fairecasoimeme/Zlinky_TIC #249: how can we get a complete quirk for ZHA implementation v12/13? I think there's promise here, but what a mess ;-)

DolmensDude, 2024-03-27:
before anyone say to use zigbee2mqtt, I tried that route 3 times already, and garbage/partial/incomplete data from ZLinky to HomeAssistant too, so I would have a slight preference for native ZHA, *if* we can (help) get the quirk to actually match all of the various parameters available.

I have EDF, HC/HP/Weekends. Have requested historical mode, without reporting per half hour from EDF.  No change yet, but still within the 72 hours they request to make such a change. We'll see... Debugging this is why I think we need a "full function" quirk, with everything exposed.

if a quirk cannot do this, then we should develop a sensors file to include in configuration file to expose everything.

Why request historical mode? 3-phase, seems according to docs seems to indicate I might need to stay with historical mode.  

I have the external antenna version of the ZLinky.  It shipped to me one week ago with version v1 of the firmware. I updated to v13 using OTA functionality of the zigbee2mqtt integration.

What I think we need just to start is having a full quirk for the ZHA implementation, including the ability to *write* certain parameters, such as setting tri-phase mode.

Nice to have would be to put the useful/common sensors to also be defined, such as converting HA to kWh for at-the-moment current draw.

if getting the quirk to be complete is only a question of copy/pasting from the developer's documentation into some kind of table for the quirk, that's probably mundane work, but if we need someone to do it, I'll *help*, someone still needs to tell me exactly what to do, and receive that table or whatever it is back from me.

Oh, I speak and write English and Dutch, but not French - I'm having to machine translate everything to understand what I need to do. Am new to France, and therefore also ZLinky.

I'm not totally convinced my ZLinky is okay. It shipped with v1 firmware, even though lixee shop said it had to ship with v13... since I updated it, that issue is moot, but simple things like resetting... The docs say hold the reset button for 10 seconds and it will reset... Mine???  It doesn't matter at all how long I hold the button in (yes, I'm def hitting the button as the hole is slightly off)... It's when I *release* the button that it seemingly resets...

fairecasoimeme, 2024-03-28:
Sorry,

But the version 1 doesn’t exist. You can verify here : https://github.com/fairecasoimeme/Zlinky_TIC/releases
I have already tell you that HA don’t find the real version in the right location. HA verify the version in cluster / attribut corresponding to update version, not the release version. After OTA update the value is updated and you can get v13 but your device was already in v13. If when you paired the device, the blue LED blinked quickly, It was last version.

So … as I already said, if you want to get all of your indexs, you have to tell your electricity supplier to change your linky mode. By default, Linky is in historical mode (historical for historical subscription). Your subscription is not historical
So you have to ask standard mode with the Enedis F185 prestation.
After that you'll get all the subscription index.

Best regards,
Fred

DolmensDude, 2024-03-28:
Hi Fred. 

“ So you have to ask standard mode with the Enedis F185 prestation.”

This should be front and center in the documentation. This means that changing the collection mode to standard in the EDF accounts preferences is not enough?

If… the z2m *also* reports the firmware incorrectly, out of the box, this too should be documented as today, OTA updates are not supported in the quirk file so no updates happen while using ZHA(since Feb of this year). 

I literally sat there and watched the update happen under z2m to v13. If it was already v13, it would never have done yet. Pointless now, it’s on v13. If it was already v13, it would be in an update loop and it’s not. 

Not in historical mode? Yes, it most certainly as is the other linkt integration in the HACS store, using historical data via Conso API must have that to work. I’m giving that up expecting this to work. 

I used ZHA to retrieve values from the Zlinky yesterday. There is a mismatch. And values that should be filled are not, so yes, the ICT port and the Zkinky are not synced. 

Will try to impress upon EDF to get that to present the correct data. 

Switching to standard mode so that when I add solar panels, it will already be in standard mode, as that seemingly is required for “producers”.

Okay with everything if it eventually works, but both routes, ZHA and z2m, produce some non-sensical data with the exception of index 1, the total of HC and HP, only everything is in HC… even that is weird. 

Will report back if it works or if there are remaining issues. 

I found the actual quirks file source. Do you need help to get it to v13? 

> On 28 Mar 2024, at 14:55, Fairesoimeme ***@***.***> wrote:
> 
> So you have to ask standard mode with the Enedis F185 prestation.