fairecasoimeme/Zlinky_TIC #254: ZHA / mode historique. Valeur puissance incohérente et peu raffraichie

liouma, 2024-04-07:
Bonjour,
Avant tout, j'ai essayé de suivre toutes les discussions ici mais c'est vraiment compliqué. Les instructions du readme concernant ZHA sont elles à jour? -> Y aurait-il des étapes que j'aurais ratées ? un custom quirk disponible ?

-J'ai recu et installé mon capteur.
-J'ai demandé et obtenu le mode standard à EDF (aucun problème grace à vos instructions précises, merci!).
-Quelques nouveaux capteurs sont apparus, dont la Puissance en W qui était jusque là à 0.

1) La valeur de puissance présente des problèmes: 
J'ai une pince ampéremétrique connectée qui me donne des valeurs que je peux comparer à celles du linky.
On trouve une belle cohérence de forme entre l'intensité du linky et la puissance de la pince.
Mais la puissance indiquée par le linky n'a aucune correspondance avec celle de la pince ni avec le courant du linky. De plus la courbe de puissance montre que la valeur est raffraichie toutes les heures.
Comment faire pour avoir la puissance ?

2) Je n'utilise pas encore les HP / HC mais je ne vois aucune information à ce sujet.
J'ai vu que vous faisiez référence à la création de sensors, je suis ok avec cela. Mais vu la date des publications à ce sujet je ne sais pas si c'est toujours d'actualité. D'où mon message.
Merci pour votre aide !!!

Informations générales sur mon installation:
- je suis à la Réunion chez EDF
- je n'utilise pas HP / HC
- le passage d'historique à standard est récent (qqs jours)
- mon module lixee a été commandé il y a 6 mois
- j'ai une installation PV mais l'injection n'est pas encore opérationnelle
- j'utilise ZHA sans quirck pour ce module
[image]
[image]
[image]

liouma, 2024-04-09:
J'ai essayé sous Z2M et le résultat n'est pas beaucoup plus concluant.

J'avoue que je suis vraiment très décu par le Lixee zlinky.
Au final j'ai une pince amperemétrique qui me donne de bien meilleurs résultat.

fairecasoimeme, 2024-04-09:
Bonjour,
Le Linky ne peut délivrer que la puissance apparente (en VA) et non pas la puissance active (en W) que peut délivrer une pince ampermétrique. La puissance en W remonté par HA est une donnée calculée et n'est pas précise.
veuillez lire la FAQ du site pour mieux comprendre : https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#4-je-ne-comprends-pas-la-puissance-affich%C3%A9e-par-le-linky-ne-correspond-pas-%C3%A0-la-puissance-mesur%C3%A9e

Si vous ne souhaitez que la Puissance Active, le Linky ne vous sert à rien. Autant utiliser des pinces ampermétriques.

Si vous avez un abonnement BASE, vous n'aurez qu'un seul index qui s'incrémente. Si vous avez d'autres abonnements, vous aurez un index pour chaque plage tarifaire.

Si vous souhaitez avec l'information d'injection, il faut suivre la FAQ : https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#3-jai-install%C3%A9-des-panneaux-photovolta%C3%AFques-et-jaimerai-r%C3%A9cup%C3%A9rer-les-infos-dinjection

liouma, 2024-04-09:
Merci pour votre réponse et cette vidéo.
Ce que je ne comprends pas c'est à quoi sert la puissance apparente. Dans quels cas de figure est elle utilisée ?
A l'inverse si je comprends bien, la puissance active permet de calculer les informations liées au cout de l'électricité ainsi que pour toutes les applications photovoltaiques.
Ai-je raté quelque chose ?

En effet, le linky sera  indispensable si je passe en HP/HC.
Merci

fairecasoimeme, 2024-04-10:
Il y a 2 possibilités pour n'avoir l'info que toutes les heures :
* un LQI un peu juste et du coup il ya des pertes de trames
* le paramétrage du config report ou du polling sur la donnée.

la puissance apparente est la puissance réellement utilisée (mais pas facturé)

liouma, 2024-04-10:
Merci. 
Est il possible de paramétrage le config report ou le polling sur ZHA ?

fairecasoimeme, 2024-04-11:
Bonjour,
je ne sais pas, j'imagine que oui mais je ne suis pas expert en HA.
Si vous avez suivi le paramétrage avancé pour la création d'entité, c'est dans ces fichiers là que le polling ou le report sont paramétrés