fairecasoimeme/Zlinky_TIC #256: Changement de fournisseur - Perte des Consommations

Coldness00, 2024-04-14:
Bonjour,

Je suis passe de chez ENEDIS (Base) a EDF avec un tarif Tempo
Jamais eu le moindre soucis, en offre de base. Cependant depuis le passage chez EDF, je n'ai plus de remonte type Live comme la puissance ni l'Energie consommée, mais j ai le numero de mon PMR, la date et l'heure, la puissance souscrite ou encore le Tarif actuel (bleu). 

Je fonctionne avec Z2M sur un Docker
Je suis passe du mode Historique a Standard (bien contrôle sur le Linky)
J'ai fait plusieurs resynchronisation
- soit en supprimant ou non le Linky du Mesh zigbee.
- j'ai débranché et rebranche le Lixee plus de 30min, mais sans succes.
- re configurtation du Lixee

Cependant, lorsque je vais dans le Dev Consol, que je sélectionne seMettering -> SumDelivered et autre information de consommation je recois bien l'information mais ne s'actualise pas apres.

Ci-joint screenshot de ma configuration ansi que de retour state:

[image]

[image]

[image]

[image]

State:
`{
    "available_power": 9,
    "current_date": "E240414183556",
    "current_index_tarif": 2,
    "current_price": "HP BLEU",
    "current_tarif": "TEMPO",
    "drawn_v_a_max_n1": 2615,
    "last_seen": "2024-04-14T16:35:56.361Z",
    "linkquality": 120,
    "message1": "PAS DE MESSAGE",
    "message2": "",
    "meter_serial_number": "VALEURCACHECONFIDENTIEL",
    "power_threshold": 9,
    "relais": 0,
    "relais_breakout": {
        "relais1": 0,
        "relais2": 0,
        "relais3": 0,
        "relais4": 0,
        "relais5": 0,
        "relais6": 0,
        "relais7": 0,
        "relais8": 0
    },
    "site_id": "VALEURCACHECONFIDENTIEL",
    "software_revision": 2,
    "status_register": "VALEURCACHECONFIDENTIEL",
    "status_register_breakout": {
        "cache_borne_dist": "ferme",
        "comm_euridis": "activee avec securite",
        "contact_sec": "ouvert",
        "depassement_ref_pow": 0,
        "etat_cpl": "nouveau_verrouille",
        "horloge": "correcte",
        "organe_coupure": "ferme",
        "pointe_mobile": "AUCUN",
        "preavis_pointe_mobile": "AUCUN",
        "producteur": 0,
        "sens_energie_active": "positive",
        "surtension_phase": 0,
        "sync_cpl": "non_synchronise",
        "tarif_dist": "index_2",
        "tarif_four": "index_2",
        "tempo_demain": "UNDEF",
        "tempo_jour": "BLEU",
        "type_tic": "standard"
    },
    "update": {
        "installed_version": 13,
        "latest_version": 13,
        "state": "idle"
    },
    "active_energy_out_d01": null,
    "active_energy_out_d02": null,
    "active_power": null,
    "active_power_max": null,
    "active_power_ph_b": null,
    "apparent_power": null,
    "average_rms_voltage_meas_period": null,
    "current_summ_delivered": null,
    "current_tier1_summ_delivered": null,
    "current_tier2_summ_delivered": null,
    "current_tier3_summ_delivered": null,
    "current_tier4_summ_delivered": null,
    "current_tier5_summ_delivered": null,
    "current_tier6_summ_delivered": null,
    "rms_current": null,
    "rms_voltage": null,
    "update_available": null
`}``

[image]

Coldness00, 2024-04-16:
[image]

Coldness00, 2024-04-23:
J'ai trouve la solution, qui en soit est ''bizarre''.
Il faut supprimer de Repporting toute les cluster/endpoints et magie, tout marche:
[image]

[image]

[image]

SBRK, 2024-09-13:
J'ai le même problème mais je ne peux rien supprimer dans "Reporting" :/