fairecasoimeme/Zlinky_TIC #259: Update vers Z2M 1.37.0 : "Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'"

meepmeep, 2024-05-05:
Hello

Depuis l'update en 1.37.0 de Zigbee2Mqtt, j'ai très régulièrement ce message qui apparait sur la console Z2M (en warning)

> Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat' 

Je ne comprends pas pourquoi ca apparait régulièrement sachant que j'ai ca dans state : 

```
{
    "active_register_tier_delivered": "HP..",
    "apparent_power": 520,
    "available_power": 60,
    "current_tarif": "HC..",
    "current_tier1_summ_delivered": 18119.856,
    "current_tier2_summ_delivered": 19373.344,
    "linkquality": 112,
    "meter_serial_number": "0xxxxxxxxxxxxxxxx",
    "rms_current": 2,
    "rms_current_max": 90,
    "schedule_h_p_h_c": 0,
    "update": {
        "installed_version": 13,
        "latest_version": 13,
        "state": "idle"
    },
    "update_available": false,
    "warn_d_p_s": 0,
    "MOTDETAT": null
}
```

Mais dans l'onglet Reporting rien ne parle de "MotDetat". 
Est ce gênant ? est ce corrigible ?

achlyss-8, 2024-05-05:
Même erreur ici, un warning toutes les minutes dans la dernière version de Z2M 😞

closed 2024-05-06 by fairecasoimeme

meepmeep, 2024-05-10:
En effet, fixé dans Z2M 1.37.1 (https://github.com/Koenkk/zigbee2mqtt/issues/21825)