fairecasoimeme/Zlinky_TIC #262: alimentation externe

nicocha75, 2024-05-19:
Bonjour
Comme le linky est assez capricieux pour l'alimentation du module, est il possible d'amener du 5V entre la broches A e I1 du zlinky?
Merci

VianneyHENON, 2024-06-13:
> bonsoir, je m'autoréponds l'alimentation via une alim 5V fonctionne très bien entre les broches A et I1. j'ai branché le linky entre I1 et I2 sans aucun soucie. j'ai déporté le zlinky à coté du portail ou j'ai le 220V qui arrive, et avec un petit module 5V, aucun soucie. Nicolas

Bonjour Nicolas,

Je rencontres des soucis également avec le lixee quand le linky est mis à jour. Le lixee ne se rallume pas et je suis obligé de le débranché 1 heure et le rebranché pour qu'il refonctionne. Je voudrais essayer de l'alimenter autrement.
Est-ce que tu pourrais me donner plus de détail sur le branchement que tu as fait et le module 5V que tu utilises. Merci.

nicocha75, 2024-06-17:
Bonjour
JE confirme, je tourne avec une alimentation 5V externe depuis maintenant 3 semaines. Plus de coupure
J'ai utilisé une alimentation que l'on trouve soit sur aliexpress soit sur ebay (500mA). 
par exemple : 
https://www.ebay.fr/itm/404954431319?itmmeta=01J0KA1KDQK5DVAEH74CQQ8GEH&hash=item5e492a3f57:g:AsAAAOSwAO1mM2xm&itmprp=enc%3AAQAJAAAA4Cqu1G0R5w369e4g1XyAIVWchSd0tG6xaUokBHZAYmvabDn1MJeBk8BArtWhfeLk%2FE5fA3tulzQy6CHS22l5H21IxZIA%2Bh6T9TEJojvucuqCejlfbh2vFrn6lr%2Bs3M9RR3wQd4kpPCDA01swzZX%2BZa6uagMRAIEqZzord408hHGEze%2FluTTbZnUxdrTTW0qez7BmvP0F06BMW7fZzTbjd%2Fes2V2g0BCr9DpVmMai52lLQL3ltP6DVmteHAZ8oX5vJxZUOPPTkPXESb6dOYQGFOAD8P1EtWavIFKtWWFOiRIX%7Ctkp%3ABFBM-LaG6oRk
Celle proposé est en plus protegé avec moins de risque sur le 220V.

VianneyHENON, 2024-06-17:
> Bonjour JE confirme, je tourne avec une alimentation 5V externe depuis maintenant 3 semaines. Plus de coupure J'ai utilisé une alimentation que l'on trouve soit sur aliexpress soit sur ebay (500mA). [...]

Merci :)

VianneyHENON, 2024-06-17:
> Bonjour JE confirme, je tourne avec une alimentation 5V externe depuis maintenant 3 semaines. Plus de coupure J'ai utilisé une alimentation que l'on trouve soit sur aliexpress soit sur ebay (500mA). [...]

Merci, pour la référence, du coup de branchement serait le suivant:
Les broches I1 et A branchées sur l'alimentation 5V et la broche I2 branchée sur le linky ? 
[image]

nicocha75, 2024-06-17:
L'alimentation entre a et i1. Et le linky connecte par les broches i1 et i2 entre le linky et le zlinky

VianneyHENON, 2024-06-17:
Cela peut servir à d'autres:

https://miniprojets.net/index.php/2021/07/28/loky-open-source-projet-pour-linky/
[image]

fairecasoimeme, 2024-06-18:
Bonjour,

vous pouvez aussi vous brancher sur le port de programmation en 3.3v
[image]

Loudramin, 2024-10-06:
J'ai eu plusieurs fois des coupures de l'alimentation, la première fois j'ai même retourné le produit en pensant qu'il était défectueux, mais le vendeur m'a indiqué qu'il était parfaitement opérationnel chez lui. 
Aujourd'hui je fais face à nouveau à des problèmes d'alimentation du Zlinky. J'ai beau l'avoir enlevé plus de 30mn du Linky comme cela m'était conseillé, cette foi les leds ne s'allume plus. J'ai contrôlé le voltage entre la borne A et I1 et cela m'indique 0.230V. J'en déduis que la tension du TIC n'est plus envoyé au zlinky. Je serai donc preneur pour avoir des photo du rendu de ceux qui ont mis une alimentation externe. J'aimerai bien voir comment cela s'intègre dans le capot du linky.

toliger, 2024-10-08:
J'avais aussi un problème d'alimentation (≃ "Zlinky_TIC rencontre des erreurs lors de l'acquisition des données du Linky") en déportant d'environ 30m le module zlinky du compteur (par 3 brins d'un cable ethernet cat 6a SFTP). J'utilise un chargeur USB afin d'alimenter le module en 5V, et ça fonctionne parfaitement.

Loudramin, 2024-10-08:
> J'avais aussi un problème d'alimentation (≃ "Zlinky_TIC rencontre des erreurs lors de l'acquisition des données du Linky") en déportant d'environ 30m le module zlinky du compteur (par 3 brins d'un [...]

Vous avez un branchement usb sur le linky ? Moi non. Ou alors je n'ai pas tout compris.

Loudramin, 2024-10-08:
> J'ai dénudé un cable USB (https://en.wikipedia.org/wiki/USB_hardware#Pinouts) en gardant le connecter USB type A, puis connecté chaque patte du zlinky à des connecteurs à levier 3 voies (Wago 221–413), puis j'ai connecté les 3 brins en direction du Linky et pour finir les 2 fils du cable correspondant +5V et du neutre. J'ai ensuite branché le cable USB dans un chargeur USB (qui délivre du 5V).

OK intéressant, merci pour l'astuce. Vous arrivez à refermer le capot vert sans soucis ?

toliger, 2024-10-08:
Dans mon cas un câble de 30 mètres sépare le zlinky du compteur, donc oui le capot se ferme ;)

VianneyHENON, 2024-10-08:
> J'ai dénudé un cable USB (https://en.wikipedia.org/wiki/USB_hardware#Pinouts) en gardant le connecter USB type A, puis j'ai connecté chaque patte du zlinky à des connecteurs à levier 3 voies (Wago 221–413), puis j'ai connecté les 3 brins en direction du Linky et pour finir les 2 fils du cable correspondant +5V et du neutre du cable USB. J'ai ensuite branché le cable USB dans un chargeur USB (qui délivre du 5V).

> > J'avais aussi un problème d'alimentation (≃ "Zlinky_TIC rencontre des erreurs lors de l'acquisition des données du Linky") en déportant d'environ 30m le module zlinky du compteur (par 3 brins d'un cable ethernet cat 6a SFTP). J'utilise un chargeur USB afin d'alimenter le module en 5V, et ça fonctionne parfaitement.
> 
> Vous avez un branchement usb sur le zlinky_tic ? Moi non. Ou alors je n'ai pas tout compris.

C'est super astucieux l'utilisation du câble RJ45 et du connecteur USB A. Merci pour 'idée.

Loudramin, 2024-10-12:
Finalement en demandant à EDF de réactiver le TIC, mon zlinky fonctionne à nouveau. J'ai quand même préparé un ancien chargeur USB pour faire le petit bricolage si ça se reproduit.

ScratMan, 2025-02-04:
Merci !! Suite à la mise en jour en v15 qui m'a obligé à le réappairer, mon Zlinky ne voulait plus démarrer.
J'ai sacrifié un câble USB pour récupérer un 5V depuis le port USB d'une multiprise placée à proximité de mon compteur, et ça a marché ! Mon Zlinky refonctionne à nouveau.

Ciredm, 2025-02-20:
Bonjour,

Il y a t'il un moyen de détecter la coupure du TIC par le Linky? Par exemple avec un délai de non réponse du Lixee? (il envoie les données périodiquement et à fréquence fixe?)

J'ai aussi perdu tout le module Lixee qui ne fonctionne plus après la mise à jour v15. J'essayerai avec une alimentation externe dès que je pourrai. Mais il est en bout de terrain et sur la rue, c'est galère de le re-appareiller.
De combien peut-on le déporter avec des câbles?

Merci de vos réponses

Loudramin, 2025-02-20:
Je suis passé de la v14 à la v16 et RAS.
Tu peux te servir de l'attribut last seen.

rippergun, 2025-05-04:
> Cela peut servir à d'autres:
> 
> https://miniprojets.net/index.php/2021/07/28/loky-open-source-projet-pour-linky/ [image]
> 
> [image]

Bonjour, j'ai fait le même branchement cet après midi et depuis mon lixee clignote à intervalle d'une seconde (1 clignotement par seconde).

Je loupe quelque chose ?

Ciredm, 2025-05-10:
Mon ZLinky était trop loin du mon routeur (limite). En le connectant avec une alimentation externe, j'ai pu le flasher correctement en passant à la v16 et récupérer la communication en le déportant (câble Ethernet) légèrement plus proche: de l'autre coté du mur) et cela fonctionne bien maintenant.

J'ai pris du temps car j'ai mis jour mon routeur (sonoff dongle-p) ne sachant pas si cela aiderai. Pour cela j'ai dû attendre d'avoir du temps de dispo pour tout reconfigurer si je plantais le réseau (heureusement cela n'a pas été le cas).

Bref, je ne mettrai plus à jour mon Zlinky!

Merci a tous!