fairecasoimeme/Zlinky_TIC #265: Plantage de Z2M suite à l'appairage du ZLinky

Xiaoh-123, 2024-05-31:
Bonjour,

J'ai récemment fait la migration du firmware de ma Sonoff Dongle Plus E vers le firmware 7.4.2 (désormais ember et plus ezsp) suite à la notification de Z2M 1.37-1.

Depuis, à chaque fois que j'appaire le ZLinky, j'ai de très nombreux messages d'erreur à l'appairage puis lors des envois de messages MQTT, souvent le point commun est le message "Not connected to MQTT server!"

J'ai mis le log après une fresh install de Z2M et l'appairage uniquement du ZLinky [ici](https://pastebin.com/0dJxcPJr).

J'ai l'impression que le fait d'appairer le ZLinky "corrompt" l'installation de Z2M car les problèmes semblent persister même après que je le supprime. L'appairage d'autres devices (Sonoff SNZB-02 et SNZB-04, Nous A1Z) se fait sans problème.

Est-ce que le problème viendrait du ZLinky ou plutôt de Z2M ou du nouveau firmware ?

gjaegy, 2024-06-14:
Bonjour,

Je pense avoir un problème assez proche, où Z2M plante de manière aléatoire, assez fréquemment. 

J'avais contacté l'auteur de Z2M à ce sujet, celui-ci semble indiquer que le ZLinky bombarde de messages, cf. le thread suivant:

https://github.com/Koenkk/zigbee2mqtt/issues/23013

Une idée de la manière de fixer le problème ?

thomasGomes, 2024-06-14:
> Bonjour,
> 
> Je pense avoir un problème assez proche, où Z2M plante de manière aléatoire, assez fréquemment.
> 
> J'avais contacté l'auteur de Z2M à ce sujet, celui-ci semble indiquer que le ZLinky bombarde de messages, cf. le thread suivant:
> 
> [Koenkk/zigbee2mqtt#23013](https://github.com/Koenkk/zigbee2mqtt/issues/23013)
> 
> Une idée de la manière de fixer le problème ?

Même chose pour moi ;)

fairecasoimeme, 2024-06-17:
Attention ce n'est pas le ZLinky qui bombarde message de lui même. C'est selon comment on le configure.
Dans Z2M pour régler le problème, il faut limiter le poll en utilisant la whitelist. ou supprimer les rapports non utiles

gjaegy, 2024-06-17:
Merci pour le retour.
J'ai bien compris le système de whitelist, par contre qu'entendez vous par la suppression des rapports non utiles ? Je suppose que cela revient au même mais je ne vois pas comment procéder dans la pratique pour cette deuxième option ?

MilesTEG1, 2024-06-17:
Bonjour,
Pour faire une whitelist, c'est bien ici que ça se passe ?
```
Tic command whitelist
List of TIC commands to be exposed (separated by comma). Reconfigure device after change. Default: all
```
[image]

Qu'est-ce qu'on doit mettre comme `TIC commands` ?
Ce sont les éléments listés dans Expose ? EAST, EASF01, EASF02 etc...
[image]

Et dernière question : `Reconfigure device after change.` c'est utiliser ce bouton ?
[image]

fairecasoimeme, 2024-06-18:
Bonjour,
Oui, c'est exactement ça @MilesTEG1

fairecasoimeme, 2024-06-18:
> Merci pour le retour. J'ai bien compris le système de whitelist, par contre qu'entendez vous par la suppression des rapports non utiles ? Je suppose que cela revient au même mais je ne vois pas comment procéder dans la pratique pour cette deuxième option ?

Dans l'onglet "rapport", vous avez un groupe de fonction. il faut désactiver (si possible) les fonctions qui ne vous sont pas utiles pour limiter le trafic

MilesTEG1, 2024-06-18:
> Bonjour, Oui, c'est exactement ça @MilesTEG1

Bonjour,
Nickel, j'ai mis ceci :
`EAST,EASF01,EASF02,EASF03,EASF04,EASF05,EASF06,LTARF,URMS1,UMOY1,SINSTS,IRMS1,Linkquality`
Et j'ai reconfiguré le ZLinky.
Depuis, je n'ai plus les erreurs occasionnelles de récupération de valeurs que j'avais encore (voir une autre issue).

Xiaoh-123, 2024-06-18:
J'ai fait une whitelist aussi et cela semble résoudre pas mal de problèmes. Après j'ai toujours un souci avec le passage d'Historique à Standard, mais ça doit venir du Linky lui-même car il affiche Historique mais mon fournisseur m'a confirmé que je suis en Standard chez Enedis (?!).
Au moins cela fonctionne à nouveau et c'est le plus important, je trouve juste ça étonnant que tout ait commencé après la màj du firmware de ma Sonoff Dongle-E.

gjaegy, 2024-06-18:
Ça semble fonctionner pour moi aussi, merci 👍