fairecasoimeme/Zlinky_TIC #266: Erreur récurrente dans Z2M : `Failed to read zigbee attributes: Error: ZCL command 0x00158d0005d25e9d/1 liXeePrivate.read(["currentPrice","currentTarif","message1","message2"]`

MilesTEG1, 2024-06-03:
Bonjour,
J'ai depuis quelques temps ces erreurs dans Z2M :

```
Failed to read zigbee attributes: Error: ZCL command 0x00158d0005d25e9d/1 liXeePrivate.read(["currentPrice","currentTarif","message1","message2"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":2695,"apsFrame":{"profileId":260,"clusterId":65382,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":30},"zclSequence":77,"commandIdentifier":1} timed out after 10000ms)

Failed to read zigbee attributes: Error: ZCL command 0x00158d0005d25e9d/1 liXeePrivate.read(["currentPrice","currentTarif","message1","message2"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":2695,"apsFrame":{"profileId":260,"clusterId":65382,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":97},"zclSequence":14,"commandIdentifier":1} timed out after 10000ms)

Failed to read zigbee attributes: Error: ZCL command 0x00158d0005d25e9d/1 liXeePrivate.read(["currentPrice","currentTarif","message1","message2"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":2695,"apsFrame":{"profileId":260,"clusterId":65382,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":21},"zclSequence":133,"commandIdentifier":1} timed out after 10000ms)
```

J'utilise l'add-on Zigbee2MQTT dans HAOS

### Zigbee2MQTT version

1.38.0 

### Adapter firmware version

7.4.3 [GA]

Même chose avec la 7.4.1 [GA]

### Adapter

HA SkyConnect with ember driver

### Setup

Add-on dans Home Assistant OS sur un NUC à base d'Intel

Je ne suis pas le seul à avoir ces erreurs, mais personne n'arrive à dire qui est en cause : Z2M ou le ZLinky.

Peut-être aurais-je plus de chance ici.

PS : je précise que mon ZLinky est en firmware limited v13, et qu'il y a plusieurs prises connectées Zigbee pas loin.
Voilà un schéma de mon maillage:
[image]

Le ZLinky est bien connecté avec un LQI satisfaisant.

fairecasoimeme, 2024-06-04:
Bonjour,

Les messages d'erreurs semblent être des "timeout" sur une demande d'info.
Apparemment, le timeout est systématiquement sur "currentPrice","currentTarif","message1","message2"
Pour savoir si c'est un des attributs qui ne réponds pas, il faudrait utiliser la console et faire une requête sur chacun des attributs

MilesTEG1, 2024-06-04:
Bonjour @fairecasoimeme 

Avec la console dev, 
- currentPrice seul : pas de soucis
- currentPrice + currentTarif : pas de soucis
- currentPrice + currentTarif + message1 : problème (timeout)
- currentPrice + currentTarif + message2 : pas de soucis
- currentPrice + currentTarif + message1 + message2 : problème (timeout)
- message1 : pas de soucis
- message1 + message2 : pas de soucis

C'est bien l'ajout de message1 qui pose problème.

[image]

> Il faut peut-être jouer avec le Meausrement poll chunk pour limiter le nombre d'attribut à poll en même temps.

Comment je fais cela ? Je veux dire, je mets quoi comme valeur ?
[image]

fairecasoimeme, 2024-06-04:
je ne suis pas expert mais de ce que je comprends, c'est que le paramètre permet de faire des demandes d'attributs par paquet.
dans votre cas, je forcerais à 1. ce qui est étrange car, apparemment c'est la valeur par défaut mais on voit bien qu'il fait des demandes par 4

MilesTEG1, 2024-06-04:
> je ne suis pas expert mais de ce que je comprends, c'est que le paramètre permet de faire des demandes d'attributs par paquet. dans votre cas, je forcerais à 1. ce qui est étrange car, apparemment c'est la valeur par défaut mais on voit bien qu'il fait des demandes par 4

Oh ! Il ne semble plus y avoir ces erreurs-là !!
Top, merci :)

En revanche, l'erreur du timeout persiste dans la console dev comme précédemment :)

fairecasoimeme, 2024-06-04:
si dans le console dev, vous faites 4 demandes, le chunck n'intervient pas

MilesTEG1, 2024-06-04:
> si dans le console dev, vous faites 4 demandes, le chunck n'intervient pas

Ok, je comprends mieux, et je m'en doutais du coup ^^
Merci pour votre aide.

closed 2024-06-04 by MilesTEG1