fairecasoimeme/Zlinky_TIC #267: Plantages réguliers depuis mars 2024

Overnuts, 2024-06-15:
Bonjour, 
J'ai un zlinky installé depuis décembre 2022, à environ 1m20 d'un coordinateur sonoff, dans la meme piece ( son lqi oscille autour de 110). 
J'utilise ZHA, j'installe les maj HA régulièrement, avec une semaine de décalage pour éviter les problèmes.
Mon réseau est principalement composé de thermomètres et prises lexman.
Tout marchait très bien jusqu'à une mise a jour HA de mars (surement la 24.3), mais depuis, mon lixee plante régulièrement, environ 3 semaines.
Obligé de le débrancher du linky, quand je le remets, tout revient tout seul dans HA.
En interrogeant le device, je suis en sw_build_id : 4000-0011
je vois qu'il y a une version 13 dispo, mais la maj ota n'aboutit pas, ca mouline, les % vont au bout, mais a la fin timeout 148.

fairecasoimeme, 2024-06-17:
Bonjour,
En effet, la mise à jour en 4000-0013 pourrait régler le problème des plantages.
Vous utilisez quel coordinateur zigbee sous ZHA ?

Overnuts, 2024-06-17:
De ce qui semblait être la dernière "bonne" façon de faire les maj, j'ai ajouté dans ma config (indenté comme il faut hein ;-)

>zha:
  custom_quirks_path: /config/custom_zha_quirks/
  zigpy_config:
    ota:
      z2m_remote_index: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json

Après quelques minutes, j'ai récupéré des maj pour des prises noname, et ça s'est installé sans problème. 
Pour Zlinky, je vois bien que la 0x0000000d est dispo

Quand je lance la maj, ca avance tranquillement jusqu'à 100% (environ 10/15 minutes), et à la fin j'ai une histoire de timeout sur un service, et plus rien ne remonte jusqu'à ce que j'aille débrancher/rebrancher le module dans le linky.

dans le log je vois ça:

> 2024-06-15 11:33:12.778 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547343648080] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 241, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2741, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2784, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 977, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1049, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 461, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/zha/update.py", line 196, in async_install
    raise HomeAssistantError(f"Update was not successful: {result}")
homeassistant.exceptions.HomeAssistantError: Update was not successful: <Status.TIMEOUT: 148>

piellick, 2024-09-05:
Hello, je suis dans le même cas. Cela fait 2 semaines depuis la reception de mon Zlinky que j'essaye de le mettre à jours. J'ai découvers par hasard qu'il était sur un truc vieux firmware.

@Overnuts j'suis également tombé sur ce hack par hasard, j'ai d'ailleurs fais une MR sur le dernier firmware dispo depuis 3 jours la v14.
https://github.com/Koenkk/zigbee-OTA/pull/552

Niveau documentations c'est compliqué, rien dans ZHA, zigpy c'est l'enfer. 

Au bout du 10iem essais je suis encore en v1:
[image]