fairecasoimeme/Zlinky_TIC #269: PTEC erroné

RobinFrcd, 2024-08-06:
Bonjour,
J'ai installé mon ZLinky ce matin et il semblerait que le sensor `PTEC` remonte une info erronée pour la plage HC/HP. Que ce soit sur ma facture ou le site d'EDF, je vois la plage suivante :
[image]
En revanche, le capteur remonte :
[image]

D'où peut venir cette différence ? C'est directement un problème du Linky qui remonte le mauvais tarif ou j'ai mal configuré quelque chose ?

Merci !

ggrandou, 2024-08-07:
J'ai eu un probleme similaire il y a quelques années (pas avec le Zlinky, mais peu importe)
Le linky ne peut pas inventer les plages horaires, il retransmet celles qu'ils recoit d'Enedis.
Le probleme etait une désynchronisation de la base EDF (le fournisseur) avec celle d'Enedis. J'ai remonté le problème à EDF et quelques mois plus tard, j'ai recu un remboursement dcorrespondant a une consommation estimée entre la consommation durant les heures creuses techniques (du Linky) et les heures creuses commerciales (d'EDF)

En tout etat de cause, rien a voir avoir le Zlinky

RobinFrcd, 2024-08-07:
Okay, merci pour l'info, je vais les contacter aussi dans ce cas !
Ils ont fini par corriger le problème ou tu as juste eu une compensation ?

closed 2024-08-07 by RobinFrcd

ggrandou, 2024-08-08:
> Okay, merci pour l'info, je vais les contacter aussi dans ce cas ! Ils ont fini par corriger le problème ou tu as juste eu une compensation ?

Ca a été corrigé