fairecasoimeme/Zlinky_TIC #272: MaJ firlware lixee zlinky tic

Sleinous, 2024-09-03:
Bonjour,
Ce post pour demander si d'autres ont l'astuce pour mettre à jour le firmware via OTA avec un taux de réussite élevé...
Depuis la v5 je galère a chaque monté de version pour passer la MAJ, en général au moins 20 essaie avant que ça passe... que je fasse la maj via Z2M, ZHA ou autre... rien y fait.
Le comportement : la maj via OTA se lance, et au bout de X% (variable) ça s'arrete et le module est hors ligne. (Il faut ensuite le débrancher rebrancher pour retenter la maj).
Je suis actuellement entrain de passer de la v13 a la v14 et comme a chaque fois je galère... J'en suis a mon 7eme essai.
Siquelqu'un a des lumieres je suis preneur :)

MilesTEG1, 2024-09-03:
@Sleinous salut 👋🏻 
Est-ce que ton ZLinky est loin de ton coordinateur Zigbee ?
Si oui c'est peut être ça la cause des échecs de MAJ OTA.

Rentente plusieurs fois, car souvent ça reprend le chargement de la Maj à l'endroit où elle s'est arrêtée.

Si ça ne fonctionne pas mieux , il faudrait mettre un répéteur zigbee (pas sur du nom là) puissant entre le coordinateur et ton Zlinky.
Et/ou des routeurs pas loin comme des prises ou ampoules connectées.

Sinon il te reste la manière brutale : faire la Maj par usb via un adaptateur usb-uart.
J'ai fait une issue sur la Maj d'un firmware standard vers un limited où j'explique pas à pas comment flasher . Cette méthode fonctionne aussi pour mettre à jour une version standard vers une version plus haute standard elle aussi.

Sleinous, 2024-09-03:
Hello @MilesTEG1 Merci pour ton retour rapide :)
En effet mon coordinateur est a moins de 20cm du lixee, il s'agit d'un zigbee coordinator en PoE de chez Tubeszb : https://tubeszb.com/product/cc2652p2-based-zigbee-to-poe-coordinator-2023/

A priori quand ça plante et que je suis obligé de débrancher rebrancher ça n'a pas l'air de reprendre de la ou ça c'est arreté :( (en tout cas je l'ai laissé 30mn sans retenter un update juste pour voir)

Pour le flash via USB-TTL, impossible de mon coté, j'ai passé 2h dessus mais ça ne marche pas avec mon USB-TTL --> FT232 FTDI Mini USB

Sleinous, 2024-09-03:
C'est enfin passé au bout de 17 essai :)

sdk69, 2024-09-04:
bonjour, j'ai essayé de faire la dernière maj en OTA hier, elle a planté et depuis plus rien ne fonctionne.
J'ai essayé de débrancher 1h, de réappairer...rien n'y fait, j'ai une erreur "Not connected to MQTT server!" alors que mes autres périphériques fonctionnent, si je débranche le lixee, tout rentre dans l'ordre.

Merci pour votre aide !

Sleinous, 2024-09-04:
C'est curieux que meme apres debranchage rebranchage ça ne revient pas, ZHA ou Zigbee2MQTT?

sdk69, 2024-09-04:
j'ai réussi à refaire fonctionner le zlinky en forçant la suppression et débrancher/rebrancher, je suis sur Z2M, mais du coup j'ai perdu tout mon historique de prod solaire et conso réseau :( et impossible de faire la maj OTA le problème se reproduit systématiquement, tant pis je vais en rester là, merci !

Mookunicorn, 2024-09-05:
Salut a tous :)

Je tente également de mettre à jour le zlinky de 13 a 14 sur ZHA. (Première mise a jour que je fais)

Durant toute l'installation le zlinky clignote
[image]

Une fois arrivé a 100% l'appareil ne réponds plus sous ZHA je dois débrancher et rebrancher.
ZHA qui d'ailleurs m'envoie un message d'erreur 
"Échec de l'exécution de l'action update/install. Update was not successful: <Status. TIMEOUT: 148>" et me rebalance la mise a jour dispo 
[image]

Le zLinky est l'appareil le plus proche du coordinateur (skyconnect) je peux d'ailleurs difficilement faire plus proche)

bombaata, 2024-09-06:
Même problème ici, depuis la tentative de MAJ via ZHA le Zlkiny ne répond plus. Je suis à distance, je débrancherai/rebrancherai ce soir.

[image]

xabix, 2024-09-06:
J'ai également le même problème. Avant j'étais sous Jeedom avec Z2M et jamais de soucis sur les updates et depuis que je suis sous HA pourtant avec Z2M j'ai pas réussi à faire d'upgrade.

Par contre j'ai du passé à EMBER depuis et je me demande si cela n'est pas lié?

SeafireGit, 2024-09-09:
Pareil de mon coté, avec HomeAssistant + Z2M, en passage de 13 à 14. La MaJ freeze à un moment random et obligation de débrancher/rebrancher pour reprendre la main dessus.

SBRK, 2024-09-12:
Je suis dans le même cas mais j'ai l'impression d'avoir des problèmes plus profonds avec le ZLinky.

Depuis des mois je n'avais plus aucune remontée d'information. Je me suis remis sur le sujet il y a quelques jours, j'ai débranché/rebranché et réappairé dans Z2M. Tout semblait ok pendant quelques minutes. Puis à nouveau aucune remontée d'info, et des erreurs de timeout sur les commandes envoyées au ZLinky. La portée du signal n'est à priori pas en cause.

Le seul moyen pour que ça fonctionne à nouveau c'est de débrancher et rebrancher le ZLinky, mais ça ne refonctionne pas si je rebranche immédiatement. La lumière bleue ne s'allume pas et je dois attendre une quinzaine de minutes avant de rebrancher. Puis au bout de quelques minutes rebelote dans Z2M, des timeouts.

Au milieu de tout ça, impossible de mettre à jour en v14, j'ai réussi à monter à 33% puis pouf, timeout à nouveau.

Mookunicorn, 2024-09-12:
> Salut a tous :)
> 
> Je tente également de mettre à jour le zlinky de 13 a 14 sur ZHA. (Première mise a jour que je fais)
> 
> Durant toute l'installation le zlinky clignote ![Screenshot_20240905-192133](https://private-user-images.githubusercontent.com/6381627/364883008-9c059e6d-8812-40dd-b32f-6ab5aaedd425.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNTU4NDUsIm5iZiI6MTcyNjE1NTU0NSwicGF0aCI6Ii82MzgxNjI3LzM2NDg4MzAwOC05YzA1OWU2ZC04ODEyLTQwZGQtYjMyZi02YWI1YWFlZGQ0MjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MTJUMTUzOTA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjI5ZTE5MmM4NTA1OWYyNDRhM2JhODgzNjhlYjk4ZDQ2MjZmNGJiZTc0YTc4YzQ5MThlOGUwYzg1ZDczM2ZhMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.sTP3GLWucCCldXp6CabMDyUDg7cl8FWxU6q2nk_x8hE)
> 
> Une fois arrivé a 100% l'appareil ne réponds plus sous ZHA je dois débrancher et rebrancher. ZHA qui d'ailleurs m'envoie un message d'erreur "Échec de l'exécution de l'action update/install. Update was not successful: <Status. TIMEOUT: 148>" et me rebalance la mise a jour dispo ![Screenshot_20240905-192920](https://private-user-images.githubusercontent.com/6381627/364889265-c2112ec9-e10a-4833-a99e-5596f7c694bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNTU4NDUsIm5iZiI6MTcyNjE1NTU0NSwicGF0aCI6Ii82MzgxNjI3LzM2NDg4OTI2NS1jMjExMmVjOS1lMTBhLTQ4MzMtYTk5ZS01NTk2ZjdjNjk0YmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MTJUMTUzOTA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDFmNzU5ZWNjYThkYWU0ZWMyMDNhYjM4NmFlOGQxYjFjMzkyZTJlYzcyZjFlMGI0MTMyZmNiYmIxZWY0OGZjNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.o-QEC69ZZVev7Ayc8VUyM86vg-XfsoEnc7VubcOQSko)
> 
> Le zLinky est l'appareil le plus proche du coordinateur (skyconnect) je peux d'ailleurs difficilement faire plus proche)

Finalement, c'est passé pour moi, quand j'ai fait la mises a jour une fois a 100% la led c'est éteinte, via ZHA j'ai refais installer (sans débrancher/rebrancher) ce qui a eu pour effet de passer en 14

r0m1CH, 2024-09-13:
Bonjour je galère pas mal à mettre à jour ce device... Comme vous ça bloque de manière random mais je ne peux pas simplement débrancher / rebrancher mon device car il ne redémarre jamais comme ça. je dois attendre X heures avant qu'il reparte. Une idée de ce que je loupe ? Merci d'avance.

SBRK, 2024-09-13:
@r0m1CH c'est le même comportement que j'observe. Le Linky se met en sécurité (cf la doc sur ce repo). J'ai l'impression que ça se produit quand je rebranche trop rapidement après avoir débranché le ZLinky. Maintenant j'attends toujours une vingtaine de secondes avant de rebrancher, et ça semble aider à ce qu'il se rallume bien.

En revanche, après moultes essais, toujours pas réussi à faire la MAJ. Par dépit j'ai commandé un adaptateur série USB pour faire la mise à jour comme ça.

r0m1CH, 2024-10-10:
@SBRK as-tu reussi à mettre à jour avec l'USB du coup ? Moi je galère toujours et je ne trouve aucun moyen de le faire correctement.

SBRK, 2024-10-10:
> @SBRK as-tu reussi à mettre à jour avec l'USB du coup ? Moi je galère toujours et je ne trouve aucun moyen de le faire correctement.

Oui et non. J'ai branché en USB mais je n'ai pas réussi à faire l'updste comme ça, en revanche j'ai pu update en OTA avec l'USB branché, ça a assuré l'alimentation et éviter que ça ne coupe.

r0m1CH, 2024-10-10:
ok ok. Bon aucune méthode bien fiable dommage. Merci pour le feedback.

Shasdo, 2024-10-14:
Même problème avec z2m, obligé de débrancher et rebrancher après chaque tentative échoué (j'en suis à 5) sinon il n'y a plus de remonté

J'ai en plus de ça des erreurs d'attributs en boucle et pour essayer d'éliminer les problèmes de réception, comme mon zLinky est à l’extérieur (4m), j'ai sortie une prise connecté sur un enrouleur servant de relais à mi chemin.

Si la maj va dans ce cas plus loin, progresse rapidement au début sans problème jusqu'à 20%,  je vois ensuite apparaitre des erreurs d'attributs à mesure que la maj ralentie.
Et plus les erreurs deviennent rapprochés, plus la maj stagne et fini par échouer.