fairecasoimeme/Zlinky_TIC #273: Remontée nouveaux index Linky (HPH/HCH/HPB/HCB)

kakik5280, 2024-09-03:
Bonjour,
Je pose ça comme ça ... 
J'ai des index sur mon compteur Linky en appuyant 10 fois sur le '+'.
Ces indexes correspondent à :
Heures Pleines saison Haute / Heures Creuses saison Haute
Heures Pleines saison Basse  / Heures Creuses saison Basse

Sur cette page j'ai vu que ça pouvait s'abréger HPH/HCH/HPB/HCB.
https://pro.engie.fr/faq/factures-paiements/votre-facture/abreviations-hce-hpe

Est-il possible d'accéder à ces données à partir de Lixee Z2M ?

fairecasoimeme, 2024-09-03:
Il faut plutôt poster côté Z2M. ça n'a pas de lien avec le produit en lui-même

closed 2024-09-03 by fairecasoimeme

kakik5280, 2024-09-03:
@fairecasoimeme Merci pour la réponse rapide.

C'était pour savoir si ces données étaient accessibles en elles-mêmes.
Quel registre ? Quelle adresse ?
Car apparemment elles seraient "nouvelles".

Je n'ai rien trouvé côté Z2M.

kakik5280, 2024-09-10:
@fairecasoimeme j'ai regardé côté z2m et dans toutes les valeurs de registre existantes aucune ne correspond à ces nouveaux indexes.
Est il possible que de nouvelles valeurs soient disponibles dans le registre de 0x0702 (Simple Metering) ?

fairecasoimeme, 2024-09-11:
peu importe le nom des index
en mode standard, les différents index sont remontées sur les 10 étiquettes prévues a cet effet.
EASF01 ...... EASF10

C'est ensuite sur Z2M que la correspondance doit se faire entre le nom réel de l'abonnement et l'étiquette correspondante.

Si votre Linky est en mode historique, vous n'aurez que l'index BASE