fairecasoimeme/Zlinky_TIC #274: Plus de remonté après MAJ v14

Sapherys, 2024-09-03:
Bonjour,

j'ai fait la MAJ OTA avec Z2M vers la version 14 aujourd'hui. 
Depuis, j'ai plus de remonté d'information. Actuellement j'ai en Whitelist seulement la donnée EAIT, ce qui me suffisait jusqe-là pour le suivi.
Est-ce que quelque chose à changé par rapport à ça pour que je n'ai plus de remonté dans HAOS?

[image]

Un grand merci à vous si vous avez des pistes! :)

Coldness00, 2024-09-03:
Salut, regarde la solution que j ai trouve sur le post:
https://github.com/fairecasoimeme/Zlinky_TIC/issues/256

Sapherys, 2024-09-03:
> Salut, regarde la solution que j ai trouve sur le post: #256

Merci pour ta réponse. Quand tu dit "supprimer" tu veux dire disable? Parce que j'en ai une vraie tonne de lignes dans les rapports (et ca va encore plus bas):

[image]

Coldness00, 2024-09-03:
Oui, exactement.
Si ca ne marche pas, tu peux faire un re-appareillage et tout reviendra comme sur le screenshot

graelo, 2024-09-03:
@Sapherys n'oublie pas stp de nous dire si ça marche pour toi, merci bcp.

MilesTEG1, 2024-09-03:
Je referais un appairage et ensuite un reboot de Z2M.

Sapherys, 2024-09-03:
> Oui, exactement. Si ca ne marche pas, tu peux faire un re-appareillage et tout reviendra comme sur le screenshot

Malheureusement ça n'a pas fonctionné chez moi, même avec un redémarrage Z2M. Je vais faire un ré appareillage en rentrant  pour voir si ça change quelque chose.
Mais déjà si d'autres ont le souci, c'est que ça ne vient pas de mon zlinky, c'est rassurant. J'en ai perdu 1 il y a 2 jours, plus de led, plus rien 😅

Coldness00, 2024-09-03:
Je te recommande d'attendre au moins quelques heures avant d'etre sur.
Et il faut generalement le laisser debrancher 30min avant de le reconnecter suite a un changement.
Assures toi d'etre aussi dans le bon mode Historique / Standards ?

Sapherys, 2024-09-04:
MAJ du jour: Pas de nouvelles remontés, même avec un appareillage pour l'instant.

graelo, 2024-09-04:
Ok, merci pour ton retour d'exp (et navré).

Sapherys, 2024-09-04:
> Je referais un appairage et ensuite un reboot de Z2M.

Ca a fonctionné chez toi?

MilesTEG1, 2024-09-04:
> > Je referais un appairage et ensuite un reboot de Z2M.
> 
> 
> 
> Ca a fonctionné chez toi?

Je ne suis pas confronté à ce souci avec la v14.
C'était une suggestion 😊
Mais en général si tu modifies une options de paramètres spécifique dans Z2M, il faut le redémarrer.

Sapherys, 2024-09-04:
De mon coté il a fallu faire une suppression du Zlinky dans Z2M puis un réapareillage. EN le renommant à l'identique, j'ai pas eu à reconfigurer le panneau des énergies. Ca fonctionne maintenant de nouveau et j'ai beaucoup moins de choses dans "Rapport" qu'avant.

closed 2024-09-04 by Sapherys

SBRK, 2024-09-13:
Chez moi, j'ai du refaire interview + reconfigure du device dans Z2M et ça a ajouté plein de trucs dans Reporting et depuis ça marche