fairecasoimeme/Zlinky_TIC #275: Parsing error mise à jour OTA

am11r, 2024-09-03:
Bonsoir,
Je sui sous HA avec zigbee2mqtt.
J'ai bien l'information d'une mise à jour 14 disponible soit directement sous HA via "parametres" soit via zigbee2mqtt.
Dans le 1er cas si je clique mies à jour puis installer: ca tourne tourne et rien ne se passe, retour à la case depart avec la mise à jour toujours disponible...
Dans le 2e cas viz zigbee2mqtt, je choisi mise à jour OTA ca se fini par une erreur de parsing....
Une idee pour que cela fonctionne?
En vous remerciant pour votre aide
[image]
[image]

am11r, 2024-09-09:
Bonsoir,
personne n'a une idee?

fairecasoimeme, 2024-09-10:
Bonjour,
Apparemment, vous utilisez une ZiGate. Malheureusement pas de MAJ OTA compatible avec Z2M et ZiGate
Cependant, vous pouvez faire la mise à jour avec un module USB TTL. 

Les infos sde MAJ non OTA sont dans le README