fairecasoimeme/Zlinky_TIC #276: Many errors in log "No converter available"

dan73800, 2024-09-05:
Good morning

I use a Zlinky_TIC with Jeedom and Zigbee2Mqtt 1.40.0
Even with the last Z2M and the last firmware of Zlinky (V14), I always have many errors in the log:

`26185|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'meter_serial_number' ("")
26186|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_power' ("")
26187|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier1_summ_delivered' ("")
26188|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_power_ph_b' ("")
26189|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier2_summ_delivered' ("")
26190|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_date' ("")
26191|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'apparent_power' ("")
26192|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'start_mobile_point1' ("")
26193|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier3_summ_delivered' ("")
26194|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'start_mobile_point2' ("")
26195|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier4_summ_delivered' ("")
26196|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'start_mobile_point3' ("")
26197|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'apparent_power_ph_b' ("")
26198|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_energy_out_d01' ("")
26199|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier5_summ_delivered' ("")
26200|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_energy_out_d02' ("")
26201|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier6_summ_delivered' ("")
26202|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_energy_out_d03' ("")
26203|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'apparent_power_ph_c' ("")
26204|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'active_energy_out_d04' ("")
26205|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_index_tarif' ("")
26206|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'tomorrow_color' ("")
26207|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier7_summ_delivered' ("")
26208|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_price' ("")
26209|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier8_summ_delivered' ("")
26210|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier9_summ_delivered' ("")
26211|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_tier10_summ_delivered' ("")
26212|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'current_summ_delivered' ("")
26213|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'stop_mobile_point1' ("")
26214|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'stop_mobile_point2' ("")
26215|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'stop_mobile_point3' ("")
26216|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'rms_current_ph_b' ("")
26217|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'rms_current_ph_c' ("")
26218|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'message1' ("")
26219|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'message2' ("")
26220|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'days_number_current_calendar' ("")
26221|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'days_number_next_calendar' ("")
26222|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'power_threshold' ("")
26223|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'days_profile_current_calendar' ("")
26224|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'days_profile_next_calendar' ("")
26225|[2024-09-05 11:45:06] [31merror[39m: 	z2m: No converter available for 'site_id' ("")
26226|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'relais' ("")
26227|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'total_apparent_power' ("")
26228|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'active_power_max' ("")
26229|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'drawn_v_a_max_n1' ("")
26230|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'active_power_max_ph_b' ("")
26231|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'drawn_v_a_max_n1_p2' ("")
26232|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'active_power_max_ph_c' ("")
26233|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'drawn_v_a_max_n1_p3' ("")
26234|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'status_register' ("")
26235|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'average_rms_voltage_meas_period' ("")
26236|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'average_rms_voltage_measure_period_ph_b' ("")
26237|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'average_rms_voltage_meas_period_ph_c' ("")
26238|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage' ("")
26239|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage_ph_b' ("")
26240|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'rms_voltage_ph_c' ("")
26241|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'software_revision' ("")
26242|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'warn_d_p_s' ("")
26243|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'schedule_h_p_h_c' ("")
26244|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'rms_current' ("")
26245|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'rms_current_max' ("")
26246|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'available_power' ("")
26247|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'MOTDETAT' ("")
26248|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'current_tarif' ("")
26249|[2024-09-05 11:45:07] [31merror[39m: 	z2m: No converter available for 'active_register_tier_delivered' ("")`

I have tried to remove and reinclude the Zlinky, but always the same errors in the log
My Zlinky works fine and all values are correctly read.

[image]

So I do not understand why I receive all these errors in the log. Logs are repeated every 5mn, with always the same messages.

Somebody can help me or tell me why ?

Many thanks

fairecasoimeme, 2024-09-05:
The error says that you have a converter error. I think you have to reboot your Z2M

dan73800, 2024-09-05:
I have already reboot it many times.
The log errors are there since my first use of Z2M with 1.36 and never disappear even with Z2M upgrade/reboot or firmware updates of the ZLinky

fairecasoimeme, 2024-09-05:
So I don't know but it's a Z2M converter problem. Perhaps, do you have an external converter
I'm not expert with Z2M. I advise you to see with Z2M

closed 2024-09-05 by fairecasoimeme