fairecasoimeme/Zlinky_TIC #277: Relevé courant haché sur Home Assistant via ZHA

piellick, 2024-09-06:
Hello,
j'ai un comportement étrange sur la partie courant via Zlinky.

Pour résumer: 
home Assistant en version `2024.9.0`
Zlinky flashé avec la dernière version (`v14`)
Linky triphasé en mode standard

la liste des capteurs:
[image]

la puissance apparente ressemble à ca:
[image]

Mais coté courant je suis a zéro avec de temps en temps une remontée de valeur:
[image]

piellick, 2024-09-12:
c'est du fonctionnement normal pour le courant ? J'ai une crete autour de 400WH, il devrait avoir un légé courant en permanence non ?

blauret, 2024-10-31:
Je pense voir un quelque chose de similaire, ma puissance apparante a change quant je suis passe en mode standard. 

[image]

Mon intuition est qu'il y a un probleme sur la remonte de la puissance apparante mais seulement en mode standard.