fairecasoimeme/Zlinky_TIC #279: mise à jour v14

am11r, 2024-09-10:
Bonjour,

J'ai finalement fait la mise à jour via USB TTL.
sous Z2M j'ai maintenant:
[image]

mais sous HA paramètres:
[image]

D'ou ma question, suis je bien en version 14? si oui pourquoi il m'est affiché version 1?

Merci à vous

MilesTEG1, 2024-09-10:
Salut 
Tu as flashé quelle version du firmware v14 ? La standard ou la limited ?

am11r, 2024-09-10:
salut,
j'ai été en standard, puis fais en limited et revenu en standard. J'ai meme fais en flashsant ou simple mise à jour mais cela n'a rien changé au final...

MilesTEG1, 2024-09-10:
Donc tu es en standard.
Tu as flashé avec l'usb et ensuite as tu essayé de flasher avec Z2M ?
Car moi après avoir flashé la v14 standard en usb , Z2M m'a dit de le mettre à jour : je l'ai fait et ça s'est fini correctement et indique la version v14.

am11r, 2024-09-10:
Oui tout pareil mais toujours v1….

am11r, 2024-09-10:
je viens de retenter et la barre bleue défile encore te encore avant que tout revienne à l'etat initial: version installé V1, disponible v14.... grrrr....

fairecasoimeme, 2024-09-11:
Il y a un bug sur Z2M qui ne va pas chercher la bonne valeur de la MAJ quand le ZLinky est flashé par USB. Donc vous êtes bien en V14. la version du firmware vous indique bien 4000-0014

closed 2024-09-11 by fairecasoimeme