fairecasoimeme/Zlinky_TIC #284: lixee: Failed to read zigbee attributes

OUARZA, 2024-09-23:
Bonjour à tous,
Depuis toujours, j'ai ces erreurs.
Cela ne semble pas poser problème dans la remontée d'information sauf peut-être une surcharge de z2m.
Comment faire pour remédier au problème ?
Merci par avance :)

`2024-09-23 20:57:50zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 haMeterIdentification.read(["availablePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (160) waiting for response TIMEOUT)
2024-09-23 20:59:01zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 haElectricalMeasurement.read(["rmsCurrentMax"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (162) waiting for response TIMEOUT)
2024-09-23 20:59:11zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 seMetering.read(["activeRegisterTierDelivered","meterSerialNumber"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (163) waiting for response TIMEOUT)
2024-09-23 20:59:22zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 haMeterIdentification.read(["availablePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (164) waiting for response TIMEOUT)`

Firmware : 4001-0013 (mise à jour en cours vers v14).
Je suis sous z2m via Jeedom.
[image]

fairecasoimeme, 2024-09-24:
Bonjour,

S'il y a quelques timeout, ça peut venir de perturbations, d'une qualité de réception moyenne ou d'un défaut de traitement côté serveur (Z2M)
pour remédier à ce dernier, il faut essayer d'utiliser la whitelist (dans paramètres spécifiques de Z2M) et de mettre que les étiquettes essentielles que vous utilisez.

OUARZA, 2024-09-24:
Bonjour,
Je ne pense pas que ça vienne de la qualité de réception.
Entre les timeout, j'ai les infos qui passent.
Je vais regarder un peu plus en détail les logs.
Mathieu

OUARZA, 2024-09-24:
`info 2024-09-24 19:48:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0005ce76da', payload '{"active_enerfy_out_d01":5126,"active_energy_out_d01":5624.11,"active_power":568,"active_power_max":1061,"active_power_ph_b":352,"active_register_tier_delivered":"","apparent_power":557,"available_power":6,"average_rms_voltage_meas_period":234,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":16023.12,"current_tarif":"BASE","current_tier1_summ_delivered":0,"current_tier2_summ_delivered":0,"device":{"applicationVersion":12,"dateCode":"20240422","friendlyName":"0x00158d0005ce76da","hardwareVersion":1,"ieeeAddr":"0x00158d0005ce76da","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":61512,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0014","stackVersion":2,"type":"Router","zclVersion":3},"drawn_v_a_max_n1":0,"last_seen":"2024-09-24T17:48:42.559Z","linkquality":255,"message1":"","message2":"","meter_serial_number":"041964502262","mot_d_etat":"000000","power_threshold":6,"relais":0,"relais_breakout":{"relais1":0,"relais2":0,"relais3":0,"relais4":0,"relais5":0,"relais6":0,"relais7":0,"relais8":0},"rms_current":2,"rms_current_max":0,"rms_voltage":235,"site_id":"","software_revision":2,"status_register":"003A4001","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_1","tempo_demain":"UNDEF","tempo_jour":"UNDEF","type_tic":"standard"},"update":{"installed_version":1,"latest_version":14,"state":"available"},"update_available":true,"warn_d_p_s":0}'

warning 2024-09-24 19:48:52zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 haElectricalMeasurement.read(["rmsCurrentMax"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (113) waiting for response TIMEOUT)

info 2024-09-24 19:49:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0005ce76da', payload '{"active_enerfy_out_d01":5126,"active_energy_out_d01":5624.11,"active_power":568,"active_power_max":1061,"active_power_ph_b":352,"active_register_tier_delivered":"","apparent_power":557,"available_power":6,"average_rms_voltage_meas_period":234,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":16023.13,"current_tarif":"BASE","current_tier1_summ_delivered":0,"current_tier2_summ_delivered":0,"device":{"applicationVersion":12,"dateCode":"20240422","friendlyName":"0x00158d0005ce76da","hardwareVersion":1,"ieeeAddr":"0x00158d0005ce76da","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":61512,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0014","stackVersion":2,"type":"Router","zclVersion":3},"drawn_v_a_max_n1":0,"last_seen":"2024-09-24T17:49:01.596Z","linkquality":255,"message1":"","message2":"","meter_serial_number":"041964502262","mot_d_etat":"000000","power_threshold":6,"relais":0,"relais_breakout":{"relais1":0,"relais2":0,"relais3":0,"relais4":0,"relais5":0,"relais6":0,"relais7":0,"relais8":0},"rms_current":2,"rms_current_max":0,"rms_voltage":235,"site_id":"","software_revision":2,"status_register":"003A4001","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_1","tempo_demain":"UNDEF","tempo_jour":"UNDEF","type_tic":"standard"},"update":{"installed_version":1,"latest_version":14,"state":"available"},"update_available":true,"warn_d_p_s":0}'

warning 2024-09-24 19:49:03zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 seMetering.read(["activeRegisterTierDelivered","meterSerialNumber"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (114) waiting for response TIMEOUT)

info 2024-09-24 19:49:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0005ce76da', payload '{"active_enerfy_out_d01":5126,"active_energy_out_d01":5624.11,"active_power":568,"active_power_max":1061,"active_power_ph_b":352,"active_register_tier_delivered":"","apparent_power":557,"available_power":6,"average_rms_voltage_meas_period":234,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":16023.13,"current_tarif":"BASE","current_tier1_summ_delivered":0,"current_tier2_summ_delivered":0,"device":{"applicationVersion":12,"dateCode":"20240422","friendlyName":"0x00158d0005ce76da","hardwareVersion":1,"ieeeAddr":"0x00158d0005ce76da","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":61512,"powerSource":"Mains (single phase)","softwareBuildID":"4000-0014","stackVersion":2,"type":"Router","zcl [... cut ...] ct_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_1","tempo_demain":"UNDEF","tempo_jour":"UNDEF","type_tic":"standard"},"update":{"installed_version":1,"latest_version":14,"state":"available"},"update_available":true,"warn_d_p_s":0}'`

fairecasoimeme, 2024-09-26:
Vous êtes en mode historique. Pour avoir plus de données, il faut passer votre Linky en mode standard. 
Pour cela, il faut suivre la procédure suivante :  

Pour passer son compteur Linky du mode historique au mode standard :

    si vous êtes chez EDF, vous pouvez effectuer la demande à travers ce [formulaire](https://particulier.edf.fr/fr/accueil/espace-client/contact/demande/formulaire/decret-conso.html#/selection-droit-exerce) de votre espace en ligne, en choisissant ensuite "Modifier le statut de la Télé Information Client (TIC)" puis en précisant que vous souhaitez passer en mode standard dans le champ commentaire.
    si vous n'êtes pas chez EDF, il faut appeler votre Fournisseur d'Electricité, et demander le passage de votre compteur Linky en mode standard, en précisant qu’il s’agit de la prestation F185 auprès d'Enedis.

Cette opération est gratuite et en général, le changement est effectué dans les 24 heures ouvrées.

OUARZA, 2024-09-26:
Merci, je suis en standard côté Enedis et côté z2m.
J'ai mis que ces paramètres : EAST,IRMS1,URMS1,Linkquality
Ces 4 données remontent bien mais je continue à recevoir des erreurs.
Que faire ?
Merci :)

[image]

[image]

```
2024-09-26 21:05:55zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0005ce76da/1 haElectricalMeasurement.read(["rmsCurrentMax"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (no response received (198) waiting for response TIMEOUT)
```

fairecasoimeme, 2024-09-27:
Bizarre qu'il demande rmsCurrentMax alors que vous ne l'avait pas dans la whitelist non ?

KipK, 2024-10-11:
Alors ja'i eu ce soucis de timeout, ca m'a pris un bon momenet avant d'en trouver la cause.
Z2M a changé le paramètre par défaut **Measurement poll chunk** sur 4 au lieu de 1 précedemment.
On ne le voit pas car le champs de  config est vide par défaut.
Et le module Lixee n'a pas l'air d'apprécier. 
Depuis que je l'ai forcé sur 1 plus d'erreur.

OUARZA, 2024-10-16:
Bonsoir @KipK ,
Je suis à 60.

Mathieu

KipK, 2024-10-16:
@OUARZA tu confond ave cle paramètre Measurement poll interval qui est à 60

c'est Measurement poll chunk qu'il faut changer. Cela dit je l'ai repaassé à 3 et ça a l'air d'etre correct pour l'instant.

MilesTEG1, 2024-10-16:
> @OUARZA tu confond avec le paramètre Measurement poll interval qui est à 60
> 
> 
> 
> 
> 
> c'est Measurement poll chunk qu'il faut changer. Cela dit je l'ai repaassé à 3 et ça a l'air d'etre correct pour l'instant.

Salut 👋🏻 
Moi je l'ai mis à 1.
Est-ce que ça peut causer des soucis d'avoir une valeur de 1 au lieu de 3 ?

KipK, 2024-10-16:
ca cause un peu plus. j'étais à 1 aussi comme évoqué plus haut pour résoudre le soucis de ce topic. Puis remonté à 3 et pas d'erreur ( sauf quand le réseau est chargé avec une ota en cours )