fairecasoimeme/Zlinky_TIC #285: Erreur depuis passage en mode standard.

blackmessor, 2024-09-27:
Bonjour, j’ai 1Z Linky avec le Firmware 14 sous zigbee2mqtt. Mon abonnement électrique est tempo. Tout fonctionnait très bien lorsque mon compteur était en mode historique, mais depuis le passage en mode standard, je n’arrive plus à faire fonctionner le Z Linky, j’ai tenté un réapairage de zéro, mais au moment d’intégrer le réseau, cela fait même planter mon zigbee2mqtt qui n’arrive plus à se connecter au broker à cause de message mal formater 

Pour faire face à cela. J’ai limité les données, renvoyer par le zlinky, et cela fonctionne mieux, pourtant, au bout de quelques minutes, à chaque fois, les données cessent de remonter, comme si le compteur était planté.  
y a t’il une solution. ? 

Merci d’avance.

freeddoo, 2024-09-29:
Salut à tous je suis exactement dans le même cas pour les valeurs
Je suis en HP/HC mode standard et il reste constamment sur HC. Si je mets le tarif sur AUTO il me plante mon installation.

Merci de votre aide

fairecasoimeme, 2024-09-30:
Pouvez-vous montrer les messages d'erreurs ou des logs ?

freeddoo, 2024-10-01:
{
    "active_energy_out_d01": 0,
    "active_energy_out_d02": 0,
    "active_power": 0,
    "active_power_max": 0,
    "active_power_ph_b": 0,
    "apparent_power": 1230,
    "available_power": 45,
    "average_rms_voltage_meas_period": 0,
    "current_date": "",
    "current_index_tarif": 0,
    "current_price": "",
    "current_summ_delivered": 0,
    "current_summ_received": 0,
    "current_tarif": "HC..",
    "current_tier1_summ_delivered": 3888.432,
    "current_tier2_summ_delivered": 3292.003,
    "drawn_v_a_max_n1": 0,
    "injected_active_load_n": 0,
    "injected_active_load_n1": 0,
    "injected_v_a": 0,
    "injected_v_a_max_n": 0,
    "injected_v_a_max_n1": 0,
    "last_seen": "2024-10-01T18:19:51.030Z",
    "linkquality": 185,
    "message1": "",
    "message2": "",
    "meter_serial_number": "032361061720",
    "reactive_power": 0,
    "reactive_power_ph_b": 0,
    "reactive_power_ph_c": 0,
    "relais": 0,
    "relais_breakout": {
        "relais1": 0,
        "relais2": 0,
        "relais3": 0,
        "relais4": 0,
        "relais5": 0,
        "relais6": 0,
        "relais7": 0,
        "relais8": 0
    },
    "rms_current": 7,
    "rms_voltage": 0,
    "site_id": "",
    "status_register": "",
    "status_register_breakout": {
        "cache_borne_dist": "ferme",
        "comm_euridis": "desactivee",
        "contact_sec": "ferme",
        "depassement_ref_pow": 0,
        "etat_cpl": "nouveau_deverrouille",
        "horloge": "correcte",
        "organe_coupure": "ferme",
        "pointe_mobile": "AUCUN",
        "preavis_pointe_mobile": "AUCUN",
        "producteur": 0,
        "sens_energie_active": "positive",
        "surtension_phase": 0,
        "sync_cpl": "non_synchronise",
        "tarif_dist": "index_1",
        "tarif_four": "index_1",
        "tempo_demain": "UNDEF",
        "tempo_jour": "UNDEF",
        "type_tic": "historique"
    },
    "total_reactive_power": 0,
    "update": {
        "installed_version": 14,
        "latest_version": 14,
        "state": "idle"
    },
    "update_available": false,
    "warn_d_p_s": 0,
    "power_threshold": null,
    "software_revision": null
}

fairecasoimeme, 2024-10-02:
Il semblerait que votre Linky soit en mode historique et non pas standard:

vous pouvez suivre le lien suivant pour faire la demande de passage en mode standard :

* si vous êtes chez EDF, vous pouvez effectuer la demande à travers ce [formulaire](https://particulier.edf.fr/fr/accueil/espace-client/contact/demande/formulaire/decret-conso.html#/selection-droit-exerce) de votre espace en ligne, en choisissant ensuite "Modifier le statut de la Télé Information Client (TIC)" puis en précisant que vous souhaitez passer en mode standard dans le champ commentaire.
* si vous n'êtes pas chez EDF, il faut appeler votre Fournisseur d'Electricité, et demander le passage de votre compteur Linky en mode standard, en précisant qu’il s’agit de la prestation F185 auprès d'Enedis.

Cette opération est gratuite et en général, le changement est effectué dans les 24 heures ouvrées.

freeddoo, 2024-10-02:
Merci de t'intéresser au problème.
Je suis bien en mode standard, confirmé par mon fournisseur d'électricité et j'ai également sur l'écran de mon linky STANDARD MODE TIC

fairecasoimeme, 2024-10-02:
OK les logs des données montrent un mode historique en mode HCHP.

C'est quoi les messages mal formatés ? ya quoi dans l'onglet "Journaux" ?

freeddoo, 2024-10-04:
Ci-joint les logs
2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_energy_out_d01":0,"active_energy_out_d02":0,"active_power":0,"active_power_max":0,"active_power_ph_b":0,"apparent_power":2127,"available_power":45,"average_rms_voltage_meas_period":0,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":0,"current_summ_received":0,"current_tarif":"HC..","current_tier1_summ_delivered":3933.638,"current_tier2_summ_delivered":3358.982,"drawn_v_a_max_n1":0,"injected_active_load_n":0,"injected_active_load_n1":0,"injected_v_a":0,"injected_v_a_max_n":0,"injected_v_a_max_n1":0,"last_seen":"2024-10-04T17:32:13.283Z","linkquality":189,"message1":"","message2":"","meter_serial_number":"032361061720","power_threshold":null,"reactive_power":0,"reactive_power_ph_b":0,"reactive_power_ph_c":0,"relais":0,"relais_breakout":{"relais1":0,"relais2":0,"relais3":0,"relais4":0,"relais5":0,"relais6":0,"relais7":0,"relais8":0},"rms_current":6,"rms_voltage":0,"site_id":"","software_revision":null,"status_register":"003A4501","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":1,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_2","tempo_demain":"UNDEF","tempo_jour":"UNDEF","type_tic":"standard"},"total_reactive_power":0,"update":{"installed_version":14,"latest_version":14,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_tarif', payload 'HC..'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/update-state', payload 'idle'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/update-installed_version', payload '14'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/update-latest_version', payload '14'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/update_available', payload 'false'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/rms_current', payload '6'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/apparent_power', payload '2127'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_tier1_summ_delivered', payload '3933.638'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_tier2_summ_delivered', payload '3358.982'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/last_seen', payload '2024-10-04T17:32:13.283Z'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/linkquality', payload '189'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_index_tarif', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_date', payload ''
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/active_energy_out_d01', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/active_energy_out_d02', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/current_price', payload ''
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/injected_active_load_n', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/injected_active_load_n1', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/injected_v_a', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/injected_v_a_max_n', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/injected_v_a_max_n1', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/message1', payload ''
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/drawn_v_a_max_n1', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/message2', payload ''
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-contact_sec', payload 'ouvert'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-organe_coupure', payload 'ferme'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-cache_borne_dist', payload 'ferme'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-surtension_phase', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-depassement_ref_pow', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-producteur', payload '1'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-sens_energie_active', payload 'positive'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-tarif_four', payload 'index_2'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/status_register_breakout-tarif_dist', payload 'index_2'
[... cut ...]
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/available_power', payload '45'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/warn_d_p_s', payload '0'
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/power_threshold', payload ''
info 2024-10-04 19:32:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky/software_revision', payload ''

J'ai toujours HC alors que je devrais avoir HP la journée par exemple.
J'ai pas la tension qui remonte 
[image]

fairecasoimeme, 2024-10-07:
Bonjour,

Désolé mais l'information NGTF qui donne HC.. n'est pas possible. HC.. existe uniquement en mode historique.

(NGTF et OPTARIF ont le même cluster/attribut). Vous avez donc paramétrer votre Z2M en standard mais votre Linky est toujours en mode historique.

chpego, 2024-10-14:
Je suis dans le même cas (quasiment) :
Mon Linky était en mode historique, et tout était paramétré correctement dans Z2M (mode, phase, production)
Depuis le passage du mode historique à standard sur le Linky, la modification **phase** dans Z2M, fait planter Z2M qui n'arrive plus à se connecter au serveur MQTT.

Les logs:
- Z2M:
``
2024-10-14 06:07:36z2m: Cannot send message: topic: 'zigbee2mqttHA1/0x00158d0005be9f06', payload: '{"active_energy_out_d01":null,"active_energy_out_d02":null,"active_energy_out_d03":null,"active_energy_out_d04":null,"active_power":null,"active_power_max":null,"active_power_ph_b":null,"active_register_tier_delivered":"","apparent_power":429,"available_power":6,"average_rms_voltage_meas_period":null,"current_date":null,"current_index_tarif":null,"current_price":null,"current_summ_delivered":1698,"current_tarif":"Base","current_tier10_summ_delivered":null,"current_tier1_summ_delivered":1699,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"current_tier7_summ_delivered":null,"current_tier8_summ_delivered":null,"current_tier9_summ_delivered":null,"days_number_current_calendar":null,"days_number_next_calendar":null,"days_profile_current_calendar":null,"days_profile_next_calendar":null,"drawn_v_a_max_n1":null,"linkquality":255,"message1":null,"message2":null,"meter_serial_number":"022261111942","mot_d_etat":"000000","power_threshold":null,"relais":null,"rms_current":2,"rms_current_max":0,"rms_voltage":null,"schedule_h_p_h_c":0,"site_id":null,"software_revision":null,"start_mobile_point1":null,"start_mobile_point2":null,"start_mobile_point3":null,"start_notice_e_j_p":0,"status_register":"003A0000","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ferme","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_1","tarif_four":"index_1","tempo_demain":"UNDEF","tempo_jour":"UNDEF","type_tic":"standard"},"stop_mobile_point1":null,"stop_mobile_point2":null,"stop_mobile_point3":null,"tomorrow_color":"","update":{"installed_version":14,"latest_version":14,"state":"idle"},"update_available":false,"warn_d_p_s":0}
``
- MQTT
``
2024-10-14 06:07:50: Client mqttjs_9f999ecd disconnected due to malformed packet.
``

fairecasoimeme, 2024-10-14:
Essayez de passer/forcer le "poll chunk" à 1

chpego, 2024-10-22:
Alors si jamais ça peut servir à d'autres; voilà ce que j'ai réalisé pour que cela fonctionne et sans plus aucune erreur lié à MQTT.
Appairage de la clé LiXee; une fois appairé il faut la déconnecter, faire les différents réglages dans Z2M (historique/standard, mono phase/tri phasé, poll chunk à 1, etc..); puis rebranché la clé LiXee