fairecasoimeme/Zlinky_TIC #286: Appliquer un diviseur au metering

pierreprinetti, 2024-09-29:
Bonjour,

J'ai connecté le Zlinky_TIC en ZHA sur mon Home Assistant (Core v2024.9.3) qui se trouve à moins de deux mètres du capteur. Les données remontent bien, et je reçois certains attributes normalement (par example dans le cluster Basic: app_version `14`, date_code `20240422`, sw_build_id `4000-0014`).

Comme Home Assistant s'attend à recevoir des kWh et le Zlinky_TIC remonte des Wh, je voudrais enregistrer un diviseur de `1000` comme attribute `divisor (id: 0x0302)` dans le cluster `Metering (Endpoint id: 1, Id: 0x0702, Type: in)`.

Cependant: j'entre `1000` dans "Value", je tape "Write attribute", j'ai un joli check blanc sur fond vert, et rien ne se passe. L'enregistrement est toujours en Wh et si je tape "Read attribute", le champ Value prend la valeur `None`.

Est-ce que j'ai mal compris comment enregistrer un diviseur?

pierreprinetti, 2024-10-03:
En attendant une solution pour régler le problème à la base, voici comment j'ai finalement ajouté un senseur "template" dans ma configuration HomeAsssistant:

```yaml
# dans configuration.yaml

# [...]

template:
  - sensor:
    - name: "Linky"
      unique_id: "lixee_zlinky_tic_summation_delivered_millesime"
      unit_of_measurement: "Wh"
      device_class: energy
      state_class: total_increasing
      state: "{{ states('sensor.lixee_zlinky_tic_summation_delivered') | float }}"
```

J'ai ensuite ajouté ce nouveau senseur virtuel au volet "Energie" pour visualiser ma consommation en électricité.

Nbird67, 2024-10-06:
Bonjour,
J’ai exactement le même problème de diviseur
Pour moi suite à Maj firmware 14 en 13 c’était bon.

Si quelqu’un a une solution je suis preneur, j’ai du également passer par des templates 

je suis ravi de ne pas être le seul.

https://github.com/fairecasoimeme/Zlinky_TIC/issues/18

https://github.com/zigpy/zha-device-handlers/issues/3365

ced30fr, 2024-10-08:
Je confirme avoir récemment acheté un module en version 14 et rencontrer le même problème. Existe-t-il une solution pour effectuer un downgrade en attendant une correction ?

Nbird67, 2024-10-08:
Le créateur @fairecasoimeme a indiqué que ce n'est pas lié au firmware à priori, mais à quoi alors ???

uowis, 2024-10-13:
I have this issue
J'ai cette issue

yomgui1, 2024-10-15:
Même chose, je suis passé de Z2M à ZHA, j'était en firmare 14, et toutes les valeurs de puissances sont multipliées par 1000.

fairecasoimeme, 2024-10-18:
> Same here.
> 
> > Le créateur @fairecasoimeme a indiqué que ce n'est pas lié au firmware à priori
> 
> Ou?

Bin ZHA

blauret, 2024-10-18:
ok, desole je n'a pas trouve le commentaire....

`
class ZLinkyTICMetering(CustomCluster, Metering):
    """ZLinky_TIC custom metring cluster."""

    # ZLinky_TIC reports current_summ_delivered in Wh
    # Home Assistant expects kWh (1kWh = 1000 Wh)
    MULTIPLIER = 0x0301
    DIVISOR = 0x0302
    _CONSTANT_ATTRIBUTES = {MULTIPLIER: 1, DIVISOR: 1000}
`

On dirait que le code est deja la pour la division.

blauret, 2024-10-21:
L'origine du probleme dans ZHA est que le Zlinky_TIC n'est plus correctement reconnue par homeassistant, selon moi, a cause d'un changement dans la signature du Zlinky_TIC

```
2024-10-21 11:16:08.645 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LiXee ZLinky_TIC (00:15:8d:00:05:be:83:47)
2024-10-21 11:16:08.645 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTIC'>
2024-10-21 11:16:08.646 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint
```

Le resultat est que le divisor qui est dans le quirk n'est plus applique. 

Apres une modification du quirk:

```

2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LiXee ZLinky_TIC (00:15:8d:00:05:be:83:47)
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks] Device matches filter signature - device ieee[00:15:8d:00:05:be:83:47]: filter signature[{'models_info': [('LiXee', 'ZLinky_TIC')], 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.METER_INTERFACE: 83>, 'input_clusters': [0, 1, 1, 3, 10, 1794, 2817, 2820, 61184, 65382], 'output_clusters': [25, 61184]}, 242: {'profile_id': 41440, 'device_type': <DeviceType.PROXY_BASIC: 97>, 'input_clusters': [33], 'output_clusters': [33]}}}]
2024-10-21 11:20:43.017 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:15:8d:00:05:be:83:47: <class 'zhaquirks.lixee.zlinky.ZLinkyTICFWV12'>
```

Le changement

```diff
diff --git a/zhaquirks/lixee/zlinky.py b/zhaquirks/lixee/zlinky.py
index 3891fe0..57c7a0d 100644
--- a/zhaquirks/lixee/zlinky.py
+++ b/zhaquirks/lixee/zlinky.py
@@ -11,6 +11,7 @@ from zigpy.zcl.clusters.general import (
     Identify,
     Ota,
     PowerConfiguration,
+    Time,
 )
 from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement, MeterIdentification
 from zigpy.zcl.clusters.smartenergy import Metering
@@ -157,13 +158,16 @@ class ZLinkyTIC(CustomDevice):
                 DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                 INPUT_CLUSTERS: [
                     Basic.cluster_id,
+                    PowerConfiguration.cluster_id,
                     Identify.cluster_id,
+                    Time.cluster_id,
                     Metering.cluster_id,
                     MeterIdentification.cluster_id,
                     ElectricalMeasurement.cluster_id,
+                    0xef00,
                     ZLinkyTICManufacturerCluster.cluster_id,
                 ],
-                OUTPUT_CLUSTERS: [Ota.cluster_id],
+                OUTPUT_CLUSTERS: [Ota.cluster_id, 0xef00],
             },
             242: {
                 PROFILE_ID: zgp.PROFILE_ID,
@@ -182,6 +186,7 @@ class ZLinkyTIC(CustomDevice):
                     Basic.cluster_id,
                     PowerConfiguration.cluster_id,
                     Identify.cluster_id,
+                    Time.cluster_id,
                     ZLinkyTICMetering,
                     MeterIdentification.cluster_id,
                     ElectricalMeasurement.cluster_id,

``

La question pour @fairecasoimeme, est-ce que le cluster 0xFE00 est documenté quelque part? Je n'ai pas vu dans le README du projet.

fairecasoimeme, 2024-10-21:
Ok je comprends mieux.
Le fait d'avoir rajouté des clusters provoque un défaut de signature.
Les clusters rajoutés sont pour la compatibilité avec les plateformes Tuya. Le cLuster 0xEF00 n'est pas utile pour ZHA.
Je pense alors qu'il faut alors rajouter une signature correspondant au dernier firmware.

blauret, 2024-10-21:
@fairecasoimeme, j'ai mis in commentaire sur zigpy/zha-device-handlers#336 a propos de la nouvelle signature. Est-ce qu'il est possible que le cluster input 1 soit deux fois dans la base?

fairecasoimeme, 2024-10-21:
Pouvez-vous essayer ce fichier zlinky.py et me dire si la signature est détectée ?

[zlinky.py.zip](https://github.com/user-attachments/files/17461284/zlinky.py.zip)

blauret, 2024-10-21:
Je n'ai pas essayer mais j'ai cree une PR avec quariment le meme contenu zigpy/zha-device-handlers#3448

fairecasoimeme, 2024-10-21:
Ok parfait alors !

Bahanix, 2024-10-30:
Bonjour, je me suis initié à HA / ZLinky il y a quelques jours et j'ai le même souci que décrit ici, et je vois que la solution discutée plus haut a été merged. Comment procéder pour récupérer ce correctif sur mon HA / ZHA ? À moins qu'il faille juste être patient pour que l'update soit proposée du HA ?

blauret, 2024-10-31:
> Bonjour, je me suis initié à HA / ZLinky il y a quelques jours et j'ai le même souci que décrit ici, et je vois que la solution discutée plus haut a été merged. Comment procéder pour récupérer ce [...]

IL faut juste etre patient et attendre une nouvelle release de homeassistant. L'autre solution est d'utilise le quirks avec le patch localement. Les instructions sont ici:

https://community.home-assistant.io/t/zigbee-guide-how-to-add-setup-local-custom-zha-device-handlers-also-known-as-quirks-in-the-zha-integration/683473

blauret, 2024-10-31:
@fairecasoimeme , a mon avis on peux cloturer ce ticket.

fairecasoimeme, 2024-10-31:
thank you for all

closed 2024-10-31 by fairecasoimeme