fairecasoimeme/Zlinky_TIC #287: Quel mode choisir pour un abonnement Heures Creuses le soir, les week-end et jours fériés ?

Gaetanb76, 2024-10-01:
Bonjour,

Je viens d'installer le module.
J'ai un abonnement Heures Creuses le soir, les week-end et jours fériés .
Je n'ai pas trouvé comment faire remonter les consommations HC et HP.

Je ne sais quel Mode choisir, historique ou  standard.
Je ne sais quel Tarif sélectionné.
Je ne sais quelles commandes utilisées pour faire remonter les valeurs HC et HP.

En **Standard** et **Auto** j'obtiens ce qui suit :
EAST        55077 kWh
EASF01    55080 kWh
EASF02    0 kWh
PREF        45 kVA
NTARF      0
NGTF       BASE
RELAIS     0
SINSTS     390 VA
Linkquality  120 lqi

En **Historique** et **HCHP** j'obtiens ce qui suit :
PTEC       TH..
HCHC     55081 kWh
HCHP      0 kWh
OPTARIF  BASE
MOTDETAT  N/A
HHPHC   N/A
IINST     1 A
ADPS   N/A A

fairecasoimeme, 2024-10-02:
Bonjour,

Votre Linky est en mode historique. le mode historique ne sait pas gérer votre abonnement.
vous devez donc demander à votre fournisseur d'énergie de passer le Linky en mode standard : 
* si vous êtes chez EDF, vous pouvez effectuer la demande à travers ce [formulaire](https://particulier.edf.fr/fr/accueil/espace-client/contact/demande/formulaire/decret-conso.html#/selection-droit-exerce) de votre espace en ligne, en choisissant ensuite "Modifier le statut de la Télé Information Client (TIC)" puis en précisant que vous souhaitez passer en mode standard dans le champ commentaire.
* si vous n'êtes pas chez EDF, il faut appeler votre Fournisseur d'Electricité, et demander le passage de votre compteur Linky en mode standard, en précisant qu’il s’agit de la prestation F185 auprès d'Enedis.

Cette opération est gratuite et en général, le changement est effectué dans les 24 heures ouvrées.

closed 2024-10-02 by fairecasoimeme

Gaetanb76, 2024-10-03:
Bonjour, 
Merci beaucoup pour votre aide.
Demande effectuée hier. Mail de confirmation aujourd'hui.
Bonne journée.

Gaetanb76, 2024-10-06:
Bonjour,
J'ai bien les remontées de valeurs pour les HP et les HC, mais pas pour les HC week-end.
J'ai rajouté EASF03 jusqu'à EASF10 mais rien n'a  changé.
Faut-il débrancher et rebrancher le module, comme je l'ai fait après le changement de TIC, pour que les données remontent ?