fairecasoimeme/Zlinky_TIC #288: Lixee, blocage de certaines mesures

jacquot68, 2024-10-03:
Bonjour,
Depuis 1 mois, j'essaie de résoudre le problème suivant :
Mon linky est en mode Standard, triphasé et sans production. Connecté via Z2M. Sa version est "Version du firmware  4000-0014"
Les infos remontent correctement pour une durée variable puis j'ai quatre grandeurs (concernant la phase 2 et 3) qui restent bloquées sur une valeur fixe alors que le reste remonte normalement.
J'ai mis en place un white list 

"EASF01,EASF02,EASF03,EASF04,EASF05,EASF06,SINSTS1,SINSTS2,SINSTS3,URMS1,URMS2,URMS3,CCASN,SINSTS,HHPHC"
 
Mais cela ne change rien, le blocage des informations concernant la phase 2 et 3 est toujours présent.

Un redémarrage de Z2M permet d'obtenir à nouveau toutes les informations pour une durée aléatoire.

Ci-dessous payload infos `bloquées:
`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_energy_out_d01":3937.9,"active_energy_out_d02":8057.3,"active_power":276,"active_power_max":20,"active_power_max_ph_b":1040,"active_power_max_ph_c":3170,"active_power_ph_b":219,"apparent_power":19,"apparent_power_ph_b":42,"apparent_power_ph_c":451,"available_power":12,"average_rms_voltage_meas_period":238,"average_rms_voltage_meas_period_ph_c":241,"average_rms_voltage_measure_period_ph_b":235,"current_date":"E240924162219","current_index_tarif":2,"current_price":"HP BLEU","current_summ_delivered":13940.1,"current_tarif":"TEMPO","current_tier1_summ_delivered":10392.8,"current_tier2_summ_delivered":2827,"current_tier3_summ_delivered":89.4,"current_tier4_summ_delivered":403.8,"current_tier5_summ_delivered":50.7,"current_tier6_summ_delivered":176.3,"drawn_v_a_max_n1":430,"drawn_v_a_max_n1_p2":4440,"drawn_v_a_max_n1_p3":2190,"linkquality":152,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"022176056507","power_threshold":12,"relais":0,"relais_breakout":{"relais1":0,"relais2":0,"relais3":0,"relais4":0,"relais5":0,"relais6":0,"relais7":0,"relais8":0},"rms_current":0,"rms_current_ph_b":0,"rms_current_ph_c":5,"rms_voltage":239,"rms_voltage_ph_b":232,"rms_voltage_ph_c":234,"site_id":"06390882722955","software_revision":2,"status_register":"013A4401","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_2","tempo_demain":"UNDEF","tempo_jour":"BLEU","type_tic":"standard"},"total_apparent_power":389,"update":{"installed_version":14,"latest_version":14,"state":"idle"},"update_available":null}'`

Payload après re-demarrage Z2M:
`2024-10-03 08:35:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_energy_out_d01":3937.9,"active_energy_out_d02":8058.1,"active_power":307,"active_power_max":20,"active_power_max_ph_b":1040,"active_power_max_ph_c":3170,"active_power_ph_b":327,"apparent_power":18,"apparent_power_ph_b":42,"apparent_power_ph_c":453,"available_power":12,"average_rms_voltage_meas_period":236,"average_rms_voltage_meas_period_ph_c":241,"average_rms_voltage_measure_period_ph_b":235,"current_date":"E240924162219","current_index_tarif":2,"current_price":"HP BLEU","current_summ_delivered":13940.8,"current_tarif":"TEMPO","current_tier1_summ_delivered":10392.8,"current_tier2_summ_delivered":2827.8,"current_tier3_summ_delivered":89.4,"current_tier4_summ_delivered":403.8,"current_tier5_summ_delivered":50.7,"current_tier6_summ_delivered":176.3,"drawn_v_a_max_n1":430,"drawn_v_a_max_n1_p2":4440,"drawn_v_a_max_n1_p3":2190,"linkquality":148,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"022176056507","power_threshold":12,"relais":0,"relais_breakout":{"relais1":0,"relais2":0,"relais3":0,"relais4":0,"relais5":0,"relais6":0,"relais7":0,"relais8":0},"rms_current":0,"rms_current_ph_b":0,"rms_current_ph_c":5,"rms_voltage":236,"rms_voltage_ph_b":230,"rms_voltage_ph_c":240,"site_id":"06390882722955","software_revision":2,"status_register":"013A4401","status_register_breakout":{"cache_borne_dist":"ferme","comm_euridis":"activee avec securite","contact_sec":"ouvert","depassement_ref_pow":0,"etat_cpl":"nouveau_verrouille","horloge":"correcte","organe_coupure":"ferme","pointe_mobile":"AUCUN","preavis_pointe_mobile":"AUCUN","producteur":0,"sens_energie_active":"positive","surtension_phase":0,"sync_cpl":"non_synchronise","tarif_dist":"index_2","tarif_four":"index_2","tempo_demain":"UNDEF","tempo_jour":"BLEU","type_tic":"standard"},"total_apparent_power":473,"update":{"installed_version":14,"latest_version":14,"state":"idle"},"update_available":null}'`

Je constate que malgré la white list toutes les informations du Lixee remontent vers Z2M! par contre sur le tableau de board de Z2M seul la white list est affichée.

Merci de m'éclairer sur le sujet afin de trouver une solution à ce désagrément.
Cdt
Jacquot

jacquot68, 2024-10-09:
Bonjour
J'ai remarqué que les mesurent qui bloquaient ne figuraient pas dans le rapport de z2m du linky. Le fait de les avoir rajoutés semble avoir résolu le blocage.

Cdt
Jacquot

Kistrof, 2025-01-13:
Bonjour, j'ai le même problème : certaines valeurs ne sont plus mises à jour, mais je suis en mode historique et j'utilise ZHA (et non Z2M).
Les index de conso (tempo) sont figés alors que la puissance apparente et l'intensité sont bien mises à jour plusieurs fois par minute.
Si je recharge ZHA, les index remontent mais une seule fois et sont de nouveau bloqués.

fairecasoimeme, 2025-01-14:
Bonjour,

La seule chose qui peut expliquer cela, c'est que lors de l'appairage, les requêtes de BIND ou config report ne passe pas. La seule manière de corriger est de cliquer sur le bouton orange de reconfiguration pour Z2M et pour ZHA de relancer l'appairage.

Kistrof, 2025-01-14:
Effectivement j'ai corrigé en supprimant le device de ZHA et j'ai ensuite refait l'appairage, mais je ne m'explique pas ce qui a cause le soucis car ça a bien fonctionné pendant plusieurs jours (upgrade ou restart de HA ?)

Kistrof, 2025-01-24:
Update: un restart de HA peut engendrer le soucis, mais si le soucis est déjà présent, un rechargement de ZHA met à jour les valeurs, mais juste une fois (cf. screenshot, j'ai rechargé ZHA il y a 27 minutes) mais je ne comprends pas le journal d'évènements du device qui indique au moment du problème qu'il semble perdre les infos du device, jusqu'à indiquer qu'il a été éteint, alors que la valeur de courant et de puissance apparente continuent de remonter en temps réel.

[image]

Si je peux déjà éviter d'avoir à le supprimer et aller débrancher et rebrancher le zlinky ...

fairecasoimeme, 2025-01-24:
Bonjour,

Pourtant un redémarrage de HA n'a pas de lien avec le ZLinky mais plutôt votre coordinateur zigbee.
Je pense qu'il faut plutôt regarder de ce côté. Quel coordinateur utilisez-vous ?

Kistrof, 2025-01-27:
C'est un Sonoff Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21)

fairecasoimeme, 2025-01-28:
Peut-être essayer de passer en mode firmware limited si vous avez un réseau zigbee avec beaucoup d'appareils qui passent par le ZLinky

Kistrof, 2025-01-28:
Je n'ai que 9 appareils Zigbee : 5 endpoints, 3 dongles Sonoff (1 coordinateur et 2 routeurs) et le ZLinky.
A priori, le ZLinky ne capte que les 3 dongles, mais je crois que parfois il capte 2 ou 3 des 5 appareils.
Je peux essayer de le passer en mode limited (pas sûr de la marche à suivre en OTA), vu son emplacement il n'a aucun intérêt à faire routeur, c'est lui l'appareil le plus éloigné de tout !

Kistrof, 2025-01-28:
Update: j'ai lu la version du firmware du ZLinky via ZHA, ça a fait planter la collecte ... comment je peux voir si le problème vient du ZLinky, du coordinateur ou de ZHA ? (logs de débuts de ZHA ?)