fairecasoimeme/Zlinky_TIC #289: Pas de remontée des HC Week-end

Gaetanb76, 2024-10-06:
Bonjour,
Après avoir rempli le [formulaire](https://particulier.edf.fr/fr/accueil/espace-client/contact/demande/formulaire/decret-conso.html#/selection-droit-exerce) j'ai les remontées de valeurs pour les HP et les HC, mais pas pour les HC week-end.
J'ai rajouté EASF03 jusqu'à EASF10 mais rien n'a changé.
Faut-il débrancher et rebrancher le module, comme je l'ai fait après le changement de TIC, pour que les données remontent ?

fairecasoimeme, 2024-10-07:
Si vous êtes sur Z2M, il faut choisir le bon tarif dans paramètres spécifique et relancer la configuration (le bouton jaune du device)
Vous pouvez aussi vérifier en faisant une requête manuelle, dans console dev et faire un read sur le cluster seMetering attribut CurrentTier3SummDelivered

Gaetanb76, 2024-10-07:
Bonsoir,
Ce tarif n'est pas proposé. Je viens de mettre en auto.
Je viens de faire la requête manuelle et cela a fonctionné : EASF03 est apparu avec la bonne valeur.
J'ai changé le tarif pour revenir à Standard - Heure Pleine Heure Creuse mais EASF03 disparait.
En choisissant Standard - Heures Super Creuses ça fonctionne.

Merci beaucoup. 👍

closed 2024-10-07 by Gaetanb76