fairecasoimeme/Zlinky_TIC #290: Maj ota normal vers limited et vis versa

KipK, 2024-10-15:
Salut

Ce n'est pas une issue, mais je savais pas trop où poster ça.

A tester, voici 2 images qui devraient permettre de passer du firmware normal vers limited et vis versa sen ota ( sans flasher en TTL).
Il s'agit des version 14, j'ai juste modifié les headers OTA pour changer l'image type et le numero de version ( à 9999 pour forcer la maj ) .
Une fois flashé, il sera bien en v14, avec le bon type. 

Les fichiers sont là:

https://drive.google.com/drive/folders/10hbOs632Vi_YDl_JNedt6z7iwb4lhdoH?usp=sharing

Je n'ai pas pu tester car mon zlinky est en prod, mais il n'y a aucun risque. Au pire ca refusera de flasher. 
Mais ça devrait fonctionner, j'ai déjà testé avec un firmware dont j'ai juste modifié le n: de version pour voir.

Pour le faire sous Zigbee2mqtt, il faut modifier un fichier  , car la gestion du local ota est broken pour le lixee avec la gestion ota actuelle ( [un PR est en cours](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8136) ) 
Ce PR permettra surtout d'avoir les maj automatiques dans z2m aussi pour le firmware limited. 

Il faut modifier le fichier :
/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lixee.js ( ou lixee.ts si vous compilez à partir des sources )
Si docker c'est ici ( il faut se connecter au terminal du container )
/app/node_modules/zigbee-herdsman-converters/devices/lixee.js

Remplacer ligne 646:
`ota: ota.lixee, `
par
`ota: ota.zigbeeOTA,`

Ensuite pour activer le local ota, ajouter dans configuration.yaml de z2m:

```
ota:
  zigbee_ota_override_index_location: localotaindex.json
```

Et créer le fichier localotaindex.json ( au même niveau que le configuration.yaml ):
[
    {
        "url": "ota/ZLinky_router_v14-limited2normal.ota"
    }
]

ou 

[
    {
        "url": "ota/ZLinky_router_v14-normal2limited.ota"
    }
]

Et placer le fichier dans le dossier ota ( le créer si absent ) 
Et redémarrer z2m.

fairecasoimeme, 2024-10-15:
Ok merci pour la contribution.
Je vais regarder tout ça (je ne suis pas expert Z2M) et tester ta procédure.
Du coup, pour les prochaines mises à jour, il faut uploader manuellement les OTA ?

KipK, 2024-10-15:
non, les fichiers restent hébergés sur ton repo git. 
La procédure est indiqué ici: https://github.com/Koenkk/zigbee-OTA

T'as juste un script à executer, et soumettre le PR après, ca fera du coup un PR par fw.

fairecasoimeme, 2024-10-15:
ok j'ai compris 👍

KipK, 2024-10-15:
J'ai soumis hier le limited pour tester. Faudra que tu le refasse pour la prochaine version si tu veux garder le controle de tout ça. 
L'avantage c'est que par la suite à chaque maj, les nouveaux PR seronts acceptés automatiquement.

KipK, 2024-10-15:
Quelqu'un avait déjà soumis le V14 en normal d'ailleurs. Mais inutile pour l'instant tant que le pr [#8136](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8136) est pas merge.

fairecasoimeme, 2024-10-15:
Ok ça me parait clair.
Merci !

fairecasoimeme, 2024-10-15:
par contre les PR sont merged une fois par mois ? non ?

KipK, 2024-10-15:
par pour ça, le index.json est telechargé par z2m pour les maj ota. Il n'est pas intégré dans les update z2m.
Peut être pour le premier PR par contre, j'ai fais la soumission hier, on va bien voir.

KipK, 2024-10-15:
https://github.com/Koenkk/zigbee-OTA/commits/master/
C'est bien mis à jours en temps réel ( pour ceux dont le 1er PR a déjà été validé ).

Donc si tu intègre ça dans ton CI/CD ca devrait pousser automatiquement par la suite.

fairecasoimeme, 2024-10-15:
parfait ! merci !

KipK, 2024-10-15:
Les PR sont passés, on est bon pour la prochaine release.

KipK, 2024-10-15:
De rien.  C'était pas grand chose au final. 
Par contre pour les passages de normal vers limited and co, pour l'instant ça n'a pas l'air de passer. Peut être que le Lixee ne supporte pas les images compressées. 
Il va falloir que je puisse tester ça ici, j'ai bien un deuxième Lixee, mais faut que je vois si je peux le faire tourner en DC sans être branché au linky.

fairecasoimeme, 2024-10-15:
Je peux t'envoyer un ZLinky si besoin. Sinon, oui tu peux alimenter une ZLinky en appliquant une tension DC de 5 ou 9 vols sur les borne A et I1.

KipK, 2024-10-15:
C'est gentil, mais j'en ai deja 2 dont un qui ne me sert pas. 
Tu crois que je peux tester les maj ota sans qu'il reçoive de data du linky? 
Jamais testé comment ça se comporte juste alimenté sur le pad 5v . Mais si je peux m'en servir comme module de dev c'est top.

fairecasoimeme, 2024-10-16:
oui tu peux tester les maj sans qu'il recoive les datas du Linky

KipK, 2024-10-16:
Super ca fonctionne. 
Alors j'ai testé déjà la mise à jours du firmware limited sans localota avec les modifs du PR, et ca se fait tout seul automatiquement depuis l'index.json hebergé.
C'est déjà une bonne nouvelle. 

Par contre pour le sujet de ce thread, les firmware de migration ne passent pas. Déjà le lixee ne supporte pas les firmware compressé en zlib, alors j'en ai refait sans ça, mais le lixee jete l'update au début de la mise à jours :

`info 2024-10-16 09:54:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d000638e13c', payload '{"active_register_tier_delivered":"","apparent_power":65535,"current_summ_delivered":0,"current_tarif":"BASE","device":{"applicationVersion":11,"dateCode":"20240422","friendlyName":"0x00158d000638e13c","hardwareVersion":1,"ieeeAddr":"0x00158d000638e13c","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":57270,"powerSource":"Mains (single phase)","softwareBuildID":"4001-0014","stackVersion":2,"type":"Router","zclVersion":3},"last_seen":"2024-10-16T07:54:34.010Z","linkquality":188,"meter_serial_number":"","mot_d_etat":"000000","rms_current":65535,"rms_current_max":0,"update":{"installed_version":14,"latest_version":9999,"state":"available"},"update_available":null,"warn_d_p_s":0}'
info 2024-10-16 09:54:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x00158d000638e13c"},"error":"Update of '0x00158d000638e13c' failed (Update failed with reason: 'ABORT')","status":"error","transaction":"eublw-3"}'
error 2024-10-16 09:54:34z2m: Update of '0x00158d000638e13c' failed (Update failed with reason: 'ABORT')`

Il doit y avoir des data dans le header du .bin qui sont vérifié par le device ( car si j'ai bien compris le header du .ota n'est pas envoyé au device, il envoie juste le .bin extrait du .ota une fois que le coordinateur a vérifié les metas du header .ota )

Ce qui veut dire qu'il faudrait compiler des firmware de migration depuis les sources qui permettrait de faire ça. Et là j'avoue que le code lixee me parle autant que du chinois pour l'instant ^^

fairecasoimeme, 2024-10-16:
Erf, ya un mécanisme de chiffrage lors de la création d'un OTA avec contrôle du CRC il me semble.

KipK, 2024-10-16:
J'ai cru lire ça oui. Voila le script que j'ai utilisé pour générer ces fichiers ota ( qui fonctionne si je genere un .ota depuis le .bin juste en changeant le n: de version ( et pas le type )  Ce qui me laisse penser que le crc est géré par la lib zigpy

`#!/usr/bin/env python
# create-ota - Create zlib-compressed Zigbee OTA file
# Copyright 2023  Simon Arlott
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

import argparse
import functools

import zigpy.ota

def create(filename, manufacturer_id, image_type, file_version, header_string):
	with open(filename, "rb") as f:
		data = f.read()

	image = zigpy.ota.image.OTAImage(
		header=zigpy.ota.image.OTAImageHeader(
			upgrade_file_id=zigpy.ota.image.OTAImageHeader.MAGIC_VALUE,
			header_version=0x0100,
			header_length=0,
			field_control=zigpy.ota.image.FieldControl(0),

			manufacturer_id=manufacturer_id,
			image_type=image_type,
			file_version=file_version,

			stack_version=2,
			header_string=header_string[0:32],
			image_size=0,
		),
		subelements=[
			zigpy.ota.image.SubElement(
				tag_id=zigpy.ota.image.ElementTagId.UPGRADE_IMAGE, data=data,
			)
		],
	)

	image.header.header_length = len(image.header.serialize())
	image.header.image_size = image.header.header_length + len(image.subelements.serialize())
	return image.serialize()

if __name__ == "__main__":
	any_int = functools.wraps(int)(functools.partial(int, base=0))
	parser = argparse.ArgumentParser(description="Create zlib-compressed Zigbee OTA file",
		epilog="Reads a firmware image file and outputs an OTA file on standard output")
	parser.add_argument("filename", metavar="INPUT", type=str, help="Firmware image filename")
	parser.add_argument("output", metavar="OUTPUT", type=str, help="OTA filename")
	parser.add_argument("-m", "--manufacturer_id", metavar="MANUFACTURER_ID", type=any_int, required=True, help="Manufacturer ID")
	parser.add_argument("-i", "--image_type", metavar="IMAGE_ID", type=any_int, required=True, help="Image ID")
	parser.add_argument("-v", "--file_version", metavar="VERSION", type=any_int, required=True, help="File version")
	parser.add_argument("-s", "--header_string", metavar="HEADER_STRING", type=str, default="", help="Header String")

	args = parser.parse_args()
	output = args.output
	del args.output

	data = create(**vars(args))
	with open(output, "wb") as f:
		f.write(data)`

KipK, 2024-10-16:
Apperement, il y a bien un autre header qui est testé, en verifiant avec zigpy le fichier ota 

`zigpy ota info ZLinky_router_v14-normal2limited.ota`

Il me sort:

ZLinky_router_v14-normal2limited.ota
Type: <class 'zigpy.ota.image.OTAImage'>
Header: OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl: 0>, manufacturer_id=4151, **image_type=2**, file_version=9999, stack_version=2, header_string=<''>, image_size=177233, *device_specific_file=False, *hardware_versions_present=False, *security_credential_version_present=False)
Number of subelements: 1
Validation result: ValidationResult.UNKNOWN

Hors le header ota est bien sur type 1, donc il retrouve cette valeur de 2 ailleurs.

KipK, 2024-10-16:
Ca y est j'ai réussis à faire passer les maj de transitions.

J'ai tenté un paquets de trucs, même éditer les .bin avec un editeur hexa, mais le firmware était rejeté à la fin du telechargement.

En fait la solution c'est qu'il manquait le meta de description qui doit etre du même nom que le firmware installé. 
Tout marche j'ai mis à jours le premier post, et les fichiers sur le drive. 
Teté dans les 2 sens sans soucis

KipK, 2024-10-16:
voila les 2 commandes pour générer les fichiers ota depuis le script fourni dans le dossier:

` python .\create-ota-nozip.py -m 4151 -i 1 -v 255 -s OM15081-RTR-JN5189-0000000000000 .\ZLinky_router_v14_limited.bin .\ZLinky_router_v14_normal2limited.ota`

`python .\create-ota-nozip.py -m 4151 -i 2 -v 255 -s OM15081-RTR-LIMITED-JN5189-00000 ZLinky_router_v14.bin ZLinky_router_v14_limited2normal.ota`

fairecasoimeme, 2024-10-17:
Merci @KipK pour le travail, je vais regarder tout ça dès que possible.

cperrier, 2024-10-29:
Bon, comme je suis un gros impatient, et surtout que je voulais pouvoir confirmer à KipK que c'est vraiment extra ce qu'il a fait, j'ai voulu faire la manip moi-même tout seul comme un grand, sans attendre la MAJ de Z2M.

Je crois avoir tout bien fait, MAIS j'ai mis les DEUX fichiers "limited2normal" et "normal2limited" dans le répertoire ota/ que j'avais créé....et j'ai créé le localotaindex.js identique à ce qu'il y a dans le premier post....en oubliant de lire "Ne gardez que l'objet necessaire," (en bon neuneu qui va trop vite)...et j'ai modifié le configuration.yaml pour ajouter la section "ota:"

Redémarrage Z2M, je lance un check des MAJ firmware, ça me propose de mettre le firmware du ZLinky à jour, ça se déroule bien et......je pense que je suis toujours en normal vu que j'ai un de mes thermostats Zigbee qui est toujours connecté au ZLinky.

Qu'à cela ne tienne, me dis-je.....j'enlève le fichier limited2normal du répertoire ota/ je modifie le localotaindex.js pour ne référencer que l'autre et je redémarre Z2M et.....bin rien.

Y'a sûrement un truc que je fais de traviole, mais je sèche. Ô grand gourou, peux-tu venir à mon aide? :-)

KipK, 2024-10-29:
pour vérifier dans quel mode tu es, regarde dans l'onglet About de l'appareil:
Firmware version 4000-0014

4000 = normal
4001 = limited

cperrier, 2024-10-29:
Ah, alors je suis en normal...

En fait, comme j'avais mis les deux FW dans le répertoire ota/ et que mon ZLinky était (il me semble) en version 13 et non 14 (il y a plusieurs semaines, la MAJ OTA en v14 m'avait été proposée mais passait son temps à échouer vu que le ZLinky passait son temps à se "planter" à cause de trop de périphériques pour qui il était routeur), eh bien, la MAJ s'est faite....en version 14 "normale".

Et si ça se trouve, cela s'est fait par le canal "classique"

Donc, il me reste maintenant à arriver à passer en limited, mais si ça se trouve, il y a qq chose que je n'ai pas compris qq part dans le post de KipK (ce qui serait absolument sans surprise, vu mon mode neuneu).

Ce que j'ai fait dans mon Home Assistant:
- ouvrir le terminal
- aller dans zigbee2mqtt
- créer un répertoire ota/
- mettre dedans les deux FW récupérés su rle drive de KipK
- créer un fichier localotaindex.json avec le contenu indiqué dans le premier post
- ajouter la section "ota:" dans le configuration.yaml de Z2M
- **chercher (et ne pas trouver) la ligne 646 mentionnée par KipK dans lixee.js, donc aucune modif là**
- redémarrer Z2M
- aller dans "Maj OTA"
- Et là, actuellement....rien...:-)

Je pense qu'il y a un truc que j'ai mal compris (ce qui, une fois de plus, serait tout à fait normal car tout cela relève un peu de la magie noire pour moi)

cperrier, 2024-10-29:
Mon z2m, il est installé dans les Modules Complémentaires de mon HA.....

KipK, 2024-10-29:
Du coup c'est du docker aussi. Cf la doc de l'addon HA:

_The repository includes two add-ons:
Zigbee2MQTT is the stable release that tracks the released versions of Zigbee2MQTT. (recommended for most users)
Zigbee2MQTT Edge tracks the dev branch of Zigbee2MQTT such that you can install the edge version if there are features or fixes in the Zigbee2MQTT dev branch that are not yet released._

trollix, 2024-11-02:
Bonjour,
Ma mise à jour s'est bien passée: normal to limited.
Mais il y a cette erreur dans les logs de z2m:

`
[2024-11-02 09:43:33] error: 	z2m: Failed to configure 'EM-LI:9AED:LIXEE', attempt 1 (Error: Bind 0x00158d0005be9aed/1 seMetering from '0x00124b00256c4f05/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lixee.ts:1803:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-02 09:44:10] error: 	z2m: Failed to configure 'EM-LI:9AED:LIXEE', attempt 2 (Error: Bind 0x00158d0005be9aed/1 seMetering from '0x00124b00256c4f05/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lixee.ts:1803:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
[2024-11-02 09:44:31] error: 	z2m: Failed to configure 'EM-LI:9AED:LIXEE', attempt 3 (Error: Bind 0x00158d0005be9aed/1 seMetering from '0x00124b00256c4f05/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lixee.ts:1803:13)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))

`
'TABLE_FULL', est-ce une erreur corrigerable ?

Merci d'avance

KipK, 2024-11-02:
Oui j'ai déjà eu cette erreur. Réinitialise le device et reappaire. Ou peut être juste relancer l'apprentissage et la configuration

cperrier, 2024-11-02:
De mon côté, la mise à jour s'est également faite avec le nouveau Z2M : j'avais ajouté un  localotaindex.json avec juste la mention du firmware "normaltolimited" et le firware correspondant dans le répertoire ota/, plus l'ajout des deux lignes de section "ota:" dans le configuration.yaml de z2m

Par contre, après la mise à jour: 
1) le ZLinky  ne renvoyait plus de mises à jour de ses capteurs, qui restaient aux valeurs précédentes. Résultat : mon chauffe-eau ne s'est pas déclenché cette nuit (j'ai une automatisation qui dépend des heures creuses Tempo). Il est reparti normalement après déconnexion/reconnexion sur le Linky (sans désapparaige/réappairage)
2) par contre, le firmware renvoyé est toujours "4000-0014" mais semble bien être limited car aucun périphérique ne l'utilise comme routeur

KipK, 2024-11-02:
Redémarre z2m voir. Sinon refais un apprentissage il devrait mettre à jours la version.

cperrier, 2024-11-02:
Merci !

Redémarrer Z2M, j'ai déjà fait mais ça n'a rien changé. Je vais voir en resettant le ZLinky (appui 10s sur le bouton de reset si je ne me trompe pas) et en le réassociant. En tout cas, un grand merci pour tout le boulot sur ce truc (ça me donne des espoirs que ça bouge aussi pour mes modules Fil Pilote Leroy-Merlin...:-) ).

KipK, 2024-11-02:
T'auras remarqué que depuis la maj les modules LM apparaissent dans l'OTA. Fait le fw maintenant..

Nitratz, 2024-11-05:
Hello,

Dans mon cas j'ai aucun appareil qui se connecte a mon LiXee mais parcontre j'ai des valeurs qui sont figées, j'ai essentiellement que la apparent_power qui change entre chaque update...

Est ce que le firmware limited va aider pour ce probleme ?

Thanks!

gauthier12, 2024-12-04:
La procédure de @KipK  a marché nickel chez moi. Merci

cperrier, 2025-01-20:
Hello à tous,

Désolé, je déterre un peu  cette discussion car, après un fonctionnement sans trop de soucis pendant 2 mois avec mon ZLinky, j'ai recommencé à "perdre la main" dessus régulièrement. Cela s'est produit après avoir migré mon installation HA sur un SSD connecté en USB (afin d'éviter de dépendre d'une brave carte mémoire).

J'ai initialement pensé à des problèmes d'interférences causés par la connexion USB du SSD. Mais même en le déplaçant sur un port USB 2.0, j'ai continué à "perdre" mon ZLinky après 1h ou 2 de fonctionnement normal. Pire, même en revenant en désespoir de cause sur la carte mémoire (donc en gros en revenant complètement en arrière), il m'a fallu je ne sais combien d'essais pour arriver un ZLinky qui s'appaire....et reste appairé (et, encore, je ne suis pas du tout certain que ça va tenir).

J'avais fait, environ fin novembre, la manip décrite dans le post 1 (un immense merci à @KipK ). Curieusement, la version de firmware reportée restait 4000-0014....mais j'étais très probablement bien en firmware limited car plus aucun périphérique Zigbee ne s'apparaît avec le ZLinky bien que j'aie depuis ajouté pas mal de périphériques sur batterie.

Aujourd'hui, après un combat épique, et 3692 aller-retours avec mon Linky, j'ai fini par arriver à réappairer le ZLinky après....avoir remis les modifications de @KipK dans mon configuration.yaml et dans un localotaindex.json.

Et là, je croise les doigts, mais ça semble OK et, en plus, la version de firmware signalée est  4001-0014, ce qui est, sauf erreur, le "v14 limited".

Bon, honnêtement, j'ai l'impression d'avoir fait pas mal de magie noire et j'étais à deux doigts de sortir les gousses d'ail et tracer un pentagramme autour de mon Raspberry Pi4...:-)

Je pense que, pou rl'instant, je ne vais plus toucher à rien (j'ai quand même pas mal d'automatisations de gestion des périodes TEMPO, qui dépendent du fait que le ZLinky arrive à envoyer les bonnes infos.....surtout que, depuis 2 semaines, EDF nous solde tous ses jours rouges!).

Je me suis dit que vous narrer toutes ces péripéties pouvait être une bonne idée, notamment pôur savoir si des choses ont bougé par ailleurs dans le monde du ZLinky....:-) (@fairecasoimeme taggué juste pour savoir et façon de remercier pour le travail fait sur le device, même si nous galérons pas mal avec!).

En fait, faudrait que je finisse par me monter une installation de dev pour faire des bidouilles trnaquillement sans tout péter. J'ai même un deuxième ZLinky dans un coin qui dort (pas sûr que je ne l'aie pas un peu bousillé, par contre, lors de tests inavouables pour mettre son firmware à jour directement avec un USB-TTL). Mais ce qui me manque, c'ets un Linky de test, par contre...:-)

Albirew, 2025-02-06:
Je viens de remarquer que pour la release de la v16, on a droit aux images limited2normal et normal2limited.
J'imagine qu'elles sont arrivée dans les releases "officielles" grâce à cet issue.
Du coup merci à toi =)

KipK, 2025-02-06:
Tu peux juste mettre l'url ca suffit.

KipK, 2025-02-09:
Tu mets juste à jours en OTA du coup, ça n'a rien à voir avec le sujet. Ici c'est pour passer de normal a limited et vis versa 🙃

luke7101, 2025-03-27:
Hello, est-ce que c'est censé marcher avec ZHA aussi ?
Si je mets le fichier ZLinky_router_v16_normal2limited.ota dans le dossier "config\zigpy_ota" avec ceci dans mon configuration.yaml
```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: advanced
          path: /config/zigpy_ota
```

luke7101, 2025-03-27:
Alors, oui! L'update finit par apparaître sur ZHA aussi, chez moi ça a pris 3h.
Donc, j'ai déclenché la mise à jour sur HA et au  bout d'une dizaine de minutes ça a atteint 100%.
`INFO (MainThread) [zigpy.device] [0x175c] OTA upgrade progress: (245886 / 245886): 100.0000%`

Le message suivant a été:
`DEBUG (MainThread) [zigpy.zcl] [0x175C:1:0x0019] Decoded ZCL frame: Ota:upgrade_end(status=<Status.SUCCESS: 0>, manufacturer_code=4151, image_type=1, file_version=255)`

Donc normalement ça devrait avoir marché !

En revanche HA détecte toujours la présence d'une mise à jour, comme si ça n'avait pas été fait, et si j'interroge le cluster 0x0000 attribut 0x4000 j'obtiens la valeur 4000-0013, donc toujours la v13 au lieu de la v16.

Qu'est-ce que ça veut dire ? Que faire ?