fairecasoimeme/Zlinky_TIC #291: MAJ OTA impossible sous Z2MQTT

TheMinimaliiiist, 2024-10-23:
Bonjour,
J’ai un Lixee depuis plusieurs mois mais je ne l’installe que maintenant.
Je n’arrive pas à le mettre à jour en OTA via Zigbee2MQTT sous Home Assistant. Je suis apparemment sous la version 1 quand la dernière est la 14. 

Que faire ?

Silverrider5, 2024-10-26:
De mon côté même problème, je suis en v13 et impossible de passer en v14 via Z2MQTT, le process se lance mais pas de progression, la barre reste en N/A....30 fois que je tente, mon coordonateur est à 15 cm du Zlinky donc ce n'est pas la distance. Une solution ?

TheMinimaliiiist, 2024-10-27:
J’ai solutionné mon problème en refaisant l’appairage sous HA et cela a tout résolu

Silverrider5, 2024-11-09:
moi ça ne marche pas impossible de faire la MAJ malgré le re appairage.

thennart, 2025-01-10:
Finalement après retenté pour la n-ieme fois ce midi, c'est passé

Silverrider5, 2025-03-01:
de mon côté toujours impossible de mette à jour mon Zlinky bloqué en v14 malgré plusieurs re-appairage. Mais Finalement je ne regrette pas étant donné les problèmes avec les firmware 15 & 16. Frustrant quand même !

y-streibel-uiris, 2025-08-22:
De mon coté, j'ai la version 1.6 du lixee, son firmware est a la version 12.
Je n'arrive pas non plus à avoir la mise à jour en OTA à travers Z2M.
J'ai tenté la mise a jour [nonOTA](https://github.com/fairecasoimeme/Zlinky_TIC#non-ota) avec le branchement du dongle USB TTL pas mieux non plus.
Auriez vous une solution fonctionnelle, merci ? Sinon, pourriez vous détaillé d'avantage le README.md du projet pour inclure des éléments de votre expérience que nous n'aurions pas !

ystreibel, 2025-08-22:
De mon coté, j'ai la version 1.6 du lixee, son firmware est a la version 12.
Je n'arrive pas non plus à avoir la mise à jour en OTA à travers Z2M.
J'ai tenté la mise a jour [nonOTA](https://github.com/fairecasoimeme/Zlinky_TIC#non-ota) avec le branchement du dongle USB TTL pas mieux non plus.
Auriez vous une solution fonctionnelle, merci ? Sinon, pourriez vous détaillé d'avantage le README.md du projet pour inclure des éléments de votre expérience que nous n'aurions pas !