fairecasoimeme/Zlinky_TIC #293: Remontée de la Puissance active totale délivrée

HiFuGa, 2024-10-27:
Bonjour,

Je viens de recevoir le Linky V2 avec Firmware 14 (J'avais jusqu'à récemment, un Zlinky V1 en firmware 13 que j'ai cramé en voulant le mettre à jour en USB...).

[image]

Il est paramétré comme ceci (Abonnement Tempo, monophasé, et Linky en mode Producteur) :

[image]
[image]

Les infos remontent :

[image]

J'ai la Puissance apparente immédiate délivrée (SINSTS) qui remonte toutes les 60s, La Puissance apparente instantanée injectée (SINSTI) qui remonte toutes les 10s, et le plus embêtant, les Puissance active totale délivrée par le fournisseur comme EASF02 qui remontent toutes les 7 à 15min.

[image]

[image]

Est-ce que le fonctionnement est normal, ou je devrais avoir une mise à jour toutes les 60s ?

J'aimerais avoir le moins possible d'heures pleines après 22h ou d'heures creuses après 6h.

[image]

A quelques secondes prêt, le changement se fait bien à ces horaires sur le relais du linky :

[image]

Merci !

KipK, 2024-10-28:
La puissance active est calculée depuis la consommation en kWh. Elle est donc surement synchro avec la remontée du compteur.

Je n'ai pas essayé de réduire la remontée de conso globale voir si la puissance active remontait avec, mais y a des chances.

miramo, 2024-10-30:
Bonjour, de mon coté j'ai la v1 du hardware et depuis le passage en firmware 14 (je pense, je ne sais pas quand la maj s'est faite), je rencontre le même soucis, `SINSTS` est mis a jours sur HA environ toutes les minutes, mais `EAST` et les `EASF01` etc ne sont plus mis a jours ou très rarement. Je passe par Z2M.
J'ai essayer de modifier `Measurement poll interval` a 60 et `Measurement poll chunk` a 4. Ça ne change rien.
Je suis en mode standard (le Linky est bien en standard également) avec TEMPO.

HiFuGa, 2024-10-30:
> TEMPO.

Intéressant, j’avais le V1 en firmware 13 qui effectivement remonté aussi toutes les valeurs toutes les 60s, et maintenant le v2 en firmware 14 avec des temps de remontées differents.

Du coup, ça doit lié au firmware 14 si vous avez la meme chose avec votre V1 en firmware 14.

miramo, 2024-10-31:
Je ne maîtrise pas encore qui a quelle responsabilité et à quel moment dans la chaîne. Mais il y a en tout cas une différence dans la fréquence de remontée des données.
En ce qui me concerne, sur HA, `SINSTS` remonte toutes les minutes, tandis que `EASF01` remonte toutes les 15 minutes, par exemple.
Est-ce que c’est le coordinateur qui décide de cela ?

miramo, 2024-11-04:
D’accord, merci pour les précisions. Du coup, j’imagine qu’on parle ici de l’onglet `Reporting` du device dans Z2M.

C’est assez étrange, car je suis quasiment certain de n’avoir rien modifié dans cet onglet ni touché aux paramètres spécifiques depuis la configuration initiale.

Le seul changement que j’ai identifié est la mise à jour du firmware.

Est-ce que je devrais ajuster certains paramètres dans `Reporting` ? (J’ai déjà tenté quelques modifications et relancé la configuration, sans succès).

[image]

fairecasoimeme, 2024-11-05:
Je vous confirme que vous n'avez surement pas changer le reporting. Mais dans un reporting comme vous pouvez le voir, il y a un min report et un MAX report. C'est à dire que le ZLinky va décider de remonter les données dans cet intervalle en fonction de son encombrement ou celui du réseau zigbee. (Cette partie n'a pas changé entre la version 13 et 14)
Pour fixer le délai de remontée d'une donnée, il faut par exemple mettre le Min à 60 et le max à 61 pour être sur d'avoir une remontée toutes les minutes.

Je ne sais pas si "Apply" envoie directement le config report au ZLinky ou s'il faut relancer la config complète.

miramo, 2024-11-05:
Je viens d'effectuer quelques tests avec par exemple :
[image]
J'ai "Apply" puis "Reconfigure" puis restart Z2M.
J'ai également essayer de baisser le `Min rep change`.

Toujours le même état, `SINSTS` est mis a jours toutes les minutes, `DATE` toutes les 10 minutes et les `EAST`, `EASF0X` toutes les 15 minutes.

fairecasoimeme, 2024-11-05:
DATE n'est pas compatible avec les reports.
Pour EAST mettre rep change à 0

miramo, 2024-11-05:
D'accord merci de la precision pour `DATE`, c'était surtout pour illustrer les differences (je ne me sert pas vraiment de cette donnée).

Je viens de tester avec :
[image]
J'ai "Apply" puis "Reconfigure" puis restart Z2M.

`EAST` est toujours mis a jours toutes les 15 minutes dans HA.

miramo, 2024-11-08:
D’accord, je vais tenter un réappairage ce soir. Pour cela, j’imagine qu’il faut supprimer puis ajouter à nouveau l’appareil ?

Par contre, non, je n’ai pas encore de module USB-TTL. Peut-être que @HiFuGa en a un ? (D’ailleurs, désolé, j’ai l’impression d’avoir un peu détourné ton issue ; j’espère que nous avons bien le même problème).

0rsa, 2024-11-18:
Hello,
Lixee V2 en V14, j'aimerai bien avoir un relevé toutes les 15 minutes :)
Voillà mon record basé sur la différence de apparent_power (EAST), un enregistrement n'est inséré que s'il existe une différence entre deux valeurs relevées (sachant que je charge une voiture en même temps, il y a une belle consommation régulière).

[image]

Ma config
[image]

Je n'ai rien modifié dans l'onglet Rapports et la valeur SINSTS est bien mise à jour toutes les minutes.

miramo, 2024-11-26:
De mon côté, j’ai également tenté le réappairage suite aux modifications du reporting, suite aux conseils de @fairecasoimeme, mais le problème des remontées dans HA toutes les 10 minutes pour certaines valeurs persiste.

sdalu, 2025-01-04:
> Je ne maîtrise pas encore qui a quelle responsabilité et à quel moment dans la chaîne. Mais il y a en tout cas une différence dans la fréquence de remontée des données. En ce qui me concerne, sur [...]

Je viens d'avoir le même problème ou les valeurs ne sont plus remontées toutes les minutes comme avant, est-ce que cela pourrait être dû à "Measurement poll interval" qui n'est pas effectué, et du coup on attend les 900 second max du reporting ?

NeWoKiKi, 2025-02-05:
Bonjour,

J'ai exactement le même souci que @TROM83 
Je suis en Triphasé et seules les valeurs des phases sont quasi instantanées (SINSTS1, SINSTS2 et SINSTS3), pas le total (SINSTS).

Du coup j'ai regardé dans le rapport également, les 3 phases sont bien dedans avec un polling rapide :

[image]

Par contre le total n’apparaît pas dans le tableau, j'ai donc voulu le rajouter comme pour les phases :

[image]

Mais j'ai un message d'erreur lorsque j'appuie sur "Appliquer" :

[image]

Pour info, les correspondances :
```
SINSTS : totalApparentPower
SINSTS1 : apparentPower
SINSTS2 : apparentPowerPhB
SINSTS3 : apparentPowerPhC
```

KipK, 2025-02-05:
> Par contre le total n’apparaît pas dans le tableau, j'ai donc voulu le rajouter comme pour les phases :

As tu essayé de mettre le param measurement poll chunk sur 1 ?

Souvent les timeout viennent de la valeur par défaut.

Sinon as tu bien mis dans la whitelist SINST ?

NeWoKiKi, 2025-02-05:
J'ai bien la valeur dans "Exposes" et elle s'actualise bien, mais très rarement contrairement aux phases.
C'est uniquement dans "Rapport" qu'elle n’apparaît.

J'avais déjà essayé de mettre le poll chunk sur 1 mais ca ne change rien (j'ai bien reconfiguré et reboot Z2M).
Dans le doute je viens d'essayer à nouveau et même chose.

Oui je l'ai bien dans la Whitelist, d’ailleurs je l'ai bien dans "Exposes"

Il n'y a pas moyen de change ce qui se trouve dans "Rapport" sans passer par l'interface ? Dans un fichier quelque part peut-être ?

KipK, 2025-02-05:
Bizarre. Bon tu peux toujours créer un template qui additionne tes 3 phases en attendant.

NeWoKiKi, 2025-02-05:
Oui c'est ce que j'ai fait pour le moment mais bon c'est dommage d'avoir un template de somme alors qu'un capteur existe et qu'il est juste mis a jour toutes les 10 min au lieu de 1 :)

KipK, 2025-02-05:
oui c'est passé à 600 depuis novembre 2023. Mais le texte dans l'UI n'a pas été corrigée.

NeWoKiKi, 2025-02-05:
Du coup, juste pour information, le "Measurement poll chunk" par défaut est à combien ?
Dans l'interface c'est indiqué "Default: 2", c'est toujours le cas ?
Si non, c'est recommandé de le mettre à combien ?

Je n'ai plus confiance aux champs vides maintenant donc je vais le remplir aussi xD

KipK, 2025-02-05:
Il est à 2 par défaut. Si des soucis de timeout apparaissent je conseille de le mettre à 1.
J'ai pas essayé avec le dernier fw, mais j'avais des soucis avant.

KipK, 2025-02-05:
D'autant que SINST est ok en monophasé

NeWoKiKi, 2025-02-05:
Je n'ai aucune idée du code ou du firmware, mais peut-être qu'en monophase `SINSTS` serait lié à `apparentPower` alors qu'en triphasé `SINSTS` est lié à `totalApparentPower` ?

Juste une piste au cas où...

fairecasoimeme, 2025-02-05:
> Je n'ai aucune idée du code ni du firmware, mais peut-être qu'en monophasé `SINSTS` serait lié à `apparentPower` alors qu'en triphasé `SINSTS` est lié à `totalApparentPower` ?
> 
> Juste une piste au cas où...

Oui c'est ce que je regarde dans la définition du cluster. je vous tiens au courant

fairecasoimeme, 2025-02-05:
Non le totalApparentPower cluster : 0x0B04 attribut : 0x0306 est bien en reportable.

fairecasoimeme, 2025-02-05:
Je viens de tester sur une ZiGate et ZWGUI avec un ZLinky v15

* j'ai fait le BIND sur le cluster 0x0b04
* J'ai lancé un config report sur le ZLinky avec le cluster 0x0b04 et l'attribut 0x0306 avec un min à 1, un max à 60 et un change à 20

--> le config report est bien passé et j'ai bien des remontées toutes les 60 au pire des cas.

Peut-être que côté Z2M le 0x0b04 /0x0306 n'est pas reportable. je vais regarder

NeWoKiKi, 2025-02-05:
De base, sans rien toucher, j'ai ça dans l'onglet rapport :

[image]

Le `totalApparentPower` n'apparaît pas

Comme dit dans un de mes posts plus haut, je ne peux pas le rajouter, ni même modifier une valeur présente, j'ai une erreur de timeout (même avec le chunk à 1) mais ça c'est peut-être un autre souci.

fairecasoimeme, 2025-02-05:
Ok, lorsque j'essaie de rajouter le totalApparentPower avec Z2M, j'ai l'erreur suivante sur le ZLinky : E_ZCL_ERR_ATTRIBUTE_TYPE_UNSUPPORTED
Du coup, je pense que Z2M n'utilise pas le bon type dans la requête de config report

fairecasoimeme, 2025-02-05:
Ok c'est bien ça. 
Le type dans la norme est un uint32 soit un type = 0x23

Et dans mon cas, le type est un uint16 soit un type = 0x21

Donc il faut que je fasse une MAJ du firmware ( :'( ) donc là de suite je vais éviter... mais je fais la modification pour la prochaine

ALaDoffe, 2025-02-05:
Même problème que @NeWoKiKi.
Abonnement EDF Tempo 3 Phases. Z2M 2.1.0

Après upgrade de mon Zlinky V1 en firmware 4000-0015. 
Suppression et Re-appairage sinon aucunes valeurs.

Remontée de SINSTS toutes les 10min. La modification du poll interval à 60 a résolu le pb.
Impossible d'ajouter, supprimer et modifier le reporting dans Z2M.
Quand je configure le tarif en STANDARD - TEMPO j'ai la même erreur que https://github.com/Koenkk/zigbee-herdsman-converters/issues/8710
Je reste donc en Tarif AUTO.