fairecasoimeme/Zlinky_TIC #294: Lixee V2

fullpower67, 2024-11-01:
Bonjour,

Je viens de recevoir le LIXEE en v2. Installé avec Z2M, màj OTA en V14 (très long mais semble ok) . 
►Compteur Linky en mode historique mono phasé, pas de production.
Qualité de la liaison + que correcte mais zéro remontée d'info. Quoi que je fasse, réinitialisation plusieurs fois faite (appui 10s), ré appairage etc. rien de rien ...tout reste à zéro ...
[image]

AlexandreKw, 2024-11-01:
Bonjour,
Mêmes symptômes ici. 
Nouveau Zlkinky_tic V2.
Installé sur Linky Sagemcom Triphasé - Mode Historique - Tarif Tempo.
Fournisseur Electricité de Strasbourg Réseaux
Zigbee par Zigbee2mqtt

La communication se fait, mais on dirait qu'il n'arrive pas à lire la TIC. 
J'ai beau changer les paramètres ou réappairer ça ne change pas.
Il me dit que je suis en tarif Base et me sort quelques lectures farfelues.

`{
    "apparent_power": 940,
    "current_summ_delivered": 2073,
    "current_tarif": "BASE",
    "linkquality": 81,
    "rms_current": 0,
    "rms_current_ph_b": 2,
    "rms_current_ph_c": 1,
    "warn_d_i_r1": 0,
    "warn_d_i_r2": 0,
    "warn_d_i_r3": 0,
    "MOTDETAT": null,
    "active_power_max": null,
    "active_register_tier_delivered": null,
    "available_power": null,
    "current_tier1_summ_delivered": null,
    "current_tier2_summ_delivered": null,
    "current_tier3_summ_delivered": null,
    "current_tier4_summ_delivered": null,
    "current_tier5_summ_delivered": null,
    "current_tier6_summ_delivered": null,
    "meter_serial_number": null,
    "presence_potential": null,
    "rms_current_max": null,
    "rms_current_max_ph_b": null,
    "rms_current_max_ph_c": null,
    "schedule_h_p_h_c": null,
    "start_notice_e_j_p": null,
    "tomorrow_color": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null,
    "warn_d_p_s": null
}`

Après certaiens tentatives il remone à HA que je suis en V1 et que la V14 est disponible... alors que la version afficher dans Zigbee2MQTT est 4000-0014.

tdaime, 2024-11-02:
Bonjour, je viens moi aussi de recevoir le Lixee en V2,  après apparairage il communique bien avec Z2M mais il est vu "Unsupported".  

[image]

fairecasoimeme, 2024-11-04:
> Bonjour, Mêmes symptômes ici. Nouveau Zlkinky_tic V2. Installé sur Linky Sagemcom Triphasé - Mode Historique - Tarif Tempo. Fournisseur Electricité de Strasbourg Réseaux Zigbee par Zigbee2mqtt - 1.40.2
> 
> La communication se fait, mais on dirait qu'il n'arrive pas à lire la TIC. J'ai beau changer les paramètres ou réappairer ça ne change pas. Il me dit que je suis en tarif Base et me sort quelques lectures farfelues.
> 
> { "apparent_power": 940, "current_summ_delivered": 2073, "current_tarif": "BASE", "linkquality": 81, "rms_current": 0, "rms_current_ph_b": 2, "rms_current_ph_c": 1, "warn_d_i_r1": 0, "warn_d_i_r2": 0, "warn_d_i_r3": 0, "MOTDETAT": null, "active_power_max": null, "active_register_tier_delivered": null, "available_power": null, "current_tier1_summ_delivered": null, "current_tier2_summ_delivered": null, "current_tier3_summ_delivered": null, "current_tier4_summ_delivered": null, "current_tier5_summ_delivered": null, "current_tier6_summ_delivered": null, "meter_serial_number": null, "presence_potential": null, "rms_current_max": null, "rms_current_max_ph_b": null, "rms_current_max_ph_c": null, "schedule_h_p_h_c": null, "start_notice_e_j_p": null, "tomorrow_color": null, "update": { "installed_version": -1, "latest_version": -1, "state": null }, "update_available": null, "warn_d_p_s": null }
> 
> Après certaines tentatives il remone à HA que je suis en V1 et que la V14 est disponible... alors que la version affichée dans Zigbee2MQTT est 4000-0014.

Bonjour,

Non vous n'avez pas les mêmes symptomes, vous recevez des données. Pour avoir plus d'informations, vous devez passer votre Linky en mode standard. Vous avez un abonnement TEMPO mais probablement pas chez EDF. Votre mode historique ne sait donc pas gérer votre abonnement et les données. https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#1-jai-chang%C3%A9-dabonnement-et-je-ne-retrouve-plus-mes-index-dans-mon-dashboard-domotique  

Fred

fairecasoimeme, 2024-11-04:
> Bonjour,
>
> Je viens de recevoir le LIXEE en v2. Installé avec Z2M, màj OTA en V14 (très long mais semble ok) . ►Compteur Linky en mode historique mono phasé, pas de production. Qualité de la liaison [...]

Si vous n'avez aucune données et que votre led clignote toutes les secondes, il y a peut-être un problème avec votre ZLinky.
je vous suggère d'envoyer un mail à contact[at]lixee.fr (si ce n'est pas déjà fait)

Fred

AlexandreKw, 2024-11-04:
> Vous avez un abonnement TEMPO mais probablement pas chez EDF.

Bonjour,
Ok merci, on va voir si c'est ça, je viens de faire la demande chez ESR. Je confirmerai quand le changement sera fait.
J'ai été un peu étonné cela dit que ça remonte "BASE", et d'autres valeurs incohérentes.

Si ça peux aider quelqu'un, sur le site de Electricité de Strasbourg Réseaux, cette demande correspond à la prestation 186 (code différend de celui de ENEDIS), la demande se fait par mail, j'ai fait ça par le formulaire de contact dispo sur leur site.

fullpower67, 2024-11-06:
> Si vous n'avez aucune données et que votre led clignote toutes les secondes, il y a peut-être un problème avec votre ZLinky. je vous suggère d'envoyer un mail à contact[at]lixee.fr (si ce n'est pas [...]

Bonjour Fred,

Oui c'est fait.
Je n'exclue pas non plus comme pour Alexandre d'être un "cas particulier" car en Alsace nous n'avons pas EDF comme fournisseur mais ESR... dans tous les cas, comme précisé dans mon email, je vous renverrai le Lixee apres le 13/11.
Eric.

AlexandreKw, 2024-11-07:
>Vous avez un abonnement TEMPO mais probablement pas chez EDF. 
>Votre mode historique ne sait donc pas gérer votre abonnement et les données. 

Bonjour  @fairecasoimeme 

La modification du Mode TIC a été effectuée le lendemain de ma demande.
Mais aucun changement. Il me remonte toujours que je suis en tarif BASE et 2 ou trois valeurs qui semblent fausses.
Le ZLinky clignotte toutes les 3 sec donc ça semble ok de ce côté là.

Donc j'imagine que,
soit y'a quelque chose lié ay ZLinky V2, (mais a priori il n'y a eu qu'une évolution matérielle)
soit c'est le firmware V14 (j'ai vu quelques soucis liés à cette version...)
soit c'est lié au fait que c'est Electricité de Strasbourg Réseaux. (ça m'éttonerait, leur doc linky et téléinfo est celle de Enedis...)
soit je suis tombé sur une mauvaise série comme @fullpower67 (mais la conincidence de fournisseur est curieuse tout de même)

Je vais donc contacter par le mail.

J'ia ce genre de warning dans Z2M:
`warning 2024-11-07 08:38:59zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006203d55/1 liXeePrivate.read(["currentTarif","motDEtat","warnDPS"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Timeout - 42746 - 1 - 226 - 65382 - 1 after 10000ms)`

Voici l'état Z2M. On voit bien la version qui n'est pas remontée, et la tarif en BASE, alors qu'on attendrait un Tempo et Tripahsé.
{
    "apparent_power": 4038,
    "current_summ_delivered": 2372,
    "current_tarif": "BASE",
    "linkquality": 61,
    "rms_current": 18,
    "status_register": "003AC401",
    "status_register_breakout": {
        "cache_borne_dist": "ferme",
        "comm_euridis": "activee avec securite",
        "contact_sec": "ouvert",
        "depassement_ref_pow": 0,
        "etat_cpl": "nouveau_verrouille",
        "horloge": "correcte",
        "organe_coupure": "ferme",
        "pointe_mobile": "AUCUN",
        "preavis_pointe_mobile": "AUCUN",
        "producteur": 0,
        "sens_energie_active": "positive",
        "surtension_phase": 0,
        "sync_cpl": "non_synchronise",
        "tarif_dist": "index_4",
        "tarif_four": "index_2",
        "tempo_demain": "UNDEF",
        "tempo_jour": "UNDEF",
        "type_tic": "standard"
    },
    "update": {
        "installed_version": 1,
        "latest_version": 14,
        "state": "available"
    },
    "warn_d_p_s": 0,
    "MOTDETAT": null,
    "active_energy_out_d01": null,
    "active_energy_out_d02": null,
    "active_power": null,
    "active_power_max": null,
    "active_power_max_ph_b": null,
    "active_power_max_ph_c": null,
    "active_power_ph_b": null,
    "active_register_tier_delivered": null,
    "apparent_power_ph_b": null,
    "apparent_power_ph_c": null,
    "available_power": null,
    "average_rms_voltage_meas_period": null,
    "average_rms_voltage_meas_period_ph_c": null,
    "average_rms_voltage_measure_period_ph_b": null,
    "current_date": null,
    "current_index_tarif": null,
    "current_price": null,
    "current_tier1_summ_delivered": null,
    "current_tier2_summ_delivered": null,
    "current_tier3_summ_delivered": null,
    "current_tier4_summ_delivered": null,
    "current_tier5_summ_delivered": null,
    "current_tier6_summ_delivered": null,
    "drawn_v_a_max_n1": null,
    "drawn_v_a_max_n1_p2": null,
    "drawn_v_a_max_n1_p3": null,
    "message1": null,
    "message2": null,
    "meter_serial_number": null,
    "power_threshold": null,
    "relais": null,
    "rms_current_max": null,
    "rms_current_ph_b": null,
    "rms_current_ph_c": null,
    "rms_voltage": null,
    "rms_voltage_ph_b": null,
    "rms_voltage_ph_c": null,
    "schedule_h_p_h_c": null,
    "site_id": null,
    "software_revision": null,
    "tomorrow_color": null,
    "total_apparent_power": null,
    "update_available": null
}

Meric pour le support.
Alex

fairecasoimeme, 2024-11-07:
Vous devez changer les paramètres spécifiques de votre module. Peut-être réappairé et redémarrer Z2M
Dans les données que vous partagez, il y a bien le status de registre mais les autres étiquettes ne correspondent pas au mode

sigalou, 2024-11-10:
Bonjour,
Je reprends cette discussion, je suis en V14.
Je n'ai pas eu de souci d'intégration du module en Zigbee
Tout fonctionne bien avec le Zlinky, juste une question de curiosité sur le basculement HC/HP pour ceux qui sont sur ce mode.
Mon Linky est en STANDARD
Voici le résultat :
[image]

Je constate donc :
- STGE et la conso correspondent et sont sur la bonne heure 02:06:07 pour les deux.
- 02:14:07 pour NTARF et pour RELAIS.

vous avez constaté un décalage ?
quelqu'un a une explication ?
Merci

julll190, 2024-11-13:
J'ai également un décalage entre l'info "PTEC" (Période tarifaire en cours, qui indique si HP ou HC) et le basculement réel du contacteur dans mon tableau : entre 2 et 8 minutes la semaine passée (mon contacteur "claque" à 22h tout pile tous les soirs, alors que HA m'indique un changement entre 22h02 et 22h08 selon les soirs).
J'ai un Linky en mode historique, abonnement Tempo avec un Zlinky V2 4000-0014

fairecasoimeme, 2024-11-19:
Bonjour,

Le décalage du PTEC est dû au fait que cette valeur ne peut être remontée que sur polling. Du coup, le PTEC sera mis à jour qu'au moment où le plugin fera la demande.
la donnée STGE remonte sur chaque changement de sa valeur. C'est pour cela qu'elle est plus réactive que PTEC.

Je suis en train de faire des modifications sur le firmware pour que PTEC remonte sur chaque changement de valeur.

dalingux, 2024-12-09:
Bonjour,
J'ai un V2, mon compteur en standard, EDF en base HP/HC.
Le V2 est bien vu dans Z2M (dernière màj) et j'ai fait la màj OTA en v14.
Malgré tout cela mon V2 clignote désespérément toute les secondes et les données ne remonte pas. J'ai (je pense) tout essayé, paramètres en auto, en rentrant les valeurs ( standard, mono...), supprimer, reappairer, laisser débrancher toute la nuit...
Je suis preneur de toutes suggestions car je sèche, dans l'onglet état il est toujours en "type_tic": "historique" alors que dans la configuration.yaml il est en linky_mode: standard.

fairecasoimeme, 2024-12-10:
> Bonjour, J'ai un V2, mon compteur en standard, EDF en base HP/HC. Le V2 est bien vu dans Z2M (dernière màj) et j'ai fait la màj OTA en v14. Malgré tout cela mon V2 clignote désespérément toute les [...]

Merci d'envoyer un mail à contact[at]lixee.fr

mixman68, 2024-12-15:
Strabsourgeois aussi sur le tarif réglementé, le Lixee est incapable de remonter la moindre info, j'ai l'impression qu'ESR n'utilise pas la meme façon de dialoguer que les autres en mode historique, c'est à dire que j'ai un index avec la somme de tous les indexs qui est rangé dans BASE, puissance apparente, intensités, les infos du contrats en kVA et c'est tout

Des retours sur un passage en mode standard ?

fairecasoimeme, 2024-12-16:
oui votre opérateur n'utilise pas les paramètres des abonnements historiques même si le nom de l'abonnement est le même.
Dans ce cas, c'est le Linky qui par défaut envoie BASE et ne remonte que l'index total.

Pour régler le problème, il faut demander à passer en mode standard.

mixman68, 2024-12-17:
> Pour régler le problème, il faut demander à passer en mode standard.

c'est bon c'est passé en mode standard
Le module semble quand même avoir du mal à tout remonter, dois je passer en limited ? On est pas nombreux sur le zigbee, une 10aine d'appareils max
Le linky est un sagemcom

mixman68, 2024-12-18:
Finalement je suis passé en limited et remontée des valeurs OK

Le module semblait reboot avant l'envoi des 4 chunk et reprend du premier, quand j'ai coupé le reste du réseau zigbee et qu'il restait que lui, les valeurs sont remontées m

le problème étant qu'avec le nouveau mode de led, on ne voit pas quand le module reboot de lui même du coup

fairecasoimeme, 2024-12-18:
Il est en effet préférable de mettre le chunk à 2 pour éviter d'avoir des requètes trop importantes. Selon les coordinateurs le chunk est plus ou moins accepté.

mixman68, 2024-12-18:
Je vais essayer 2

avec 4 (la valeur par défaut) j'avais des soucis

dalingux, 2024-12-26:
Bonjour,
suite à l'échange standard du module ça fonctionne, les valeurs remontent.
Reste à faire les jolis graphes :)
Merci.

KikiLaPine77, 2025-04-04:
Bonjour à tous.

Nouveau sur HA et nouveau sur cette plateforme. Je m'excuse si le topic a déjà été traité mais je ne trouve pas de réponses à mes erreurs.

HA en dernière Version, j'ai acheter la dernière Zlinky V2 avec antenne. Première installation avec Z2M en mode historique OK. J'ai demandé le passage en mode standard et depuis c'est l'hécatombe... Je pense avoir compris qu'il y avait trop d'informations remontées et cela fait plante la connexion entre Z2M et MQTT (mosquito broker) Le Zlinky est en version 16.

J'ai cru comprendre qu'il fallait flasher en Limited, est-ce la solution?

Merci d'avance pour votre aide.