fairecasoimeme/Zlinky_TIC #296: Pas de remontées des EASF01 & EASF02

Stef33fr, 2024-11-11:
Bonjour,

Depuis une semaine je n'ai plus de remontées de 
[image]
ni de 
[image]

Alors que toutes les autres données sont bien dynamique comme le courant, la tension, la charge active et la puissance apparente.

julll190, 2024-11-13:
Dans Z2M, onglet "paramètres (spécifiques)", vérifie que ton paramétrage est toujours bon (Linky mode, Energy phase, ...). Le mien a sauté lundi dernier, du coup j'avais certaines valeurs qui ne remontaient plus malgré l'avoir débranché puis rebranché.

Stef33fr, 2024-11-14:
> Dans Z2M, onglet "paramètres (spécifiques)", vérifie que ton paramétrage est toujours bon (Linky mode, Energy phase, ...). Le mien a sauté lundi dernier, du coup j'avais certaines valeurs qui ne remontaient plus malgré l'avoir débranché puis rebranché.

Bonsoir, j'ai déja tout vérifié de ce coté et tout est OK
En revanche, et ce sans rien faire, j'ai récupéré EASF01 & EASF02

closed 2024-11-15 by fairecasoimeme