fairecasoimeme/Zlinky_TIC #298: zLinky V2 : PTEC affiche mauvaise valeur et est figé

julll190, 2024-11-15:
Bonjour,

J'ai un abonnement Tempo triphasé avec le Linky en mode historique.

Depuis mon achat du zLinky V2 il y a quelques semaines, les jours Tempo étaient toujours bleus et le PTEC changeait bien en HPJB et HCJB au bon moment (même si le zLinky avait entre 1 et 8 minutes de retard sur le changement réell du relais selon les jours).

Le 13/11/2024 (jour bleu) à 22h05 (heures creuses débutaient à 22h00), la valeur de PTEC est bien passée à HCJB (normal).
Le 14/11/2024 à 6h00 le premier jour Tempo blanc est arrivé, mais le PTEC est resté sur HCJB au lieu de passer à HPJW...
A 22h au passage en heures creuses (jour toujours blanc), puis re-passage en heures pleines ce matin (encore jour blanc) le PTEC est resté figé sur HCJB.

Pourtant le zLinky envoie bien les autres données (au minimum les IINST1,2,3 ou mes index de conso qui changent selon mes appareils allumés).

Pourquoi ce comportement, ai-je mal configuré quelquechose ?

Voici une capture de l'hitorique PTEC (1ère ligne), et de l'historique donné par l'intégration RTE (3 lignes suivantes) :
[image]

Merci !

Ma config :
- box HA Green
- clé Connect ZBT-1
- zLinky appairé via zigbee2mqtt

julll190, 2024-11-15:
Petit ajout :
J'ai redémarré z2m pour une autre raison, et je me suis aperçu que 10 minutes après mon PTEC s'était mis à jour avec la bonne valeur....
Je me demande pourquoi seulement cette valeur était figée et pas les autres, à surveiller...

Yamakaky, 2025-01-04:
As-tu regardé sur le linky si la valeur remonté par le zlinky était la bonne ? Je suis passé en tempo hier, et aujourd'hui le linky (et donc zlinky aussi) affiche un tarif bleu alors qu'on est en blanc. Je vais voir si ça se résout tout seul.

julll190, 2025-01-05:
Non je n'avais pas regardé, mon Linky est à la rue donc pas toujours pratique de vérifier rapidement en cas de problème.
Depuis ce redémarrage de z2m le souci n'est jamais réapparu.

jpouyaud-perso, 2025-02-03:
hello
j'ai un comportement similaire

-monophasé
-historique (tempo => BBR)
-opérateur local (non EDF)

ca marchait très bien et ce matin le PTEC c'est figé sur HCJB

[image]