fairecasoimeme/Zlinky_TIC #299: Only updates when peak electricity is increasing.

traylenator, 2024-11-17:
Once my zlinky_tic is added to the energy dashboard of home assistant I seem to only get regular measurements when
the peak electricity is increasing.

So I see all my off peak electricity in one lump once the pea electricity starts increasing.

Linky Peak is sensor.linky_hchp
Linky OffPeak is sensor.linky_hchc

[image]

I'd like to see the updates over night at hourly resolution.

traylenator, 2024-11-17:
Here are the off peak meansurements:

[image]

Off peak starts at 02:00 but does not appear till the massive step at 06:00.

steenlarsen, 2024-11-20:
Does this happen always? I had this problem sporadically a few releases back but now it never happens. Having said that I also had my Linky replaced because it was faulty and at some time stopped counting electricity completely.

traylenator, 2024-12-07:
Indeed for whatever reason this situation has fixed itself now.

closed 2024-12-07 by traylenator