fairecasoimeme/Zlinky_TIC #300: Intégration LIxee V2 sur HA

Jrbebel, 2024-11-26:
Bonjour, 
J'aurai besoin d'aide pour connecter mon Lixee V2.
Aussitôt reçu,je l'ai connecté sur mon linky, Je n'ai reçu aucune valeur du module.
J'ai demandé à mon fournisseur de passer en mode standard en même temps.
Actuellement je suis en TIC en mode standard et en firmware 14 mais j'ai toujours des valeurs null qui me sont retourné.
Pouvez vous m'aider sur mon problème 

Voici mes l'etat : 

`{
    "active_register_tier_delivered": "",
    "apparent_power": 0,
    "current_summ_delivered": 0,
    "current_tarif": "BASE",
    "current_tier1_summ_delivered": 0,
    "current_tier2_summ_delivered": 0,
    "current_tier3_summ_delivered": 0,
    "current_tier4_summ_delivered": 0,
    "current_tier5_summ_delivered": 0,
    "current_tier6_summ_delivered": 0,
    "linkquality": 98,
    "meter_serial_number": "",
    "mot_d_etat": "000000",
    "rms_current": 0,
    "rms_current_max": 0,
    "schedule_h_p_h_c": 0,
    "start_notice_e_j_p": 0,
    "tomorrow_color": "",
    "update": {
        "installed_version": 14,
        "latest_version": 14,
        "state": "idle"
    },
    "warn_d_p_s": 0,
    "MOTDETAT": null,
    "available_power": null,
    "device": {
        "applicationVersion": 14,
        "dateCode": "20240422",
        "friendlyName": "0x00158d0006203af9",
        "hardwareVersion": 1,
        "ieeeAddr": "0x00158d0006203af9",
        "manufacturerID": 4151,
        "manufacturerName": "LiXee",
        "model": "ZLinky_TIC",
        "networkAddress": 52809,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "4000-0014",
        "stackVersion": 2,
        "type": "Router",
        "zclVersion": 3
    },
    "update_available": null,
    "active_energy_out_d01": null,
    "active_energy_out_d02": null,
    "active_power": null,
    "active_power_max": null,
    "active_power_ph_b": null,
    "average_rms_voltage_meas_period": null,
    "current_date": null,
    "current_index_tarif": null,
    "current_price": null,
    "drawn_v_a_max_n1": null,
    "message1": null,
    "message2": null,
    "power_threshold": null,
    "relais": null,
    "rms_voltage": null,
    "site_id": null,
    "software_revision": null,
    "status_register": null
}`

[image]

fairecasoimeme, 2024-11-27:
La led clignote-t-elle toutes les secondes ou toutes les 3 secondes ?

Jrbebel, 2024-11-27:
Elle clignote toutes les secondes

fairecasoimeme, 2024-11-27:
ok dans ce cas, il faut contacter le SAV sur contact[at]lixee.fr

closed 2024-11-27 by fairecasoimeme