fairecasoimeme/Zlinky_TIC #302: Possibilité de régler le reporting max sous les 60 secondes

KipK, 2024-11-30:
Salut @fairecasoimeme 

J'aimerais pouvoir régler le reporting max de Sinst sous les 60 secondes qui est la limite  logicielle pour l'instant.

Depuis quelques mois j'ai mon sinst réglé en reporting min sur 10s avec un changement minimum sur 1.
Cela fonctionne bien sur mon linky, j'ai bien sinst toutes les 10s.

Aujourd'hui je voudrais changer un truc pour avoir plus de reactivité sur les gros changements, et je me rend compte que la valeur reporting max est bloquée à 60 sec minium.

Je voudrais pouvoir regler le reporting change sur 1, le min reporting sur 1, et le max sur 20s, avec une valeur de changement minimum de 500W pour que seules les changements de puissance > 500VA soit mis à jours immediatement et le reste à 20.
Au final ca ferait moins de messages à envoyer, avec le confort d'avoir bien les changements importants en quasi temps réél. 
J'ai testé déjà avec la limite actuelle, donc le max sur 61sec, le min sur 1sec, et changement min sur 1 et ca fonctionne impec.

Possible d'avoir ça pour la prochaine maj ? 

Merci à toi

fairecasoimeme, 2024-12-02:
Bonjour,

Oui, je peux créer une version spécifique pour commencer avec cette modification et que tu puisses la tester.
De mémoire tu utilises une version limited ?

KipK, 2024-12-02:
Top merci. 
Je suis en version standard maintenant.

fairecasoimeme, 2024-12-03:
ok je compile une version asap

fairecasoimeme, 2024-12-04:
J'ai pas pu tester mais normalement c'est ok

[ZLinky_router_v14_MAX_Report.zip](https://github.com/user-attachments/files/18009852/ZLinky_router_v14_MAX_Report.zip)

KipK, 2024-12-04:
Merci d'avoir pris le temps pour ça.

Apperement ca ne passe pas, si je mets une valeur inférieure à 61 pour "Max rep Interval" sur SINST, il renvoie toujours la même erreur:

`z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ZCL command 0x00158d000638e0dc/1 haElectricalMeasurement.configReport([{"attribute":"apparentPower","minimumReportInterval":1,"maximumReportInterval":20,"reportableChange":500}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'`

fairecasoimeme, 2024-12-06:
erf OK
Je vais faire plus de tests ASAP alors

fairecasoimeme, 2024-12-07:
oui c'est exactement ce que j'ai fait en modification mais apparemment ça ne suffit pas. Il faut que j'investigue plus

KipK, 2024-12-07:
> oui c'est exactement ce que j'ai fait en modification mais apparemment ça ne suffit pas. Il faut que j'investigue plus

Et tu avais définis aussi

`ZCL_ATTRIBUTE_REPORTING_SERVER_SUPPORTED`
?

fairecasoimeme, 2024-12-07:
Oui, c'est plus un problème (je pense) de timer. je cherche actuellement, mais je pense qu'il doit y avoir un timer à 60 sec quelque part qui bloque. je cherche par rapport à l'erreur retournée : E_ZCL_ERR_INVALID_VALUE

fairecasoimeme, 2024-12-07:
Normalement c'est ok. 
[ZLinky_router_v14_MAX_Report2.zip](https://github.com/user-attachments/files/18048016/ZLinky_router_v14_MAX_Report2.zip)

KipK, 2024-12-07:
C'est tout bon 👍 Merci

fairecasoimeme, 2024-12-13:
C'était juste la compilation qui ne prenait pas en charge mon fichier .h :) 
As-tu un retour sur le fonctionnement de ton nouveau reporting ?

KipK, 2024-12-13:
oui ca fonctionne impec. J'ai bien la remontée SINST en instantané pour ce qui est superieur à 500VA et un refresh toutes les 20s dans tout les cas. 
Ca change pas mal la donne, je n,'arrive plus à faire sauter le compteur avec la borne de recharge maintenant :)

fairecasoimeme, 2024-12-13:
👍

closed 2024-12-13 by fairecasoimeme