fairecasoimeme/Zlinky_TIC #307: Uniquement "heures creuses" remontées

madd0, 2024-12-13:
Bonjour,

Depuis plusieurs mois mon module (Lixee ZLinky_TIC v1) remonte bien deux indexes: heures pleines et heures creuses. Puis, soudainement, le 30 novembre, il s'est mis à ne remonter qu'une valeur (qui correspondait avant aux heures creuses).

Je suis sous Home Assistant avec ZHA, et mon compteur est en mode "historique".

Quelqu'un a une idée de la cause probable ? Est-ce que demander un passage en mode "standard" risque d'améliorer ou, au contraire, empirer la situation ?

En investiguant le problème ces derniers jours, j'ai découvert au passage que j'avais la version 1 du firmware. J'ai fait la mise à jour OTA vers la v14, mais ça n'a rien changé.

Toute idée est bienvenue.
Je vous remercie d'avance

fairecasoimeme, 2024-12-13:
Bonjour,

N'y a-t-il pas eu un changement d'opérateur ? 
A ma connaissance, rien d'autres ne peut expliquer ce phénomène.

Fred

madd0, 2024-12-15:
Vu par email. Je reporte ici au cas où d'autres personnes auraient le même problème.

Il n'y a pas eu de changement d'opérateur, mais il y a peut-être eu un changement chez l'opérateur. J'ai demandé de passer en mode "standard" pour voir si ça solutionne le problème.

Je mettrai à jour ce fil quand le changement sera fait.

Merci

madd0, 2024-12-23:
L'opérateur a passé le Linky en mode standard et, après avoir débranché le ZLinky pendant environ 1h, puis rebranché, j'ai de nouveau les données !

Merci pour votre aide

closed 2024-12-23 by madd0