fairecasoimeme/Zlinky_TIC #308: Lixee ZLinky v1 model incorrect

yomgui1, 2024-12-18:
J'ai un LiXee ZLinky v1. Mais le model n'est pas "ZLinky_TIC" mais "ZLinky" tout cours.
Je suis passé il y a qq mois à la V14 du firmware, mais je viens seulement de m’apercevoir de cela en tentant de mettre le quirks v2 pour ZHA. J'ai changé dans son script le nom du model pour que cela fonctionne.

C'est normal d'avoir ce nom de model?

[image]

fairecasoimeme, 2024-12-18:
Bonjour,

Non ce n'est pas normal d'avoir ce nom de model. il faut avoir "ZLinky_TIC"
Avez-vous pris le bon firmware officiel ?

yomgui1, 2024-12-18:
Je suis toujours passé par l'OTA. C'est étrange, j'ai moi même du mal à comprendre.
Pour l'histoire, depuis l'achat j'étais sous Z2M, je suis passé cette année sous ZHA. Mais je n'ai pas fait plus attention au nom du modèle.
Ok merci de l'info. Je vais certainement reflasher la carte en mode USB pour être certain du firmware.

closed 2024-12-18 by yomgui1