fairecasoimeme/Zlinky_TIC #312: current_tarif = BASE au lieu de HPHC

bax137, 2025-01-02:
Bonjour,

Tout fonctionnait parfaitement depuis plusieurs mois sur mon zlinky en mode historique et tarif HPHC, mais je me suis rendu compte que depuis 1 mois current_tarif envoie BASE au lieu de HPHC.

J'ai essayé de forcer le Linky Mode et le Tarif à Historique/HPHC, de redémarrer le zlinky, de mettre à jour zigbee2mqtt, mais rien n'y fait, le tarif remonté reste en BASE.

fairecasoimeme, 2025-01-02:
bonjour,

êtes-vous chez l'opérateur octopus ?

bax137, 2025-01-02:
oui

bax137, 2025-01-02:
Je viens de me rentre compte aussi que tout remonte en HC depuis fin octobre, puis plus rien depuis début décembre. Là en forçant historique/HCHP ça remonte à nouveau en HC (au lieu de HP).

fairecasoimeme, 2025-01-02:
L'opérateur a changé tous ses abonnements et malgré le nom, ne font plus parti des abonnements historique EDF. Pour avoir les toutes les informations, il faut leur demander de passer votre Linky en mode standard.

bax137, 2025-01-02:
OK merci, je vais leur demander.

closed 2025-01-02 by bax137

bax137, 2025-01-02:
Pour info je les ai appelés il vont faire ça sous 24 heures !
Merci encore.