fairecasoimeme/Zlinky_TIC #313: Erreur paramétrage rapport table_full

ritonls, 2025-01-03:
Bonjour,

Je ne peux pas modifier les rapports de mon zlinkyvia l'onglet Rapport de Z2M.
J'ai supprimé mon zlinky et reset via le bouton, reboot Z2M & Homeassistant et enfin réassocié le Zlinky et Z2M

Quand je clique sur le bouton reconfigurer, j'obtiens cette erreur:
`error 2025-01-03 17:10:26z2m: Failed to configure 'ZLinky_TIC', attempt 5 (Error: ZCL command 0x00158d00062034a1/1 haElectricalMeasurement.configReport([{"attribute":"totalReactivePower","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed ({"target":41197,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":163},"zclSequence":178,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2131:55 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2070:16) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at async Promise.allSettled (index 11))`

Et lors de mes tentatives, j'obtenais également une erreur table_full

Est il possible de reinitialiser les rapports dans le zlinky ? ou de le remettre en configuration usine?

Zlinky en version 14limited
Home assistant avec Z2M:
Version de Zigbee2MQTT
[1.42.0]
Type de coordinateur
EmberZNet
Révision du coordinateur
7.4.5 [GA]
Adresse IEEE du Coordinateur
0x8c65a3fffeb09af4
Version de l'interface
0.7.6
Version Zigbee-herdsman-converters
20.58.0
Version Zigbee-herdsman
2.1.9

ritonls, 2025-01-03:
J'ai retrouvé le log d'erreur avec la TABLE_FULL
`error 2025-01-03 18:59:05z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0x00158d00062034a1/1 genIdentify from '0x8c65a3fffeb09af4/1' failed (Status 'TABLE_FULL')'`

fairecasoimeme, 2025-01-06:
Quand il y a ce type d'erreur, c'est qu'il y a un bind qui s'est mal passé ou avec un cluster de trop.
Le seul moyen est de reset et refaire l'appairage

closed 2025-01-06 by fairecasoimeme

ritonls, 2025-01-08:
> Quand il y a ce type d'erreur, c'est qu'il y a un bind qui s'est mal passé ou avec un cluster de trop.
> Le seul moyen est de reset et refaire l'appairage

J'ai déjà effectué ces manipulation:
- Suppression du device dans Z2M
- reset du zlinky-V2
- re appairage

Mais pas mieux :(, Existe T'il une possibilité de faire un reset usine du ZLinky ?

ritonls, 2025-01-11:
@fairecasoimeme 
Je viens a nouveau de faire l'essai et ca ne fonctionne toujours pas.
Est il possible de faire un reset usine SVP?

fairecasoimeme, 2025-01-13:
Pour faire un reset, il faut appuyer plus de 10 secondes sur le bouton. Le ZLinky redémarre avec la led qui clignote rapidement. les données de BIND sont alors remis à zéro.
Normalement, il n'y a pas de raisons d'avoir l'erreur de Bind. Cette erreur ne peut arriver que si un Bind a été mal enregistré.
Je n'ai pas de retour d'autres personnes avec cette erreur qui n'a pas été résolu avec le reset.

ritonls, 2025-01-14:
Ok et bien meme après plusieurs reset et reassociation, c'est pareil.
J'ai meme rebasculer le firmware de limited vers normal, puis suppression et reasso

fairecasoimeme, 2025-01-15:
vous avez systématiquement l'erreur de TABLE FULL a chaque remise à zéro ?