fairecasoimeme/Zlinky_TIC #316: erreurs d'acquisition avec un SAGEMCOM - TEMPO

bwallrich, 2025-01-21:
Bonjour,

Je tente de connecter mon ZLinky_TIC à mon Linky SAGEMCOM (avec abonnement TEMPO).
Si coté  zigbee tout semble bien se passer, le voyant lumineux persiste à clignoter 1x par seconde, signe qu'il rencontre des erreurs lors de l'acquisition des données du Linky.

Les messages recu sur coté zigbee (renvoyé sur ma passerelle MQTT) arrivent bien mais les valeurs sont à 0.

Je n'arrive pas à trouver ce qui ne colle pas.

`
MQTT publish: topic 'home/linky', payload {
    "active_enerfy_out_d01": 0,
    "active_power": 0,
    "active_power_max": 0,
    "active_power_ph_b": 0,
    "apparent_power": 0,
    "average_rms_voltage_meas_period": 0,
    "current_date": "",
    "current_index_tarif": 0,
    "current_price": "",
    "current_summ_delivered": 0,
    "current_tarif": "BASE",
    "current_tier1_summ_delivered": 0,
    "current_tier2_summ_delivered": 0,
    "drawn_v_a_max_n1": 0,
    "linkquality": 132,
    "message1": "",
    "message2": "",
    "meter_serial_number": "",
    "relais": 0,
    "rms_current": 0,
    "rms_voltage": 0,
    "site_id": "",
    "status_register": "",
    "update": {
        "installed_version": 1,
        "latest_version": 14,
        "state": "available"
    }
}
`

fairecasoimeme, 2025-01-22:
Bonjour,
Si le ZLinky clignote toutes les seconde et que vous êtes en mode standard, il est possible que votre ZLinky ait un problème matériel.
veuillez contacter par mail contact[at]lixee.fr

bwallrich, 2025-01-22:
merci ! je fais ca.

closed 2025-01-22 by bwallrich