fairecasoimeme/Zlinky_TIC #319: Le sensor LTARF indique "HP BLEU" alors que NTARF a la valeur 6 (HP ROUGE)

cperrier, 2025-02-03:
Bonjour,

Je viens de constater, pour la première fois, un comportement curieux de mon ZLinky. Depuis 2 jours, le sensor LTARF indique "HP BLEU" alors que le Linky indique "HP ROUGE" (dernier jour rouge 2024/2025 aujourd'hui!).

J'ai cru à un moment que le ZLinky était planté, mais tous les autres sensors sont correctement mis à jour. C'est la première fois que je constate ce comportement.

La version de firmware est 4001-0014 (version 14 en mode "limited").

Je n'ai pas encore totalement réinitialisé le ZLinky, j'ai juste essayé de le déconnecter quelques secondes et le reconnecter, mais cela n'a rien résolu.

Je suis bien sûr en tarif EDF TEMPO et le Linky est en mode Standard.

Au besoin je peux faire d'autre investigations si cela peut aider à trouver l'origine du problème. J'ai adapté mes automatisations (j'utilise HA) afin qu'elles se basent sur NTARF en plus de LTARF.

jpouyaud-perso, 2025-02-03:
J’ai le même problème 
Tempo également 

J’ai upgrade en v15 et depuis j’ai carrément plus rien

cperrier, 2025-02-03:
Plus rien dans quel sens ? Plus de problème ou bien plus de retour sur la période tarifaire ?

Pour ma part, je n'ai pas vu arriver de version 15 "limited" du firmware en OTA sur Z2M, d'ailleurs. Peut-être cette version est-elle en fait assez récente ?

Après le très gros travail de KipK (https://github.com/fairecasoimeme/Zlinky_TIC/issues/290) qui a fait en sorte qu'on pouisse avoir une méthode pour faire une MAJ de firmware "normal-->limited" en OTA dans Z2M, j'attendais de voir si les MAJ de firmware arrivaient "automatiquement" dans Z2M pour ceux d'entre nous qui sont passés en firmware "limited" pour éviter les "plantages" qui se produisent quand des capteurs Zigbee s'appairent avec le ZLinky. Mais, bon, c'est peut-être plus compliqué à faire pour @fairecasoimeme ou bien il n'a pas le temps non plus (ce que je comprendrais tout à fait).

fairecasoimeme, 2025-02-04:
Bonjour,

Oui après la mise à jour, il faut :

* Supprimer le ZLinky de Z2M
* Réappairer

Pour je ne sais quelle raison sur Z2M, après la mise à jour, le ZLinky ne fait plus l'acquisition de données du Linky

Je vais regarder pour la prochaine mise à jour pour corriger ce point

Concernant LTARF, c'est une donnée qui est en read only et qui ne peut pas remonter sans qu'on le poll. Du coup, il peut y avoir un décalage par rapport à la réalité.
Normalement le meilleur moyen de récupérer les couleurs du jour et du lendemain est d'utiliser le registre de statut STGE (en mode standard) qui lui remonte à Z2M sur chaque changement de valeur.

La valeur STGE est brute mais il faudra faire des modifs dans le plugin pour décomposer les valeurs

Dans la nouvelle version v15 du ZLinky, il y a aussi un nouvel attribut : "tariffPeriod" qui remontera sur chaque changement de valeur et plus facile à gérer côté plugin mais pour le moment, il y a un problème

cperrier, 2025-02-04:
Bonjour et un grand merci pour la réponse détaillée...

Actuellement mon contournement pour utiliser NTARF fonctionne bien (par exemple, cet index est 4 à l'instant même alors qu'on est en HP BLANC, ce qui est logique : 1=HC BLEU, 2=HP BLEU, 3=HC BLANC, etc.)

STGE me semble affectivement bien cryptique à comprendre (je vois actuellement "423ACC01"), je vais donc plutôt m'en tenir là pour l'instant.

Concernant la MAJ firmware, je vois justement que Z2M me propose une mise à jour en version 15. Par contre je me demande si ce sera alors une version 15 "limited" ou "normal" si je la fais (je suis en 14 "limited" actuellement grâce aux échanges de #290 ).

Dans l'issue 290, KipK mentionnait la possibilité que ces montées de version de firmware deviennent possibles en OTA mais je ne sais pas si cela a pu être fait pour la version 15 du firmware. AU pire, je peux le faire manuellement à nouveau avec la bidouille décrite dans #290 ce qui me permettrait au moins de voir apparaître ce nouvel attribut tariffPeriod (même s'il n'est pas encore utilisable à ce que je comprends).

cperrier, 2025-02-05:
Effectivement, j'ai fait la MAJ de Z2M récemment (je suppose qu'il doit y avoir moyen de retrouver quand et confirmer si c'est au même moment que le ZLinky a cessé de mettre à jour LTARF.

Merci pour l'info. En passant si quelqu'un connaît un moyen de retrouver le moment où la MAJ d'un module complémentaire (ou d'une itnégration, d'ailleurs) a été réalisée.

En regardant le journal de Z2M, je vois apparaître ceci très régulièrement : 
[2025-02-05 08:09:14] warning:  zhc:lixee: Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'tariffPeriod'

Ces messages ne sont dans le journal de Z2M que depuis le dernier redémarrage du module, qui semble avoir eu lieu le 2/2 à 18h52, ce qui est probablement le moment où j'ai du faire la mise à jour (Z2M recrée un nouveau répertoire dans "log" dès qu'on le redémarre).

Or, la dernière fois où LTARF a changé correctement, c'est le 2/2/2025 à 6h du matin, donc au changement de tarification de ce jour là (HC BLANC --> HP BLEU)

Il semble donc très probable que le problème vienne de la MAJ de Z2M, oui.

jessycler, 2025-02-07:
Bonne nouvelle !
Z2M a été mis à jour en version 2.1.1 à l'instant.
Grâce à cette mise à jour, mon zlinky affiche à nouveau LTARF 👍 (je suis toujours en V15 sur le zlinky).

cperrier, 2025-02-07:
Même chose pour moi. Je pense donc qu'on peut clôturer ce ticket.

Merci pour les réponses des uns et des autres et merci pour son travail, à @fairecasoimeme

closed 2025-02-07 by cperrier

AURELB87, 2025-02-07:
Bonsoir,

Pour ma part, je suis passé en V2.1.0 pour Z2M, pas de problème. 
J'ai eu des problèmes avec la màj OTA V15. 
Ce soir, je suis passé en V2.1.1, j'ai désappairé le Zlinku, refais l'appairage: aucune donnée ne remonte et pire, cela me fait planter Z2M (erreur '130').

Si quelqu'un a une idée, je suis preneur.