fairecasoimeme/Zlinky_TIC #320: Plus de remontée

95JOE210, 2025-02-03:
Bonjour,

Je n'ai plus de remonter de mon ZLinky_TIC v2 depuis la mise à jour vers la version V15. Y a t-il quelque chose a modifié après la maj?

[image]

fairecasoimeme, 2025-02-03:
En général sur Z2M après une mise à jour, il faut redémarrer Z2M

95JOE210, 2025-02-03:
Le redémarrage n'a pas suffit il a fallu que je supprime le ZLinky_TIC v2 et que je le réadopte. Le problème est résolu.
Merci

Mr-PoPoP, 2025-02-03:
Bonjour,

J'ai le même problème depuis la MàJ v15, mes compteurs sont vides (sauf certains), j'ai essayé aussi de réappairer (en utilisant le petit bouton de reset sur le  Zlinky) et aussi supprimer le Zlinky de la liste d'appareils, redémarrage Z2M, mais rien y fait, les compteurs sont toujours vides et j'ai des erreurs Z2M "Failed to read..."

Si quelqu'un a une idée, je suis preneur :)

`warning 2025-02-03 18:22:13zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 liXeePrivate.read(["currentTarif","message1"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)
warning 2025-02-03 18:22:13zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 haMeterIdentification.read(["softwareRevision"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '17738'.)
warning 2025-02-03 18:22:18zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 liXeePrivate.read(["message2","relais"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)
warning 2025-02-03 18:22:22zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 liXeePrivate.read(["statusRegister","drawnVAMaxN1"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)
warning 2025-02-03 18:22:27zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 haElectricalMeasurement.read(["activePower","activePowerPhB"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)
warning 2025-02-03 18:22:32zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 haElectricalMeasurement.read(["averageRmsVoltageMeasPeriod","rmsVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)
warning 2025-02-03 18:22:37zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006205710/1 haElectricalMeasurement.read(["activePowerMax"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (Delivery failed for '34330'.)`

[image]

[image]

Lesuperlolo, 2025-02-03:
Même problème ici.
Forcer la suppression sous Z2M et appairer à nouveau a été sans effet, le ZLinky ayant même le statut indisponible au bout de plusieurs tentatives.
Seule issue trouvée : faire un retour à une sauvegarde plus ancienne, où le Zlinky est reconnu et discute en mqtt et remonte les informations.
Mais chose étrange, il est affiché en firmware 14, et ne propose pas de mise à jour, alors que je suis certain d'avoir fait une mise à jour en v15 avec succès...

fairecasoimeme, 2025-02-04:
Bonjour,

Oui après la mise à jour, il faut :

* Supprimer le ZLinky de Z2M
* Réappairer

Pour je ne sais quelle raison sur Z2M, après la mise à jour, le ZLinky ne fait plus l'acquisition de données du Linky

Je vais regarder pour la prochaine mise à jour pour corriger ce point

Mr-PoPoP, 2025-02-04:
Pour ma part le problème est résolu. Il fallait juste un peu de patience à priori... les données sont remontées une demi-heure après la MaJ.

ghostsam45, 2025-02-04:
Même soucis pour moi ce matin, suite à la mise à jour j'ai du supprimer le Zlinky de Z2M et réappairer comme la souligner @fairecasoimeme.
Et ça prends bien 30mn environ à remonter tous les données.

NeWoKiKi, 2025-02-05:
Du coup, juste pour information, le "Measurement poll chunk" par défaut est à combien ?
Dans l'interface c'est indiqué "Default: 2", c'est toujours le cas ?
Si non, c'est recommandé de le mettre à combien ?

Je n'ai plus confiance aux champs vides maintenant donc je vais le remplir aussi xD

PierrickP, 2025-02-05:
Un peu pareil.

Il faut essayer plusieurs fois.
+ Débrancher
+ Supprimer
+ Rebrancher / Appairer
+ Forcer les paramètres (poll / précision)

Attendre 20/30mins.
Refaire si ca marche pas.
Redemarrer zigbee2mqtt