fairecasoimeme/Zlinky_TIC #323: Lixee HS depuis la mise à jour v15

Seven-94, 2025-02-06:
Bonjour,

Comme beaucoup j'ai fait la mise à jour v15 et depuis je n'ai plus de remontée dans Home Assistant et Z2M. j'ai donc lu quelques issues et j'ai tenté les manipulations decrites dans les commentaires or cela ne focntionne pas chez moi et je pense meme que la mise à jour à completement grillé mon Lixee car je n'ai plus la led qui clignote. 

En ouvrant mon boitier linky j'ai vu que le LIxee ne clignotait plus j'ai donc tenté de le supprimmer le Z2M puis de l'ajouter de nouveau mais la led ne s'allume plus ni ne clignote plus. Il m'est donc impossible de l'ajouter de nouveau dans Z2M. J'ai tente de laisse le Lixee hors du linky pendant 10 minutes puis de le reinstaller mais rien n'y fait il n'y a plus aucun activité sur le Lixee et meme en tentant un apparairage en restant appuyé 10s ou 30s ou 60s, Z2M ne detecte pas la tentative d'appairage.

J'espere me tromper mais à premiere vue le Lixee est mort à cause de la V15. J'aurais du regarder le repo avant de faire la mise à jour mais comme il n'y avait aucune information sur les modifications apportées par la V15 j'ai pensé à une mise à jour mineure. En gros NE FAITES PAS LA MAJ.

[image]

fairecasoimeme, 2025-02-06:
Bonjour,

les modifications sont dans chaque changelog de version

Il est peu probable que la mise à jour ait cassé votre ZLinky. Si vous avez un module USB TTL, il est probablement possible de le récupérer.
https://github.com/fairecasoimeme/Zlinky_TIC?tab=readme-ov-file#non-ota

Vous pouvez m'envoyer un mail directement sur contact[at]lixee.fr pour un échange ou reprogrammation si nécessaire.

Seven-94, 2025-02-06:
Merci pour votre réponse.

Si ce n'est pas la mise à jour alors c'est etrange car il est installe depuis mars 2024 sans aucun souci et le boitie linky est ferme depuis ce jour là. Il n'y a eu ni intervention ni modification de l'installation, uniquement la mise à jour.

Je vais vous contacter par mail.

LeBombe31, 2025-02-06:
J'ai exactement le même phénomène depuis la mise à jour... depuis que celle-ci est terminée mon Lixee ne s'allume plus

fairecasoimeme, 2025-02-07:
Oui le fait que la led ne s’allume plus est lié à la mise à jour v15.
Une fois branché sur le linky, il faut appuyer 15 sec sur le bouton puis relâcher. Normalement, au bout de 5 sec la led va se remettre à clignoter rapidement.

Seven-94, 2025-02-07:
Merci ! Je pense que ca rentre dans l'ordre, je l'ai laissée debranché toute la nuit puis j'ai rebranche ce matin en appuyant 15s (ni plus ni moins) sur le bouton et apres avior relache au bout de quelques secondes la led s'est mise à clignoter. Le probleme doit surement venir de fait que je restais appuyé trop longtemps sur le bouton, j'avais testé en restant appuyé 20s, 30s et meme 60s mais ca ne fonctionnait pas. 

Les données semblent remontées meme si pour l'instant BASE reste bloqué sur une valeur, je vais attendre quelques heures voir comment ca evolue.

Seven-94, 2025-02-07:
Ok tout fonctionne bien maintenant. Merci

closed 2025-02-07 by Seven-94