fairecasoimeme/Zlinky_TIC #324: tariffPeriod v16

Cadkey, 2025-02-07:
J'ai installé la v16, le ZLinky fonctionne, je n'ai pas d'erreur.
Je fonctionne sous deCONZ
sous l'application DeConz, je ne vois pas l'attribut 0x10 dans le cluster 0xFF66
Je ne vois pas de nouvel attribut créé dans le cluster 0xFF66 par rapport à la v14. C'est identique.
Je suis bien en v16: "swversion": "4000-0016",

[image]

fairecasoimeme, 2025-02-07:
Bonjour, oui c'est normal car Deconz doit mettre à jour de son côté le cluster
N'y a-t-il pas moyen avec les fichiers DDF de le faire ? je ne connais pas bien Deconz et si on peut faire du spécifique.

Cadkey, 2025-02-07:
Si on peut modifier le DDF et faire du spécifique, c'est ce que j'ai fait pour le ZLinky. Dans mon DDF je n'utilise pas tous les cluser/attribut, et dans l'image que je montre au-dessus, ce sont tous les cluster/attributs que detecte deCONZ.
Je vois que deCONZ ne detecte pas non plus MOTDETAT bizarrement dans la liste de tout ce que detecte l'app deCONZ.
Alors sur le module, je suis passé de v8 à V14 il y a un mois environ, puis V16. et je ne vois que ce qui est posterieur à V8 si je ne me trompe pas. C'est ce qui m'interpelle.

fairecasoimeme, 2025-02-07:
il faut alors rajouter dans le cluster 0xff66 l'attribut 0x0010 pour qu'il apparaisse dans Deconz
MOTDETAT est apparu en v13

Cadkey, 2025-02-07:
Merci. Je vais vérifier si je n'ai pas oublié de valider quelque part.

Cadkey, 2025-02-08:
Je suis en tests pour les remontées de PTEC et tariffPeriod sur decONZ.
Ce matin, les deux sont remontés en même temps 06:08.
Je n'ai aucun gain, je ne comprends pas.

fairecasoimeme, 2025-02-08:
PTEC est une valeur que l'on peut "poll" uniquement
TariffPeriod est une valeur qui remonte automatiquement dès changement de la valeur sur le Linky
sur deconz si le poll est à la minute vous aurez la meme heure:minute que tariffPeriod
Si vous avez les secondes, normallement tariffPeriod remontera avant

sur Z2M, s'il y a trop de trafic, il est possible que le Poll, même configuré à la minute prenne du retard. Avec tariffPeriod ce ne sera pas le cas.

PS : au cas où vous vous posiez la question, le changement de tarif sur le Linky ne se fait pas à heure fixe

Cadkey, 2025-02-08:
D'accord, je sais que le changement de tarif ne se fait pas à heure fixe. Je peux comparer avec le teleinfo usb, le claquement du contacteur chauffe-eau aussi. Ce matin, plusieurs minutes de retard, pas important sur PTEC, je vais me fier à tariffPeriod, mais mon essai n'a pas été concluant. Je n'exclue pas un problème decONZ, je regarde.
Je n'utilise pas Z2M, uniquement decONZ et son API avec des commandes curl.
Que veut dire 'poll' exactement? et quelles sont les valeurs qui remontent automatiquement? par exemple comment remonte ADPS?
Je compare avec un teleinfo usb, qui lit la trame à 1/2h fixe.

fairecasoimeme, 2025-02-08:
"poll" veut dire que pour obtenir la valeur PTEC, il faut qu'il ait une demande read attribute sur cette valeur pour qu'elle soit retournée.

alors que pour tariffPeriod, la valeur sera envoyée automatiquement par le ZLinky
ADPS remonte aussi automatiquement

Mais pour que ces valeurs remontent automatiquement, il faut qu'elles soient marquées RP dans le tableau https://github.com/fairecasoimeme/Zlinky_TIC?tab=readme-ov-file#mode-historique

ET

Faire un Bind sur le cluster + Config report. Je ne sais pas comment c'est paramétrable sur Deconz

Sur Z2M, c'est fait automatiquement à l'appairage

ScratMan, 2025-02-09:
Je n'ai pas le paramètre LTARF dans les remontées sur Z2M 2.1.1, je suis en FW 16, mode historique. Je vois que LTARF n'est pas listé dans leq paramètres du mode historique, mais mon PTEC ne se met plus à jour sur mon Zlinky depuis vendredi, il reste bloqué sur HPJW.
Mes indexes sont pourtant bien incrementés correctement en heures creuses et heures pleines, en jours blanc vendredi et samedi et bleu aujourd'hui (dimanche).

Comment débloquer le PTEC ? Merci.

Cadkey, 2025-02-09:
Il est possible d'utiliser le cluster pxff66 attribut 0x0010 tariffPeriod, créé depuis la V15 qui remonte les valeurs PTEC/LTARF.

Cadkey, 2025-02-11:
Bonjour,
Un petit retour pour confirmer que tariffPeriod remonte très vite la valeur de PTEC. C'est parfait.
Merci 👍

closed 2025-02-11 by Cadkey