fairecasoimeme/Zlinky_TIC #325: Version 0016 non disponible sur Zigbee2MQTT

Jezza34000, 2025-02-07:
Hello,
Mon zigbee2mqtt est actuellement en version 0015 et ne me propose pas la version 0016 ?
J'ai beau forcer la recherche de mise à jour dans `OTA` > `Vérifier les mises à jour` mais aucune mise à jour.
Est-ce normal ou y-a til quelquechose à faire en particulier ?
Merci

Seven-94, 2025-02-07:
Salut, 
Voici le commentaire de @fairecasoimeme sur une autre issue

>  Pour ceux qui veulent tester en manuel (non ota), voici un firmware qui corrige les problèmes :
> https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v16.0
> 
> Pour Z2M en OTA je le mettrai pour Lundi (si tout est ok)
> 
> Désolé pour les gênes occasionnées

Donc il faut attendre.

Jezza34000, 2025-02-07:
Merci j'attendrai Lundi alors 👍🏻

WarC0zes, 2025-02-08:
Bonjour,
c'est automatique la mise a jour du fichier OTA pour Z2M. C'est hebdomadaire les actualisations des fichiers. Donc pour lundi, ça sera disponible.

> LiXee fait partie du téléchargement automatique hebdomadaire. Pas besoin de PR manuel, tout est mis à jour automatiquement.

[image]

Jezza34000, 2025-02-08:
Merci pour l'information, je viens d'apprendre quelque-chose 🙂

closed 2025-02-08 by Jezza34000