fairecasoimeme/Zlinky_TIC #327: Tarif Zen Week-End Plus

t-r-i-s-t-a-n, 2025-02-10:
Bonjour,

Je viens d'installer mon ZLinky et je ne trouve pas mon abonnement.

Je suis en Zen Week-End Plus avec une option tarifaire Heures Creuses + WE + Vendredi, je suis en triphasé, je ne sais pas si cela a son importance.

À quelle tarif mon abonnement correspond et si aucun, est-il possible de l'ajouter ?

Merci

t-r-i-s-t-a-n, 2025-02-10:
Après la mise à jour en 16 puis un réappairage, j'ai plus de capteurs qui remontent, pour le moment ils sont vides, voyons d'ici quelques heures

[image]

Cadkey, 2025-02-11:
Votre option tarifaire ne s'affiche pas en mode Historique. Voir le readme OPTARIF en première page.

t-r-i-s-t-a-n, 2025-03-01:
Je suis enfin en mode TIC standard, j'ai supprimé le Zlinky de Z2M puis appairé de nouveau.
Et au bout de 5-10min environ, c'est le drame, mon Z2M fait completement planté mon Mosquitto.

Une idée du pourquoi ?

fairecasoimeme, 2025-11-27:
> Bonjour, des nouvelles pour le mode Zen Week-End Plus ?

La gestion des abonnements n'a pas de lien avec le ZLinky. Si vous avez des demandes particulières sur la gestion des abonnements c'est du côté du plugin de Z2M qu'il faut aller voir.

fairecasoimeme, 2025-11-27:
> Nan, impossible de le faire marcher, dès que je le branche, les informations envoyées à Home Assistant sont trop nombreuses et il plante.

De mémoire, vous avez la possibilité sous Z2M d'utiliser la whitelist pour limiter le nombre de messages.
Mais encore une fois ce n'est pas une issue du ZLinky