fairecasoimeme/Zlinky_TIC #328: Changement relevé EAST depuis la MAJ 15

Nivalam, 2025-02-11:
Bonjour,

J'utilise la fonction dérivée de HA pour approximer la puissance instantané du Linky comme décrit ici : https://rdr-it.com/domotique/home-assistant-afficher-sa-consommation-electrique-en-temps-reel-avec-zlinky/

Depuis la MAJ 15 (et 16) sur z2m, ce n'est plus fonctionnel. J'ai l'impression que l'EAST n'est plus remonté toute les minutes et du coup il la dérivée ne marche plus.

Est ce qu'il y a eu une modif qui peut expliquer ça ?

starox, 2025-02-11:
Je suis juste un utilisateur de zlinky. Je me demandais quelle est la différence entre votre solution et le champs SINSTI ?

SckyzO, 2025-03-04:
Tu as bien une data dans EAST ? 

Dans la release note il est écrit : 

⚠️ Attention sur Z2M, après chaque nouveau appairage, il faut relancer la reconfiguration image plusieurs fois pour que tous les rapports soient pris en compte

SckyzO, 2025-03-04:
> Je suis juste un utilisateur de zlinky. Je me demandais quelle est la différence entre votre solution et le champs SINSTI ?

- EAST (Énergie Active Soutirée Totale) : C'est l'index d'énergie totale consommée par le compteur en Wh (Watt-heures). Cette valeur est cumulative, c'est-à-dire qu'elle augmente continuellement avec le temps au fur et à mesure de la consommation.

- SINSTI (Puissance Apparente Instantanée) : C'est la puissance instantanée consommée à un moment donné, exprimée en VA (Volt-Ampères). Contrairement à EAST, cette valeur est mise à jour en temps réel et donne une indication immédiate de la puissance en cours de consommation.

Différence clé :
- EAST représente une énergie cumulée (en Wh).
- SINSTI représente une puissance instantanée (en VA).

SckyzO, 2025-03-04:
Ouais après je suis d'accord, pourquoi utiliser EAST plutôt que SINSTI ... Bonne question. 

Je viens de tester avec SINSTI j'arrive à avoir pareil que le lien que tu mets : 

[image]

Et tu peux même faire un truc mieux c'est avoir la valeur en Kw si tu veux en faisant un truc style, créer un nouveau sensor basé sur sinsti : 

Je pense que c'est même mieux, après je me trompe peut etre je débute aussi 😁

```
sensor:
  - platform: template
    sensors:
      consommation_kw:
        friendly_name: "Consommation instantanée (kW)"
        unit_of_measurement: "kW"
        value_template: "{{ (states('sensor.zlinky_sinsti') | int / 1000) | round(2) }}"
```

Gnomuz, 2025-03-26:
Transformer une valeur exprimée en VA en kW par une division par 1000 néglige le facteur de puissance. Si les charges sont très majoritairement résistives, l'approximation sera correcte, mais il y a généralement  un écart de 10-15% dans uns installation résidentielle.

max3163, 2025-03-29:
Mis à part le débat stérile sur la différence entre les Watt et les VA sur de l'alternatif, le problème d'origine a t il une solution. 
Je suis également passé au firmware 16 et j'ai une remontée de la valeur EAST ( current_summ_delivered ) qui est mis a jours de manière totalement sporadique… Avec la version 14, c'était très régulièrement ( environ 10 ou 15 seconde ) et je pouvais donc en déduire une conso en Watt assez réaliste.

Pouvez-vous me dire comment rétablir la remontée régulière des informations ? voir comment remettre le firmware 14 ?

SckyzO, 2025-03-30:
> Transformer une valeur exprimée en VA en kW par une division par 1000 néglige le facteur de puissance. Si les charges sont très majoritairement résistives, l'approximation sera correcte, mais il y a généralement  un écart de 10-15% dans uns installation résidentielle.
> 

SINSTI te donne la puissance apparente en VA (Volt-Ampères).

kW correspond à la puissance active, celle réellement consommée, et dépend du facteur de puissance (cos φ).

La relation réelle est :

```
P (kW) = S (kVA) × cos(φ)
```

Mais comme on n’a pas le cos φ avec ZLinky, on ne peut pas calculer précisément les kW.

Tu as raison, le facteur de puissance n’est pas pris en compte et ça donne une approximation de la puissance active. Mais pour du suivi visuel simple dans Home Assistant, ça me suffit. Si ZLinky expose un jour le cos φ, je corrigerai.

SckyzO, 2025-03-30:
> Mis à part le débat stérile sur la différence entre les Watt et les VA sur de l'alternatif, le problème d'origine a t il une solution.
> Je suis également passé au firmware 16 et j'ai une remontée de [...]

Alors debar stérile je ne suis pas d'accord. Non on est plus intelligent que ça 

😁

Moi vu les retours je ne suis pas passé en 16 encore je suis tres bien avec le 14
Je pense qu'en utilisant la fonction de mise a jour local offline de z2m tu dois pouvoir y arriver ?

https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Nico04, 2025-04-27:
Moi aussi j'utilise Zlinky principalement pour avoir une consommation (presque) instantanée, en utilisant EAST (plus précis). C'était régulier (environ toutes les minutes) sur la 14, et depuis ma mise à jour 16 c'est effectivement irrégulier, toutes les 5 minutes à tous les 15min, voir plus. J'ai déjà fait 3 reset physique du Zlinky, reboot HA plusieurs fois, refait l'appairage, débranche le Zlinky pendant 1 jour... 😭
Une idée ?

Jmi91, 2025-05-04:
Même problème pour moi, j'ai une mise à jour de l'EAST toutes les 3/4 minutes. J'ai essayé en downgradant en v14 mais pas mieux... Il y a t-il une solution ?
Merci

Jmi91, 2025-05-05:
Bon après avoir modifié comme cela ce paramètre:

[image]

Cela fonctionne toutes les minutes.

Nico04, 2025-05-07:
> Bon après avoir modifié comme cela ce paramètre:
> <img alt="Image" width="1032" src="https://private-user-images.githubusercontent.com/50298900/440228896-1d0cc234-f280-4533-95bb-10061e2ae451.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDY2MjU5MjYsIm5iZiI6MTc0NjYyNTYyNiwicGF0aCI6Ii81MDI5ODkwMC80NDAyMjg4OTYtMWQwY2MyMzQtZjI4MC00NTMzLTk1YmItMTAwNjFlMmFlNDUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTA3VDEzNDcwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2MGQ4Nzg4ZDZjZjkxOGYxNTU4ZTg4OGZlNDFlNDY0YWM5NjYwNmY1MTcwZDhiMWE2MjUxOGFmYTgyMmFiMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.LLzHz0Ed_TDGHuAX7Igj1QY1SqmGdMZrQwqk-GLKOaQ">
> 
> Cela fonctionne toutes les minutes.

It worked for me, awesome, thanks !

sbraz, 2025-11-13:
Pas de problème ici en v17. J'ai seulement dû mettre  `kWh_precision: 2` (je me demande pourquoi c'est à 0 [par défaut](https://github.com/Koenkk/zigbee-herdsman-converters/blob/28f3190ff3c9c657db066387cb58644009fb5c10/src/devices/lixee.ts#L173)) dans ma config Zigbee2MQTT 2.6.3.

Je laisse le reporting par défaut (de toute façon quand je le change, il est écrasé immédiatement, peut-être un bug du converter) et j'ai bien une remontée à chaque augmentation de 100 Wh, ce qui correspond à la valeur de _reportable change_ :

[image]

```
homeassistant ~ $ mosquitto_sub -t 'zigbee2mqtt/bridge/devices' -C 1 | jq '.[] | select(.model_id == "ZLinky_TIC").endpoints["1"].configured_reportings[] | select(.attribute == "currentSummDelivered")'
{
  "attribute": "currentSummDelivered",
  "cluster": "seMetering",
  "maximum_report_interval": 900,
  "minimum_report_interval": 60,
  "reportable_change": 100
}
```