fairecasoimeme/Zlinky_TIC #329: PTEC ne remonte plus avec FW v16 et Z2M 2.1.1 en TEMPO mode Historique

ScratMan, 2025-02-11:
Bonjour, depuis la mise à jour en v15 puis en v16 j'ai des problèmes de remontée d'infos sur le Zlinky_TIC, le PTEC ne se met plus à jour, sauf à demander la reconfiguration du Zlinky dans Z2M et faire un reset/pairage.
Une fois le PTEC remis à jour sur la valeur, il ne change plus et reste là dessus indéfiniment.

[image]

Je constate que j'ai beaucoup de warnings sur le Lixee dans les logs de Z2M, indiquant des erreurs de lecture d'attributs (dont le activeRegisterTierDelivered correspondant au PTEC):

[image]

Que se passe-t-il ?

Merci

ScratMan, 2025-02-12:
J'ai poursuivi mes essais, et il semble que le problème vienne  du paramètre `Measurement poll chunk` qui était à 1, la valeur par défaut. J'ai mis 4, après avoir trouvé l'info dans un autre fil de discussion qu'il semble que ça puisse provoquer des erreurs de lecture.
Depuis les warning sur Lixee ont disparus et le PETC est bien remonté en HPJW ce matin à 6h.
Je vais vérifier ce soir que ça continue bien.

Question : lorsque l'on fait un reset du Zlinky et un nouvel appairage, sans avoir supprimé l'appareil de Z2M ou HA au préalable, est-ce que les paramètres sont remis à leurs valeurs par défaut dans le Zlinky et dans Z2M ? Et d'où le système tire-t-il ses valeurs par défaut ? Du convertisseur Z2M ou du firmware du Zlinky ?

N'y aurait-il pas un mismatch entre les les réglages par défaut de Z2M et du Zlinky qui provoquerait ces problèmes de remontées ?
Je n'avais jamais eu à toucher aux paramètres du Zlinky dans Z2M à part pour choisir le nombre de décimales sur les indexes, et depuis la v15 j'ai l'impression que si je ne fais pas des réécritures des paramètres dans le Zlinky après un pairage, ça ne fonctionne pas correctement.

fairecasoimeme, 2025-02-12:
dans le plugin il y a une coquille car dans le text le measurement poll chunk est à 1 alors que dans le réel il est à 4 par défaut

voir le [Lixee.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lixee.ts)

ScratMan, 2025-02-12:
Lorsque j'avais mis à jour en v15 et que j'ai rencontré les problèmes de remontées, j'avais regardé dans les paramètres spécifiques dans Z2M et ils étaient vides, j'avais alors remis les valeurs qui étaient indiquées comme étant celles par défaut. C'est peut-être pour ça que c'est resté à 1 par la suite et que ça a continué à dysfonctionner en v16.

Là le PETC est bien passé à HCJW à 22h. Le problème semble résolu.

closed 2025-02-12 by ScratMan