fairecasoimeme/Zlinky_TIC #330: Soucis de délai remonté Z2M HA

BretonLud, 2025-02-14:
Bonjour, 

Tout d'abord merci pour le développement du Zlinky, 

Je dispose du Zlinky V2.0 en firmware 16. 

J'ai un soucis de délai pour certaine entité dont celles qui m'intéresse pour faire ma domotique. 

Je me suis mis devant mon compteur et à 22h00:10 exactement je suis passé en heure creuse et le relais d'enedis c'est enclenché. 

Hors sur Z2M le payload envoie le relais à 1 qu'a 22h02:05 (image 1)  surtout que le STGE a bien changé lui à 22h00:11 (image 2)

[image]

[image]

J'ai à la base pris le Zlinky pour permettre de me passer de l'api de rte tempo :)

fairecasoimeme, 2025-02-17:
Le mois prochain il y aura surement une mise à jour du plugin Z2M permettant de mieux gérer la tarification et les données du STGE

closed 2025-02-17 by fairecasoimeme

BretonLud, 2025-02-17:
Ok merci de l'information 

Pour le moment je récupère les données en requete mqtt (par exemple la couleur du lendemain qui n'est pas affiché dans Z2M) 

Je ne sais pas ou c'est géré la partie zlinky dans Z2M, mais si y a possibilité de créer les entités "utiles"

fairecasoimeme, 2025-02-17:
attention les couleurs du lendemain sont envoyé par le Linky entre 22h et 6h il me semble.

BretonLud, 2025-02-17:
> attention la couleur du lendemain est envoyée par le Linky entre 22h et 6h il me semble.

20h normalement mais ça c'est géré :)