fairecasoimeme/Zlinky_TIC #336: Zlinky HS après mise à jour v16 ?

looloo06, 2025-03-01:
Bonjour,

J'ai effectué une mise à jour OTA de mon Zlinky (v1) depuis la version 12 vers la version 16 depuis Home Assistant (Z2M).
Une fois la mise à jour terminée, plus de communication avec le Zlinky.
En allant voir ce dernier, la led bleue clignotte régulièrement.
J'ai testé : 
- débrancher puis rebrancher le Zlinky
- Retrait ZLinky de Z2M et Reset ZLinky pour réappairage : le Zlinky continue à clignotter après le reset, il est vu et réappairé dans Z2M puis j'ai des erreurs et aucune communication avec le Zlinky.

Y'aurait-il un moyen de réparer cela ?

Merci !

looloo06, 2025-03-01:
Pour compléter, voici l'état dans Z2M après plusieurs tentatives d'interview suivant le reset et réinclusion dans Z2M (et redémarrage de Z2M également) : 

{
    "current_tarif": "TEMPO",
    "linkquality": 255,
    "update": {
        "installed_version": 16,
        "latest_version": 16,
        "state": "idle"
    },
    "MOTDETAT": null,
    "active_energy_out_d01": null,
    "active_energy_out_d02": null,
    "active_power": null,
    "active_power_max": null,
    "active_power_ph_b": null,
    "active_register_tier_delivered": null,
    "apparent_power": null,
    "available_power": null,
    "average_rms_voltage_meas_period": null,
    "current_date": null,
    "current_index_tarif": null,
    "current_price": null,
    "current_summ_delivered": null,
    "current_tier1_summ_delivered": null,
    "current_tier2_summ_delivered": null,
    "current_tier3_summ_delivered": null,
    "current_tier4_summ_delivered": null,
    "current_tier5_summ_delivered": null,
    "current_tier6_summ_delivered": null,
    "drawn_v_a_max_n1": null,
    "message1": null,
    "message2": null,
    "meter_serial_number": null,
    "power_threshold": null,
    "relais": null,
    "rms_current": null,
    "rms_current_max": null,
    "rms_voltage": null,
    "schedule_h_p_h_c": null,
    "site_id": null,
    "software_revision": null,
    "status_register": null,
    "tomorrow_color": null,
    "warn_d_p_s": null
}

fairecasoimeme, 2025-03-03:
Depuis la version 14, le principe de clignotement de led a changé.
https://github.com/fairecasoimeme/Zlinky_TIC?tab=readme-ov-file#voyant-lumineux

closed 2025-03-03 by fairecasoimeme