fairecasoimeme/Zlinky_TIC #338: Vérifier la version du Zlinky ?

stadros83, 2025-03-04:
Bonjour,

Comment fait-on pour vérifier la version installée du ZLINKY avec Home assistant et ZHA ?
J'ai Micrologiciel : 0x00000001.
J'ai essayé de faire la mise à jour via OTA mais je n'arrive pas à voir si elle est passée ou non.
Est-ce que dans "Gérer l'appareil Zigbee" on peut trouver ça ?

starox, 2025-03-04:
Je suis sous zigbee2mqtt.
Lorsque je regarde sous home-assistant, j'ai ` Micrologiciel : 4000-0014 `
Ce qui correspond à la version 14.

stadros83, 2025-03-04:
Arf, je commence à me demander si mes précédentes mises à jour avaient vraiment fonctionné ...

BrunoSpy, 2025-10-26:
La version est donnée en hexa dans ZHA. La dernière version est 0x00000011 (soit 17) : 

[image]