fairecasoimeme/Zlinky_TIC #340: Données d'injection avec Jeedom

Fab5938, 2025-03-21:
Bonjour
je suis en Jeedom V4, j'ai un boitier Lixee Zlinky . Chez Enedis je suis en mode standard et reconnu comme producteur. 
Comment puis-je faire pour avoir mes données d'injection ?
Cordialement

fairecasoimeme, 2025-03-24:
Si vous êtes en mode standard avec le mode producteur, il suffit d'interroger les clusters / attributs correspondant à la production comme EAIT et SINSTI, vous retrouverez les infos dans le README

closed 2025-03-24 by fairecasoimeme

fairecasoimeme, 2025-03-26:
Je pense qu'il faut voir avec eux. je ne suis pas spécialiste Jeedom

Fab5938, 2025-04-27:
Bonsoir
J'ai vu avec l'équipe Jeedom, il faut forcer la synchronisation dans Jeedom après avoir mentionner que l'on est producteur pour prendre en compte cette information. Pour que tout ce stabilise il faut 2-3j, je suis passé en version 16 du firmware sans souci.
Par contre je constate que les données remontées pour CCASN et CCAIN sont mises à jour que toutes les 30mn. 
Est-ce normal ?
Peut-on réduire cet intervalle ?
Cordialement