fairecasoimeme/Zlinky_TIC #342: Remontée triphasée a la place de monophasée

foenyx-dev, 2025-04-05:
Bonjour,

J'ai un problème bizarre avec la remontée d'informations du ZLinky. J'ai déménagé il y a peu et, avant, tout semblait correct dans les informations remontées, c'est-à-dire que je n'avais pas ces informations complémentaires sur les phases A/B/C.

Sur le Linky de mon nouveau logement, j'ai l'impression qu'il détecte le compteur comme étant en triphasé plutôt qu'en monophasé.

J'ai bien vérifié au cas où, mais je suis bien en monophasé. En arrivant dans ce nouveau logement, j'ai demandé le passage au mode standard, ce qui a été fait et vérifié sur l'écran du Linky.

Je viens de mettre à jour le firmware du ZLinky de la v1 à la v16 pour être sûr que cela ne soit pas dû à un bug, mais cela n'y change rien.

Je suis sur ZHA dans Home Assistant.
Le Linky est en monophasé et en mode TIC standard.
La version du ZLinky sur le circuit imprimé est la v2.1.

Voici les métriques remontées par le ZLinky.

[image]

[image]

Que puis-je faire ?

foenyx-dev, 2025-04-12:
J'ai un peu creusé aujourd'hui,
Je suis bien en linky_mode 1 qui correspond au standard monophasé.
Le mystère s'épaissit...

[image]

foenyx-dev, 2025-04-16:
J'ai acheté la v2 en me disant que peut-être le hardware était défectueux et en fait, même souci.

Soit c'est mon Linky et sa prise TIC qui déconnent sot le soft du ZLinky.

Ou moi qui comprend peut-être rien et que cette phase B soit normale, mais ça paraît étrange.

@fairecasoimeme avez vous un avis ?

fairecasoimeme, 2025-04-16:
Bonjour, Je pense qu'il faut plutôt regarder du côté de ZHA. le Linky ne remonte pas de W mais de VA. Le ZLinky ne remonte pas de puissance autre que la puissance apparente. Vérifiez du côté de vos scripts ZHA ou plug-in

foenyx-dev, 2025-04-17:
Merci pour la réponse, je vais essayer d'aller creuser par là-bas. Je n'utilise pas de script ou de plugin custom, simplement ZHA out of the box.

foenyx-dev, 2025-04-17:
Après quelques recherches j'ai l'impression que c'est effectivement l'intégration [ZHA mise a jour il y a 3 mois](https://github.com/home-assistant/core/blame/cadbb623c75d2af831dbc14de5213da25870da82/homeassistant/components/zha/sensor.py#L53) qui ajoute cette phase B/C etc... 

De ce que je comprend de votre retour ZHA n'a de ce fait jamais fonctionné correctement chez moi car j'avais bien des W sur le sensor puissance.

Du coup, le mapping des sensors sous ZHA est farfelu si je comprend bien.

Si je récapitule :  

ZLinky (v1.2 PCB, v16 firmware ou v2 PCB, v16 firmware) ✅ ----> [ZHA quirks](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lixee/zlinky.py) (Quirk: zhaquirks.lixee.zlinky.ZLinkyTICFWV15) ✅ ----> [ZHA Sensor](https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py) mapping ❌

Je sais pas trop comment je vais m'en sortir, je vais surement essayer zigbee2mqtt pour comparer, mais j'aimerais que ZHA fonctionne out of the box.

Sinon j'imagine qu'a l'aide du quirk je peux faire mes prorpes sensor ? J'avoue n'avoir encore jamais essayé, je vais explorer cette piste.